raspibolt/raspibolt #233: raspibolt welcome script not showing any LND data

Sowiedu, 2018-09-23:
Hi,
I updated to the latest lnd and bitcoin versions and also recreated the startup script of the bonus section which should show me data about the status of my raspi, bitcoin and lnd.

The first two parts work falwlessly, but in the Lightning (LND) tab I get 0 information.
When viewing lncli getinfo I see active channels and so on, but it seems that the motd script doesnt see these.

Not sure which information you need to help me, here's a screenshot:

[image]

q0de, 2018-10-18:
still having this issue, I've tried Sowiedu script but no dice

q0de, 2018-10-23:
> Yeah, it's been neglected lately and is not quite up-to-date. Will fix that soon.

any update on this? thank you

CodingMuziekwijk, 2018-10-23:
@q0de I did a pull request (#255 ) a week ago to fix it. but Stadicus hasn't been active since. You can check the code changes (it's not that much).

oitdmser, 2018-11-17:
> any update on this? thank you

The raspibolt startup script has been updated to work with lnd v0.5 and should be working now

closed 2018-11-22 by Stadicus

starkana, 2018-12-16:
I seem to have the same issue with v0.5.1.  It displays the info on the MOTD but not when you execute the command.

rbuckhoy, 2019-01-03:
Same issue. It was working at first. Im wondering if order of operation on startup is an issue.

rbuckhoy, 2019-01-04:
Wow. Worked immediately with sudo. Does admin need some permissions or?

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>