raspibolt/raspibolt #236: Feature request: Use EPS with raspibolt via internet

Sowiedu, 2018-09-23:
Hi,
I just successfully configured my raspibolt to be used as electrum backend and it works flawlessly! Thank you so much for sharing your know-how, it is much appreciated!
Now is it possible (via dyndns or some other stuff) to be able to use my raspibolt as electrum backend when I am out and about, and not only in the local network? If so how would I do that? Can you add a bonus guide?

Stadicus, 2018-09-24:
That should be pretty straight-forward. I just noticed that I did not include the UFW setup in the EPS bonus guide and fixed that. 

You just need to extend the protective UFW rule to allow access to port 50002 from everywhere and set up port-forwarding in your router.

```
$ sudo ufw allow 50002 comment "allow EPS"
$ sudo ufw enable
$ sudo ufw status
```

Please be aware that this opens an additional attack vector and might make your RaspiBolt vulnerable to DoS attacks. Use at your own risk.

Sowiedu, 2018-09-24:
I'm not quite sure how to proceed. I added the rule and I opened a dyndns account which is working(verified by pinging the address and it finds the correct ip and the ping is successfull).
I made port forwards in my router to my raspibolt for the port 50002. When I input my dyndns address and the port 50002 in my electrum wallet it doesn't work tho. Any pointers? Tell me if this is out of scope then I will look elsewhere.

Sowiedu, 2018-09-30:
I switched my DynDNS provider and nmap show the port as "filtered" now, not as closed anymore. Still my Electrum wallet can't seem to connect to it despite pinging the dynamic dns name works fine.

Stadicus, 2019-01-08:
Hej @Sowiedu, were you able to resolve this issue?

Sowiedu, 2019-01-09:
Hey @Stadicus. Not yet, I moved and had to set up stuff again and actually got it all working fine a couple of days ago. I will try to make this specific issue work til end of january since I will need it to work around that time :P. It is not forgotten.

Sowiedu, 2019-01-09:
While configuring LND and bitcoind to work with / use ipv6 I noticed that the port forwards of my rotuer defaulted to the link local ipv6 adress and no the global one. When I forced to forward the ports to that global ipv6 adress these issues were resolved and I am hoping that the same will take place with EPS via internet. Will report back.

Sowiedu, 2019-01-11:
Sadly no matter what I do, nothing seems to fix it. Strange thing is that according to several websites, the port 8333 for bitcoin is open, yet any other port that has been opened the exact same way in my router doesn't seem to work at all. I even registered with a port mapping provider now to circumvent the DSLite tunneling problem that messes everything up. I try to connect with my phone to my electrum wallet via the mobile network. If I'm not mistaken we don't have IPv6 on mobile in germany yet. So with the Portmapping provider it should work. I messed around for hours today and can't seem to find a solution. I even think my ISP could be blocking ports. I can't really try to connect to it from outside without my smartphone since I cant install electrum on my notebook for various reasons. I know it's asking a lot but if someone more experienced with networking and IPv4/IPv6/DSLite/Forwarding/Mapping could troubleshoot this with me from the outside it may be a tremendous help.

Sowiedu, 2019-01-12:
Upon further research I might have found the issue that leads to only one port forwarding rule being actually enabled - someone with more knowledge please give some input:

Apparently with DSLite you can have only exactly one forwarding rule due to the fact that you are still kinda using IPv4. Can someone confirm?

Stadicus, 2019-01-29:
Hej @Sowiedu, kudos for your endurance. Unfortunately this is way out of my league... I'll leave it open for a few more weeks though, maybe someone stumbles upon that.

Sowiedu, 2019-01-29:
Yeah I have no more time to take care and look for a solution since I will be gone for a few months from that machine. Feel free to close it since 1) I cant verify potential solutions and 2) It's probably an Issue that has to do with my ISP and its IPv4/6 configuration.

ClaudeNovaMB, 2019-01-29:
Hey guys, Im stuck at the step of making the directory for EPS. Says Permission denied. I tried sudo but it asks for a password. not sure how to proceed. I created the link without a problem, just not the directory.
[image]

badokun, 2019-01-29:
Shouldn't you execute that as the admin user, instead of the bitcoin one? If you enter exit, followed by the mkdir command that should work

ClaudeNovaMB, 2019-01-30:
The step prior says open a bitcoin user session and change to home directory. On my other pi I installed eps as an admin user outside the bitcoin user session which did work. not skilled enough to know the security and "other" impacts if any

Stadicus, 2019-01-31:
This command should be run by user "bitcoin", which also should be the owner of the external hdd. Try running this command again first:
`$ sudo chown -R bitcoin:bitcoin /mnt/hdd/`

ClaudeNovaMB, 2019-01-31:
That worked perfectly. Thanks so much!

closed 2019-02-08 by Stadicus