raspibolt/raspibolt #238: Can't start LND after update

Argonath, 2018-09-23:
Hi, I'm running into an issue after updating my raspibolt to LND v0.5 and Bitcoid v0.16.3.

`Sep 23 14:32:41 raspibolt systemd[1]: Started LND Lightning Daemon.
Sep 23 14:32:41 raspibolt systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Sep 23 14:32:41 raspibolt systemd[1]: lnd.service: Unit entered failed state.
Sep 23 14:32:41 raspibolt systemd[1]: lnd.service: Failed with result 'exit-code'.
`

`● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Sun 2018-09-23 14:40:45 -03; 25s ago
  Process: 969 ExecStart=/usr/local/bin/lnd --externalip=${PUBLICIP} (code=exited, status=1/FAILURE)
 Main PID: 969 (code=exited, status=1/FAILURE)
`

The problem seems to be related to the Public IP file. Everything was working as intended before the update. But I don't know if the update was actually the cause.

Stadicus, 2018-09-24:
Can you post the full command output of LND using this method?
https://github.com/Stadicus/guides/issues/220#issuecomment-423812828

dennisreimann, 2018-10-12:
@TomasHo Thanks for the pointer, this was my problem as well. There seems to be [an explicit error for that](https://github.com/lightningnetwork/lnd/commit/58ab6c1912a1fcb78e06fdc94f4cd5c41141a4a5), but unfortunately it did not show up for me in the output. x

ManveerJarosz, 2018-10-31:
@TomasHo @dennisreimann What did you guys change the second port to? Is there any rule of thumb we should follow?

dennisreimann, 2018-10-31:
@ManveerJarosz My config now looks like this:

```
zmqpubrawblock=tcp://127.0.0.1:28332
zmqpubrawtx=tcp://127.0.0.1:28333
```

I have used [these recommendations](https://github.com/alexbosworth/ln-service#install-lnd-andor-your-bitcoin-chain-daemon) for the settings.

ManveerJarosz, 2018-10-31:
@dennisreimann thank you :)

My LND still won't start though 😞 I think there are more issues with my system 😅

ManveerJarosz, 2018-10-31:
Looks like removing `debughtlc` from lnd.conf did the trick!

dennisreimann, 2018-11-02:
I just found that the FAQ describes the [LND upgrade](https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_faq.md#how-to-upgrade-lnd) and mentions this too :)

Stadicus, 2018-11-13:
So, everything working again? :-)

closed 2018-12-23 by Stadicus