raspibolt/raspibolt #240: After update to lnd 0.5-beta and bitcoin-core 0.17.0 cannot unlock wallet on a testnet

dimarkov, 2018-09-24:
Hi, 

after following your instructions for updating lightning client and bitcoin core to newest version, 
I end up with following error when I try to unlock freshly created wallet as admin:

[lncli] unable to read macaroon path (check the network setting!): open /home/admin/.lnd/data/chain/bitcoin/mainnet/admin.macaroon: no such file or directory

I cannot figure out why is it looking for mainnet admin.macaroon as I created wallet only on the testnet, 
and tried to unlock it as admin
 
`lncli --network=testnet unlock`

Any idea what could be forcing lncli to look for the mainnet macaroon? 

I have checked both bitcoin.conf and lnd.conf, both are set for the testnet. 

Which i also see if i run lnd as bitcoin user

> 2018-09-25
01:23:05.536 [INF] LTND: Version 0.5.0-beta commit=3b2c807288b1b7f40d609533c1e96a510ac5fa6d
2018-09-25 01:23:05.536 [INF] LTND: Active chain: Bitcoin (network=testnet)
2018-09-25 01:23:05.538 [INF] CHDB: Checking for schema update: latest_version=6, db_version=6
2018-09-25 01:23:05.549 [INF] RPCS: password RPC server listening on [::]:10009
2018-09-25 01:23:05.549 [INF] RPCS: password gRPC proxy started at 127.0.0.1:8080
2018-09-25 01:23:05.550 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.

Janaka-Steph, 2018-10-11:
I had the same issue. I ended up keeping the default paths for macaroons files (not specifying any in lnd.conf), and --network=testnet is mandatory when on testnet.
Useful trick, add `alias tlncli='lncli --network=testnet'` to /home/bitcoin/.bashrc 

Have a look at https://github.com/lightningnetwork/lnd/issues/533

Stadicus, 2018-11-13:
I just updated the whole guide from beginning to testnet, building a new RaspiBold from scratch. Can you check out if your settings match the ones I just validated?

dimarkov, 2018-11-16:
Thanks for the updated guides, unfortunately it still does not work (although settings match). I will just wipe everything and start from scratch.

closed 2018-11-16 by dimarkov