raspibolt/raspibolt #248: Failed to start Bitcoin daemon. bitcoind.service: Failed with result 'signal'.

RobJah, 2018-10-03:
Hi there people,

I guess they dont come more newbie than me, since i bought my first raspberry specifically to buidl it into a raspibolt. So yeh, I dont really understand what im doing, eventhough im learning super much from the guide. I solved a lot of problems that occured myself already but now i really have to get some of you wizards to save meh from this one. 

As seen below i tried to manually start the deamon since it didnt boot on startup, or rather it stayed on 'activating' before ultimately failing again and again.

I hope this makes sense to some one. 

FIRST COMMAND

root@LIGHTNINGpi:/home/pi# sudo service bitcoind start
Job for bitcoind.service failed because a fatal signal was delivered to the control process.
See “systemctl status bitcoind.service” and “journalctl -xe” for details.

NEXT COMMAND 

— Unit bitcoind.service has begun starting up.
Oct 03 19:43:07 LIGHTNINGpi bitcoind[913]: ************************
Oct 03 19:43:07 LIGHTNINGpi bitcoind[913]: EXCEPTION: N5boost10filesystem16filesystem_errorE
Oct 03 19:43:07 LIGHTNINGpi bitcoind[913]: boost::filesystem::create_directory: File exists: “/home/bitcoi
Oct 03 19:43:07 LIGHTNINGpi bitcoind[913]: bitcoin in AppInit()
Oct 03 19:43:07 LIGHTNINGpi bitcoind[913]: bitcoind: chainparamsbase.cpp:70: const CBaseChainParams& BaseP
Oct 03 19:43:07 LIGHTNINGpi systemd[1]: bitcoind.service: Control process exited, code=killed status=6
Oct 03 19:43:07 LIGHTNINGpi systemd[1]: Failed to start Bitcoin daemon.
— Subject: Unit bitcoind.service has failed
— Defined-By: systemd
— Support: https://www.debian.org/support
—
— Unit bitcoind.service has failed.
—
— The result is failed.
Oct 03 19:43:07 LIGHTNINGpi systemd[1]: bitcoind.service: Unit entered failed state.
Oct 03 19:43:07 LIGHTNINGpi systemd[1]: bitcoind.service: Failed with result ‘signal’.
Oct 03 19:43:36 LIGHTNINGpi systemd[1]: Starting Cleanup of Temporary Directories…
— Subject: Unit systemd-tmpfiles-clean.service has begun start-up
— Defined-By: systemd
— Support: https://www.debian.org/support
—
— Unit systemd-tmpfiles-clean.service has begun starting up.
Oct 03 19:43:36 LIGHTNINGpi systemd[1]: Started Cleanup of Temporary Directories.
— Subject: Unit systemd-tmpfiles-clean.service has finished start-up
— Defined-By: systemd
— Support: https://www.debian.org/support
—
— Unit systemd-tmpfiles-clean.service has finished starting up.
—
— The start-up result is done.
Oct 03 19:43:37 LIGHTNINGpi systemd[1]: bitcoind.service: Service hold-off time over, scheduling restart.
Oct 03 19:43:37 LIGHTNINGpi systemd[1]: Stopped Bitcoin daemon.
— Subject: Unit bitcoind.service has finished shutting down
— Defined-By: systemd
— Support: https://www.debian.org/support
—
— Unit bitcoind.service has finished shutting down.
Oct 03 19:43:37 LIGHTNINGpi systemd[1]: Starting Bitcoin daemon…
— Subject: Unit bitcoind.service has begun start-up
— Defined-By: systemd
— Support: https://www.debian.org/support

NEXT COMMAND

root@LIGHTNINGpi:/home/pi# systemctl status bitcoind.service
● bitcoind.service – Bitcoin daemon
Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
Active: activating (auto-restart) (Result: signal) since Wed 2018-10-03 19:46:08 CEST; 22s ago
Process: 965 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/h
Process: 961 ExecStartPre=/bin/sh -c sleep 30 (code=exited, status=0/SUCCESS)

Oct 03 19:46:08 LIGHTNINGpi systemd[1]: Failed to start Bitcoin daemon.
Oct 03 19:46:08 LIGHTNINGpi systemd[1]: bitcoind.service: Unit entered failed state.
Oct 03 19:46:08 LIGHTNINGpi systemd[1]: bitcoind.service: Failed with result ‘signal’

RobJah, 2018-10-04:
Maybe i found another clue to why this is not working:

Im using a NTFS hdd with the btc blckchain allready on it. Long story short: there was no other way.
I managed to install the ntfs-3g driver to mount the hdd correctly. After that the raspibolt guide told me to add a symbolic link that points to the external harddisk.

to do so i used: $ ln -s /mnt/hdd/bitcoin /home/bitcoin/.bitcoin
If i do it again now it says the symbolic link already excists. But i wonder if this is the correct link, since i use the ntfs-3g drive and therefore my mounting command line is different right?
shouldnt i use something like: $ ln ntfs-3g -s /mnt/hdd/bitcoin /home/bitcoin/.bitcoin
cause that piece of extra code 'ntfs-3g' was necessary to mount the hdd aswell (i used ntfs-3 /mnt/hdd in my /etc/fstab) 

currentlymy symobolic link looks like this:
[image]

RobJah, 2018-10-04:
i managed to get it done.
it was a combination of 2 small things i didnt do right earlier on in the guide. problem is that some things dont work anymore like the guide says so i had to find work arounds which now turned out to bite me in the ass. that plus me running a NTFS hdd.

if you, the author, ur open for it, we could do a noob-review of the guide :)

Stadicus, 2018-11-13:
Hi @RobJah, please feel free to point out any difficulties you encountered. That's how the guide grew into what it is today, and I'm sure there's lots of room to improve.

One lesson was that NTFS just causes too many troubles. This is why I removed it completely (it was an option in earlier versions).

Glad you got it working. Kudos for your perseverance! :-)

closed 2018-12-23 by Stadicus

RobJah, 2018-12-29:
Stadicus!

I only now see ur reply. Im sorry i dont remember all the things a i ran into BUT i do remember the NTFS hdd not mounting because the guide didnt tell me to first install a driver for my pi to be able to read and write NTFS.

If i ever run into new stuf, cause im on here regularly and am actually still a noob really, i will let you know in new tickets !

RobJah, 2018-12-29:
Plus if I get the time on my week off (2 weeks from now) I will walk through the whole current guide and let u know what i coming up with
ill include the tor part, which gave me some trouble a week agow  hehe

Stadicus, 2018-12-29:
Would appreciate that, thanks!