raspibolt/raspibolt #249: Use auto nat discover starting in lnd 0.5-beta

frennkie, 2018-10-06:
As of lnd 0.5-beta the software can discover the external IP address of a NAT device/home router using either UPnP or NAT-PMP. This eliminates the need for the Public IP script and passing the `--externalip=${PUBLICIP}` parameter. I tested this on my network and it is working perfectly fine. This also detects IP address changes (e.g. after a disconnect).

I had to take the following steps for this (my router is capable of UPnP):

* setup a `ufw` rule to allow successful UPnP discovery over SSDP
  * `sudo ufw allow proto udp from 192.168.0.0/24 port 1900 to any comment 'allow local LAN SSDP for UPnP discovery'`
* remove `--externalip=${PUBLICIP}` from `/etc/systemd/system/lnd.service`
* set `nat=true` in `/home/bitcoin/.lnd/lnd.conf`
* on my NAT device/home router (AVM Fritzbox)
  * configure my router to allow UPnP in general
  * remove the static port forwarding for port 9735
  * allow the host/ip running `lnd` to automatically setup port forwardings (this is **important**! `lnd` will currently only use the discovered external IP if it can setup the port forwarding itself)

Without the `ufw` rule I had the following log messages:

```
2018-10-06 11:09:35.658 [INF] SRVR: Scanning local network for a UPnP enabled device
2018-10-06 11:09:45.658 [ERR] SRVR: Unable to discover a UPnP enabled device on the local network: context canceled
```

After a 10 second timeout it gave up the UPnP discovery.

Now it looks like this:

```
2018-10-06 11:17:15.059 [INF] SRVR: Scanning local network for a UPnP enabled device
2018-10-06 11:17:19.357 [INF] SRVR: Automatically set up port forwarding using UPnP to advertise external IP
```

waffi121, 2018-10-06:
nice! a dream comes true: D

works for me, FritzBox: 2 via UPnP open ports (TCP 9735, UDP 9735)

Thank you.

dennisreimann, 2018-11-11:
Awesome, thanks a lot, this made it work for me too!

Stadicus, 2018-11-13:
Thanks for documenting this! I tried it myself but did not think to open the UDP port. Finally, we're able to get rid of the hacky getip script. Works like a charm with my ASUS RT-AC66U router.

Stadicus, 2018-11-13:
https://github.com/Stadicus/guides/commit/e0f3f264aea40de7eb4bca2181373b77d086baa8#diff-cf8452295da60ae23ef1c54a61d7bf2e

closed 2018-11-13 by Stadicus

frennkie, 2018-11-14:
@Stadicus Go you get a new/different external IP assigned from your provider? And if yes: What happens to the forwarded port? In my case the UPnP forwarding was removed on the router. And `lnd` did not renew the forwarding. Therefore I'm still working with `lnd` team to get this running completely stable...

starkana, 2018-12-15:
I have an Apple AirPort Extreme router which only seems to support NAT-PMP.  No matter what I try I keep getting the error:

`2018-12-15 12:44:09.098 [ERR] SRVR: Unable to discover a NAT-PMP enabled device on the local network: fork/exec /usr/bin/ip: no such file or directory

2018-12-15 12:44:09.098 [ERR] SRVR: unable to create server: Unable to discover a NAT-PMP enabled device on the local network: fork/exec /usr/bin/ip: no such file or directory

2018-12-15 12:44:09.098 [INF] LTND: Shutdown complete
Unable to discover a NAT-PMP enabled device on the local network: fork/exec /usr/bin/ip: no such file or directory`

- I have NAT Port Forwarding Protocol (NAT-PMP) enabled.
- Port 8333 (TCP & UDP), 18333 (TCP morning & UDP), and 9735 (TCP & UDP) forwarding to my Raspberry Pi Bitcoin & Lightning node on (10.0.1.2).
- I am following @Stadicus 's guide here: https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_40_lnd.md
- I am using lightning v0.5.1 (the guide references v0.5)

I spent the last 2 days trying to troubleshoot this and am feeling very discouraged and ready to give up... I suspect it may be the Apple router but there is a chance I missed to configure something. Any help is greatly appreciated. 

The model my router connects to is a ARRIS TM1602A. It's completely locked and there is no admin login. All I have to work with is the Apple router. ISP is Charter/Spectrum.

frennkie, 2018-12-16:
@starkana From the error message I assume your main/only(?!) issue is that `/usr/bin/ip` is not found. I recall having had an issue like this.

Make sure that  the iproute2 tools are installed:  `sudo apt-get install iproute2` and then check: `ls -l /usr/bin/ip`. If the file does not exists you can create a symlink to test whether this resolves your issue: `sudo ln -l /sbin/ip /usr/bin/ip`.

starkana, 2018-12-16:
@frennkie I love you so much right now! That did it. It needed symlink.  Damn debugging! Spent 2 days and I missed something so simple! =/

Stadicus, 2018-12-16:
Great you got it fixed. Did you use exactly the same os image (Raspbian Stretch) as mentioned? Strange I never ran into this...

frennkie, 2019-01-04:
@cilphex There are two issue: 

1. port forwardings not being re-created after a change in the external IP
2. coupling between the `nat=True` setting and the port forwarding 

For 1. I have a finished PR which I hope will be merged soon.  Until then I'm patching it manually.
For 2. I started a PR, but did not have the time to finish it.

cilphex, 2019-01-05:
@frennkie Do you mean PRs for this guide, or something else?

frennkie, 2019-01-05:
PRs on the LND project: 2014 and 2163.

KayBeSee, 2019-02-22:
> @frennkie what do you mean by this part in your original post/how do you do it?
> 
> > allow the host/ip running lnd to automatically setup port forwardings (this is important! lnd will currently only use the discovered external IP if it can setup the port forwarding itself)

I am wondering too what this means. How do you config automatic port forwardings?

I am currently stuck at the following error. I removed the static port forwarding and did the other configs you described in the original post.

```
Attempting automatic RPC configuration to bitcoind
Automatically obtained bitcoind's RPC credentials
2019-02-21 22:10:46.474 [INF] LTND: Version: 0.5.2-beta commit=v0.5.2-beta, build=production, logging=default
2019-02-21 22:10:46.474 [INF] LTND: Active chain: Bitcoin (network=mainnet)
2019-02-21 22:10:46.475 [INF] CHDB: Checking for schema update: latest_version=7, db_version=7
2019-02-21 22:10:46.479 [INF] RPCS: password RPC server listening on 127.0.0.1:10009
2019-02-21 22:10:46.479 [INF] RPCS: password gRPC proxy started at 127.0.0.1:8080
2019-02-21 22:10:46.479 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.
2019-02-21 22:11:04.682 [INF] LNWL: Opened wallet
2019-02-21 22:11:05.040 [INF] LTND: Primary chain is set to: bitcoin
2019-02-21 22:11:05.046 [INF] LTND: Initializing bitcoind backed fee estimator
2019-02-21 22:11:05.046 [INF] LNWL: Started listening for bitcoind block notifications via ZMQ on tcp://127.0.0.1:28332
2019-02-21 22:11:05.046 [INF] LNWL: Started listening for bitcoind transaction notifications via ZMQ on tcp://127.0.0.1:28333
2019-02-21 22:11:10.621 [INF] LNWL: The wallet has been unlocked without a time limit
2019-02-21 22:11:10.628 [INF] LTND: LightningWallet opened
2019-02-21 22:11:10.637 [INF] LNWL: Caught up to height 0
2019-02-21 22:11:10.653 [INF] HSWC: Restoring in-memory circuit state from disk
2019-02-21 22:11:10.654 [INF] HSWC: Payment circuits loaded: num_pending=0, num_open=0
2019-02-21 22:11:10.655 [INF] SRVR: Scanning local network for a UPnP enabled device
2019-02-21 22:11:20.661 [ERR] SRVR: Unable to discover a UPnP enabled device on the local network: context canceled
2019-02-21 22:11:20.662 [INF] SRVR: Scanning local network for a NAT-PMP enabled device
2019-02-21 22:11:20.694 [ERR] SRVR: Unable to discover a NAT-PMP enabled device on the local network: read udp 192.168.200.20:35328->192.168.200.1:5351: recvfrom: connection refused
2019-02-21 22:11:20.694 [ERR] SRVR: unable to create server: Unable to discover a NAT-PMP enabled device on the local network: read udp 192.168.200.20:35328->192.168.200.1:5351: recvfrom: connection refused

2019-02-21 22:11:20.694 [INF] LTND: Shutdown complete
Unable to discover a NAT-PMP enabled device on the local network: read udp 192.168.200.20:35328->192.168.200.1:5351: recvfrom: connection refused
```

frennkie, 2019-02-22:
@KayBeSee I see two ways how this could be going wrong for you:

1) Either your Router/NAT device does not support UPnP (or NAT-PMP) or it is disabled/misconfigured
2) The UPnP/NAT-PMP traffic is being blocked - e.g. by the firewall on your LND host

I recommend your check the firewall rule set.

KayBeSee, 2019-02-24:
This is my ufw status. I just copied the code from this guide. I ssh into the pi using `ssh xxx@192.168.200.20`, so I _think_ this is the correct setup. I have UPnP enabled on my router.

[image]

KayBeSee, 2019-02-24:
Actually, I was able to get the node up and running (past the `Unable to discover a NAT-PMP enabled device on the local network` step), but I still don't think the node is connected correctly b/c I can't find it in any explorers and can't open any channels to it. It looks like I am getting peers though.

When I try to open a channel using 0312bd1afe7ce6d67ac12902451e46e606ee7d3deb913a8dc58539c25189700b62@50.5.55.60:9735

I get `Unable to connect to peer, dial tcp 50.5.55.60:9735: connect: connection refused`

frennkie, 2019-02-24:
So if you were able to "get past `Unable to discover..`" does this mean a NAT device was detected? If yes, are there now dynamic port forwardings for your node?

KayBeSee, 2019-02-24:
Yes! I got it working!

Some other questions related to my experience:
Is it possible to run two lnd nodes on the same network? I was able to finally get the node connected by unplugging my Casa node from my router. 

I also noticed that the node didn't appear on any explorers until I opened a channel with someone else. Is this normal?

Finally, when I funded my node I generated a newaddress and sent an onchain transaction to it. I noticed that the address I sent it to looks like it confirmed the incoming transaction and then immediately sent the funds off somewhere else ([address here](https://www.smartbit.com.au/address/3CnLTHxNmSXy8h1d3GXf7fQFkKiVi8imxf)). Why is that?

Thanks again for helping me out! This has been a very educational exercise about how lightning works!