raspibolt/raspibolt #251: error: couldn't connect to server: unknown (code -1)

Start2Bitcoin, 2018-10-09:
Hi, 

When running Bitcoind on my raspibolt, I always get stuck at this error when calling `bitcoin-cli getblockchaininfo`

> error: couldn't connect to server: unknown (code -1)
> (make sure server is running and you are connecting to the correct RPC port)

All my connections and the RPC port are in order (followed by the provided guide).

This is my debug.log file:

> 2018-10-09 10:48:08 Cache configuration:
> 2018-10-09 10:48:08 * Using 12.5MiB for block index database
> 2018-10-09 10:48:08 * Using 8.0MiB for chain state database
> 2018-10-09 10:48:08 * Using 79.5MiB for in-memory UTXO set (plus up to 47.7MiB of unused mempool space)
> 2018-10-09 10:48:08 init message: Loading block index...
> 2018-10-09 10:48:08 Opening LevelDB in /home/bitcoin/.bitcoin/blocks/index
> 2018-10-09 10:48:08 Opened LevelDB successfully
> 2018-10-09 10:48:08 Using obfuscation key for /home/bitcoin/.bitcoin/blocks/index: 0000000000000000
> 2018-10-09 10:48:34 LoadBlockIndexDB: last block file = 1393
> 2018-10-09 10:48:34 LoadBlockIndexDB: last block file info: CBlockFileInfo(blocks=94, size=90416721, heights=544619...544805, time=2018-10-06...2018-10-07)
> 2018-10-09 10:48:34 Checking all blk files are present...
> 2018-10-09 10:48:35 LoadBlockIndexDB: transaction index disabled
> 2018-10-09 10:48:35 : You need to rebuild the database using -reindex to change -txindex.
> Please restart with -reindex or -reindex-chainstate to recover.
> 2018-10-09 10:48:35 Aborted block database rebuild. Exiting.
> 2018-10-09 10:48:35 Shutdown: In progress...
> 2018-10-09 10:48:35 scheduler thread interrupt
> 2018-10-09 10:48:35 Shutdown: done

Thanks in advance for helping me out.

Kind regards

saubyk, 2018-10-09:
Please check your bitcoin.conf file.
Do you have have txindex=1in your config?

Start2Bitcoin, 2018-10-09:
Yes, the hdd has a fully indexed blockchain.
This is my bitcoin.conf file:

> server=1
> daemon=1
> txindex=1

saubyk, 2018-10-09:
Was the blockchain copied from another computer?
If yes, was txindex enabled on that computer when you were building the blockchain?

Appears to be an indexing error.

saubyk, 2018-10-09:
Please refer to this thread as well...

https://bitcoin.stackexchange.com/questions/60709/when-should-i-use-reindex-chainstate-and-when-reindex

Start2Bitcoin, 2018-10-09:
Yes, the blockchain is copied  from my laptop with index=1 (fully sync.). There it worked perfect without errors. After it, I copied everything to the HDD (blocks, chainstate, index, ...)

- Is a reindex recommend, taking in consideration the cpu power a raspberry only has?

saubyk, 2018-10-09:
To be honest, never faced such an issue.
I did the same thing as you, copied the blockchain on HDD...and it ran without a hitch.

So cannot recommend anything specifc.

But I will take my chances with reindex-chainstate...could be a while on pi though.

Another option is to sync the chain again on your laptop and try copying again...see if that fixes it.

Start2Bitcoin, 2018-10-09:
No problem. Thanks for the help.
Regarding the slack, I'm not a member of the BTC slack. I will try. Thx

saubyk, 2018-10-09:
Btw reindex-chainstate on laptop seems to be a good idea...I will try posting your issue on Bitcoin slack. Will let you know if there is an update.

Start2Bitcoin, 2018-10-09:
Thanks, really appreciate it!

Jeans5, 2018-10-10:
I got Bitcoind running, but still got the same issue. Also after rebuilding it with reindex-chainstate. When calling `Bitcoin-cli getblockchaininfo`, I got another error message (maybe this is not related to the above problem):

> error: Could not locate RPC credentials. No authentication cookie could be found, and RPC password is not set.  See -rpcpassword and -stdinrpcpass.  Configuration file: (/home/admin/.bitcoin/bitcoin.conf)

saubyk, 2018-10-11:
What version of Bitcoind are you running on rpi?
What was the version on the mac?

Jeans5, 2018-10-12:
On the rpi, it's  v0.16.3. On my laptop it's v0.17 I believe.

saubyk, 2018-10-18:
Hi, sorry for a late response. Life keeps on intervening!

Hoping your problem is addressed now.

I've had to re-index on testnet chains when I upgraded to 0.16.3. But didnt do that on rpi mainnet chain.

If you've not already, you can try updating rpi to 0.17 and see if that helps.

If not, let me know what error your getting. Will try to engage again with Bitcoin slack.

jbbasics, 2018-10-29:
Hey guys,

I am getting the same errors. I reindexed on my laptop and upgraded the pi to 17.0 (same as laptop) but get the following:
2018-10-29T20:59:16Z Bitcoin Core version v0.17.0 (release build)
2018-10-29T20:59:16Z InitParameterInteraction: parameter interaction: -whitelistforcerelay=1 -> setting -whitelistrelay=1
2018-10-29T20:59:16Z Assuming ancestors of block 0000000000000000002e63058c023a9a1de233554f28c7b21380b6c9003f36a8 have valid signatures.
2018-10-29T20:59:16Z Setting nMinimumChainWork=0000000000000000000000000000000000000000028822fef1c230963535a90d
2018-10-29T20:59:16Z Using the 'standard' SHA256 implementation
2018-10-29T20:59:16Z Default data directory /home/bitcoin/.bitcoin
2018-10-29T20:59:16Z Using data directory /home/bitcoin/.bitcoin
2018-10-29T20:59:16Z Using config file /home/bitcoin/.bitcoin/bitcoin.conf
2018-10-29T20:59:16Z Using at most 40 automatic connections (1024 file descriptors available)
2018-10-29T20:59:16Z Using 16 MiB out of 32/2 requested for signature cache, able to store 524288 elements
2018-10-29T20:59:16Z Using 16 MiB out of 32/2 requested for script execution cache, able to store 524288 elements
2018-10-29T20:59:16Z Using 8 threads for script verification
2018-10-29T20:59:16Z scheduler thread start
2018-10-29T20:59:16Z HTTP: creating work queue of depth 16
2018-10-29T20:59:16Z Config options rpcuser and rpcpassword will soon be deprecated. Locally-run instances may remove rpcuser to use cookie-based auth, or may be replaced with rpcauth. Please see share/rpcauth for rpcauth auth generation.
2018-10-29T20:59:16Z HTTP: starting 4 worker threads
2018-10-29T20:59:16Z Cache configuration:
2018-10-29T20:59:16Z * Using 2.0MiB for block index database
2018-10-29T20:59:16Z * Using 12.2MiB for transaction index database
2018-10-29T20:59:16Z * Using 8.0MiB for chain state database
2018-10-29T20:59:16Z * Using 77.8MiB for in-memory UTXO set (plus up to 47.7MiB of unused mempool space)
2018-10-29T20:59:16Z init message: Loading block index...
2018-10-29T20:59:16Z Opening LevelDB in /home/bitcoin/.bitcoin/blocks/index
2018-10-29T20:59:16Z Opened LevelDB successfully
2018-10-29T20:59:16Z Using obfuscation key for /home/bitcoin/.bitcoin/blocks/index: 0000000000000000
2018-10-29T20:59:28Z LoadBlockIndexDB: last block file = 1413
2018-10-29T20:59:28Z LoadBlockIndexDB: last block file info: CBlockFileInfo(blocks=79, size=72452666, heights=547557...547695, time=2018-10-27...2018-10-28)
2018-10-29T20:59:28Z Checking all blk files are present...
2018-10-29T20:59:28Z Opening LevelDB in /home/bitcoin/.bitcoin/chainstate
2018-10-29T20:59:28Z Opened LevelDB successfully
2018-10-29T20:59:28Z Using obfuscation key for /home/bitcoin/.bitcoin/chainstate: 7505727cceec2d5b
2018-10-29T20:59:29Z : Error initializing block database.
Please restart with -reindex or -reindex-chainstate to recover.
2018-10-29T20:59:29Z Aborted block database rebuild. Exiting.
2018-10-29T20:59:29Z Shutdown: In progress...
2018-10-29T20:59:29Z scheduler thread interrupt
2018-10-29T20:59:29Z Shutdown: done
2018-10-29T20:59:30Z

Any help would be greatly appreciated.

>

Stadicus, 2018-11-18:
It's essential to run the same version of Bitcoin Core to download the blockchain and on the Pi. Please refer to this answer here:
https://github.com/Stadicus/guides/issues/257#issuecomment-439680005

If you copy the v0.16 blockchain to the Pi that is running 0.17, it will automatically upgrade the internal data structure (can take a few hours).

If you copy the 0.17 blockchain to the Pi that is running 0.16, just upgrade your Pi to use the latest Bitcoin Core binaries (see FAQ: https://github.com/Stadicus/guides/issues/257#issuecomment-439680005)

Jeans5, 2018-11-23:
Hi, my apologies for my delayed response. When updating to 0.17, the errors got fixed. Thanks for helping me out with this.

closed 2018-11-23 by Jeans5