raspibolt/raspibolt #252: eps issue: transaction only local

wwwurst, 2018-10-11:
#1: thanks to Stadicus for the excellent guides !

#2 I am unning eps 0.15 with bitcoind 0.17.0 on a Raspberry Pi in a "raspibolt" setup.

When my Electrum wallet connects to eps, one transaction shows up as "Status: local". The tx is from back in May, and all block explorers and public Electrum servers show it correctly and confirmed. The bitcoind update to 0.17.0 already ran a blockchain reindex. Is there anything else I can do do or try to further troubleshoot?

#3 it would be nice to include also the eps status in the raspibolt welcome message

Stadicus, 2018-12-23:
Sorry for the late reply, I gradually worked through all open issues while rebuilding my own RaspiBolt and the "bonus" section unfortunately is last... :-)

Were you able to resolve this issue or is it still open?

wwwurst, 2018-12-24:
Hi Stadicus, I made no progress on this issue. Now it seems electrum 3.3.2 will no longer connect to eps 0.15 at all. Older electrum version still have the issue dicribed above.  I'd really appreciate if you could add the upgrade procedure to eps 0.16 , given it's new installation scheme...

Stadicus, 2018-12-24:
Will do, thanks for the hint. :-)

Stadicus, 2019-01-08:
Guide has been updated for EPS 0.1.6
https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_64_electrum.md

closed 2019-01-08 by Stadicus