raspibolt/raspibolt #254: Part2 - Autounlock script edit for macaroon directory in LND v0.5

zwarbo, 2018-10-16:
For the unlock script in part 2 of the guide, I noticed that the LN_ROOT directory won't cover the new LND v0.5 placement for the admin.macaroon file location.

 Here is the snippet of the code and the changes I made:
- Added LN_MACN=/home/bitcoin/.lnd/data/chain/bitcoin/mainnet
- Changed LN_ROOT to LN_MACN in ${LN_MACN}/admin.macaroon)
Does that make sense? Seems to work ok.

```
LN_ROOT=/home/bitcoin/.lnd
LN_MACN=/home/bitcoin/.lnd/data/chain/bitcoin/mainnet

curl -s \
        -H "Grpc-Metadata-macaroon: $(xxd -ps -u -c 1000 ${LN_MACN}/admin.macaroon)" \
        --cacert ${LN_ROOT}/tls.cert \
        -d "{\"wallet_password\": \"$(cat /etc/lnd/pwd | tr -d '\n' | base64 -w0)\"}" \
        https://localhost:8080/v1/unlockwallet > /dev/null 2>&1
```

CodingMuziekwijk, 2018-10-17:
Thanks for pointing this out. I made a pull request (#255) with your solution added to the file. Also added a check for testnet/mainnet.

Stadicus, 2018-11-18:
Thanks for your contributions! The guide has been updated with the new unlock script (although mainnet is still hardcoded). Does that work for your?

closed 2018-11-22 by Stadicus