raspibolt/raspibolt #257: rebuild the database using -reindex to change -txindex.

grymjack, 2018-10-18:
I copied the two subdirectories "blocks" and "chainstate" from my main system to the Thundroid at "\mnt\hdd\bitcoin\".  Then started the service.  It seems to be cycling on an error telling me I have to "Please restart with -reindex or -reindex-chainstate to recover".   Any thoughts on why this is happening and how to fix this?

              -Thank you

*** logfile

2018-10-18 10:31:38 LoadBlockIndexDB: last block file = 1403
2018-10-18 10:31:38 LoadBlockIndexDB: last block file info: CBlockFileInfo(blocks=19, size=15648263, heights=546085...546222, time=2018-10-17...2018-10-18)
2018-10-18 10:31:38 Checking all blk files are present...
2018-10-18 10:31:38 LoadBlockIndexDB: transaction index disabled
2018-10-18 10:31:38 : You need to rebuild the database using -reindex to change -txindex.
Please restart with -reindex or -reindex-chainstate to recover.
2018-10-18 10:31:38 Aborted block database rebuild. Exiting.
2018-10-18 10:31:38 Shutdown: In progress...
2018-10-18 10:31:38 scheduler thread interrupt
2018-10-18 10:31:38 Shutdown: done

grymjack, 2018-10-19:
**Windows Client used to download the blockchain**
---------------------------------------
Bitcoin Core version v0.17.0 (64-bit)

**Bitcoin Core binary downloaded**
---------------------------------------
wget https://bitcoincore.org/bin/bitcoin-core-0.16.3/bitcoin-0.16.3-arm-linux-gnueabihf.tar.gz

**Thundroid daemon**
---------------------------------------
Bitcoin Core Daemon version v0.16.3

grymjack, 2018-10-19:
ok, I feel stupid.  I went and got the new binaries and updated them and it appears to be operating correctly now.  Being somewhat new to the Linux cli world, why wouldn't the bitcoin binaries get updated in a general "sudo apt update"?

jbbasics, 2018-10-29:
Hey guys, I'm having the same issue. Can you tell me how you solved it? Thanks, Josh

grymjack, 2018-10-30:
upgrade the bitcoin version to 0.17.0 on your ODroid

jbbasics, 2018-10-31:
I ended up getting it after deleting the blocks again and transferring. I think it may have been running when doing the first transfer. Thanks

Stadicus, 2018-11-18:
Bitcoin Core 0.17 uses different internal data structures than 0.16 and migrates them automatically when upgrading. But they are not backwards compatible, so upgrading the Thundroid to the latest bitcoin binaries is the correct solution.

Another common issue (but not here, afaict) is to download the Bitcoin blockchain on your regular comuter without setting the `txindex=1`. Then, the whole chain needs to be reindexed on your regular computer before being usable on the Raspberry Pi / Odroid.

closed 2018-12-23 by Stadicus