raspibolt/raspibolt #260: lnd.service: Main process exited, code=exited, status=1/FAILURE

grymjack, 2018-10-19:
Lightning Node service doesn't seem to want to sync and keeps going into a fail state and restarting.  Bitcoin is syncing fine, however, 

_when I do a "lncli getinfo", I get this error_
----------------------------------------
admin@GrymNodeOne:~$ lncli getinfo
[lncli] rpc error: code = Unavailable desc = grpc: the connection is unavailable
----------------------------------------

_This is the output from "sudo journalctl -f -u lnd"_
----------------------------------------
Oct 19 18:04:47 GrymNodeOne lnd[812]: 2018-10-19 18:04:47.358 [INF] LNWL: Started listening for blocks via ZMQ on tcp://127.0.0.1:29000
Oct 19 18:04:49 GrymNodeOne lnd[812]: 2018-10-19 18:04:49.936 [INF] LNWL: The wallet has been unlocked without a time limit
Oct 19 18:04:49 GrymNodeOne lnd[812]: 2018-10-19 18:04:49.936 [INF] LTND: LightningWallet opened
Oct 19 18:04:50 GrymNodeOne lnd[812]: 2018-10-19 18:04:50.751 [INF] LNWL: Catching up block hashes to height 546481, this will take a while...
Oct 19 18:04:51 GrymNodeOne systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Oct 19 18:04:51 GrymNodeOne systemd[1]: lnd.service: Unit entered failed state.
Oct 19 18:04:51 GrymNodeOne systemd[1]: lnd.service: Failed with result 'exit-code'.
Oct 19 18:05:51 GrymNodeOne systemd[1]: lnd.service: Service hold-off time over, scheduling restart.
Oct 19 18:05:51 GrymNodeOne systemd[1]: Stopped LND Lightning Daemon.
Oct 19 18:05:51 GrymNodeOne systemd[1]: Started LND Lightning Daemon.
Oct 19 18:05:51 GrymNodeOne lnd[916]: Attempting automatic RPC configuration to bitcoind
Oct 19 18:05:51 GrymNodeOne lnd[916]: Automatically obtained bitcoind's RPC credentials
Oct 19 18:05:51 GrymNodeOne lnd[916]: 2018-10-19 18:05:51.736 [INF] LTND: Version 0.4.1-beta commit=
Oct 19 18:05:51 GrymNodeOne lnd[916]: 2018-10-19 18:05:51.736 [INF] LTND: Active chain: Bitcoin (network=mainnet)
Oct 19 18:05:51 GrymNodeOne lnd[916]: 2018-10-19 18:05:51.736 [INF] CHDB: Checking for schema update: latest_version=0, db_version=0
Oct 19 18:05:52 GrymNodeOne lnd[916]: 2018-10-19 18:05:52.012 [INF] RPCS: password RPC server listening on 127.0.0.1:10009
Oct 19 18:05:52 GrymNodeOne lnd[916]: 2018-10-19 18:05:52.012 [INF] RPCS: password gRPC proxy started at 127.0.0.1:8080
Oct 19 18:05:52 GrymNodeOne lnd[916]: 2018-10-19 18:05:52.012 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create wallet, or `lncli unlock` to unlock already created wallet.

TheCHGoon, 2018-10-22:
i get the same error.
my output of ls -la /home/bitcoin/.lnd/

drwxrwxr-x 4 bitcoin bitcoin 4096 Oct 22 16:38 .
drwxr-xr-x 5 bitcoin bitcoin 4096 Oct  9 16:12 ..
drwx------ 4 bitcoin bitcoin 4096 Oct 14 12:46 data
lrwxrwxrwx 1 bitcoin bitcoin   12 Oct 14 11:10 lnd -> /mnt/hdd/lnd
-rw-rw-r-- 1 bitcoin bitcoin  289 Oct 22 16:38 lnd.conf
drwx------ 3 bitcoin bitcoin 4096 Oct 14 12:23 logs
-rw-r--r-- 1 bitcoin bitcoin  733 Oct 14 12:23 tls.cert
-rw------- 1 bitcoin bitcoin  227 Oct 14 12:23 tls.key

also:

admin@Goondroid:~$ su bitcoin
Password:
bitcoin@Goondroid:/home/admin$ systemctl status lnd
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2018-10-22 17:29:17 CEST; 2h 24min ago
 Main PID: 1125 (lnd)
   CGroup: /system.slice/lnd.service
           └─1125 /usr/local/bin/lnd
bitcoin@Goondroid:/home/admin$ lncli unlock
Input wallet password:

lnd successfully unlocked!
bitcoin@Goondroid:/home/admin$ lncli addinvoice --amt 100000
[lncli] rpc error: code = Unavailable desc = grpc: the connection is unavailable
bitcoin@Goondroid:/home/admin$ lncli getinfo
[lncli] Wallet is encrypted. Please unlock using 'lncli unlock', or set password using 'lncli create' if this is the first time starting lnd.
bitcoin@Goondroid:/home/admin$ lncli unlock
Input wallet password:

lnd successfully unlocked!
bitcoin@Goondroid:/home/admin$ lncli addinvoice --amt 100000
[lncli] rpc error: code = Unavailable desc = grpc: the connection is unavailable
bitcoin@Goondroid:/home/admin$

any help would be greatly appreciated

gbwx-net, 2018-11-02:
I have received that issue too, has this issue started since upgrading to lnd 0.5.0-beta? If so it could be:

bitcoin.conf not having separate ports for zmq ie:
    zmqpubrawblock=tcp://127.0.0.1:28332
    zmqpubrawtx=tcp://127.0.0.1:28333

lnd.conf still having debughtlc set ie:
    debughtlc=true 
    Either remove or comment out to read:
    #debughtlc=true

Could be an issue with the admin.macaroon

wesbabler, 2018-11-09:
Has anyone figured out a fix for this? I am still having the issue when going from testnet to mainnet. I had two fully working nodes and now after update I'm without any!

grymjack, 2018-11-10:
Nope, waiting a bit hoping that future release fixes the problem.   :(

Stadicus, 2018-11-18:
I think the use of systemd complicates debugging a lot, sorry for that. Can you please try to start LND manually, as described here?
https://github.com/Stadicus/guides/issues/276#issuecomment-439678798

MrPaz, 2018-11-19:
```
$ lnd
Attempting automatic RPC configuration to bitcoind
Automatically obtained bitcoind's RPC credentials
2018-11-19 09:20:06.359 [INF] LTND: Version 0.5.0-beta commit=3b2c807288b1b7f40d609533c1e96a510ac5fa6d
2018-11-19 09:20:06.359 [INF] LTND: Active chain: Bitcoin (network=mainnet)
2018-11-19 09:20:06.361 [INF] CHDB: Checking for schema update: latest_version=6, db_version=6
2018-11-19 09:20:06.401 [INF] RPCS: password RPC server listening on 127.0.0.1:10009
2018-11-19 09:20:06.401 [INF] RPCS: password gRPC proxy started at 127.0.0.1:8080
2018-11-19 09:20:06.403 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.
2018-11-19 09:20:28.472 [INF] LNWL: Opened wallet
2018-11-19 09:20:29.078 [INF] LTND: Primary chain is set to: bitcoin
unable to create chain control: unable to connect to bitcoind: unable to subscribe for zmq block events: dial tcp 127.0.0.1:28332: connect: connection refused
2018-11-19 09:20:29.294 [INF] LTND: Shutdown complete
unable to connect to bitcoind: unable to subscribe for zmq block events: dial tcp 127.0.0.1:28332: connect: connection refused

```

Stadicus, 2018-11-22:
* Is bitcoind up and running?
* Are the ZMQ ports set correctly to 28332 and 28333?

```
$ sudo systemctl status bitcoind
$ bitcoin-cli getblockchaininfo
$ sudo cat /home/bitcoin/.bitcoin/bitcoin.conf
...
zmqpubrawblock=tcp://127.0.0.1:28332
zmqpubrawtx=tcp://127.0.0.1:28333
...
```

MrPaz, 2018-11-23:
I believe everything is set up correctly... 

```
$ sudo systemctl status bitcoind
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2018-11-19 08:33:27 CST; 4 days ago
  Process: 710 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/pi/bitcoinData/bitcoin.conf
 Main PID: 712 (bitcoind)
   CGroup: /system.slice/bitcoind.service
           └─712 /usr/local/bin/bitcoind -daemon -conf=/home/pi/bitcoinData/bitcoin.conf -pid=/h

Nov 19 08:33:27 piNode systemd[1]: Starting Bitcoin daemon...
Nov 19 08:33:27 piNode systemd[1]: bitcoind.service: PID file /home/pi/bitcoinData/bitcoind.pid 
Nov 19 08:33:27 piNode systemd[1]: Started Bitcoin daemon.

$ bitcoin-cli getblockcount
551210

# RaspiBolt LND Mainnet: bitcoind configuration
# /home/bitcoin/.bitcoin/bitcoin.conf <--example. For my pi it's: /home/pi/.bit$

# remove the following line to enable Bitcoin mainnet
# testnet=1

# Bitcoind options
server=1
daemon=1
txindex=1
disablewallet=0

# Connection settings
rpcuser=myuser
rpcpassword=mypw
zmqpubrawblock=tcp://127.0.0.1:28332
zmqpubrawtx=tcp://127.0.0.1:28333

# Raspberry Pi optimizations
dbcache=100
maxorphantx=10
maxmempool=50
maxconnections=25
maxuploadtarget=5000
```

MrPaz, 2018-12-25:
yes, unfortunately.  lnd seems to still be unable to connect to bitcoind.

```
$ lnd
Attempting automatic RPC configuration to bitcoind
Automatically obtained bitcoind's RPC credentials
2018-12-25 09:24:08.873 [INF] LTND: Version 0.5.0-beta commit=3b2c807288b1b7f40d609533c1e96a510ac5fa6d
2018-12-25 09:24:08.873 [INF] LTND: Active chain: Bitcoin (network=mainnet)
2018-12-25 09:24:08.898 [INF] CHDB: Checking for schema update: latest_version=6, db_version=6
2018-12-25 09:24:08.940 [INF] RPCS: password RPC server listening on 127.0.0.1:10009
2018-12-25 09:24:08.944 [INF] RPCS: password gRPC proxy started at 127.0.0.1:8080
2018-12-25 09:24:08.946 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.
2018-12-25 09:25:12.296 [INF] LNWL: Opened wallet
2018-12-25 09:25:12.767 [INF] LTND: Primary chain is set to: bitcoin
**unable to create chain control: unable to connect to bitcoind: unable to subscribe for zmq block events: dial tcp 127.0.0.1:28332: connect: connection refused**
2018-12-25 09:25:12.931 [INF] LTND: Shutdown complete
**unable to connect to bitcoind: unable to subscribe for zmq block events: dial tcp 127.0.0.1:28332: connect: connection refused**
```

I know there was a new release of lnd a couple of weeks ago. maybe I will install that tomorrow and see if it somehow helps.

Stadicus, 2018-12-26:
Are bitcoind & lnd running with the same user? The fact that the connection has been "refused" would point to failed authentication and the lnd user needs to be able to access the user / pw credentials of bitcoind.

MrPaz, 2018-12-28:
yes they were on same user.  i was trying to combine my original pi node setup from like 5 years ago with your guide to spin up a lightning node and eps, so it was a janky setup to start with. i couldn't figure this out, and had the past couple days off, so decided to go scorched earth, flash the microsd and start over.  lnd is working for me now.  so i don't need anything further regarding this.  thanks

Stadicus, 2018-12-29:
Congrats!

closed 2018-12-29 by Stadicus