raspibolt/raspibolt #270: Mainnet wallet creation and macaroons questions

GrannyCryptomaster, 2018-11-07:
I followed the guide ad literam (as a newbie that I am) on a Raspberry Pi 3 B+, first I worked on testnet and now I moved to mainnet.
My problems:
1.the macaroon files... in the **/home/bitcoin/.lnd/data/chain/bitcoin/mainnet**/ are 5 files: admin.macaroon, invoice.macaroon, readonly.macaroon, macaroon.db and wallet.db. What files should I copy to **/home/admin/...** ? Only the admin.macaroon? 
2.when creating the wallet on mainnet, I kept the same password as on testnet, but for the seed part, should I use the seed from the testnet wallet? Or choose No (I don't have a seed) and generate a new seed?
3.I already created the mainnet wallet from the testnet wallet's seed... If this is wrong, and I must create a new one with fresh new seed, should I delete some files first? Will I have 2 wallets?
What are the exact and correct steps?... because I don't know anything about nodes and wallets...
I just want it up and runnig to see if it's profitable and as a personal curiosity... 
Thanks!

CodingMuziekwijk, 2018-11-09:
Commenting from my phone, so an short answer (for now). 

1. The way macaroon files are stored changed in lnd 0.5. Stadicus has been on holiday for a while and hasn't fully updated the guide yet. You can check some pull requests in the meantime to get it working I think one of them has this thing in it. 

2. You should het a new wallet/seed. It is more secure and will prevent future problems.

3. Don't know how to do that atm. You could lookup the lncli rpc command and check if it's there.

And "to see if it's provitable" I can tell you it won't earn you any significant money. Opening a channel cost about 230sat and you will get maybe 0 of 1 sat for routing a payment. Many people are offering the routing for free so if you ask 1 sat, they probably skip your node most of the time.

It is profitable in a way that is can save you a lot of transaction cost and waiting time. :)  succes on your lightning adventure ^^

wesbabler, 2018-11-13:
I also updated to lnd 0.5 and had a lot of issues. I tried multiple fixes I found on github but it seems to of only made things worse. My mainnet channel balance was very low, do you think my best bet is to fully start over? Or will that reproduce the same issues? (missing macaroons, balance showing up as 0)

waffi121, 2018-11-13:
https://github.com/Stadicus/guides/issues/220

I also had to start over completely. but now it works perfectly, is not a big thing :)

Stadicus, 2018-11-18:
I don't think it's possible to downgrade, as LND migrated all internal data structures to a new format. 

What should work, but it takes time, is to set up LND 0.4 again with your original seed and just wait for other peers to close your channels. The funds will end up in your wallet eventually, but there's no way to force or even contact these peers. From my own experience, I had a VPS running with $2000 incoming liquidity. After being down for about a month, all channels have been closed.

closed 2018-12-23 by Stadicus