raspibolt/raspibolt #275: Bitcoind on startup/reboot

Knopfler23, 2018-11-16:
First, let me say that your guide is amazing, and that I have my node up and running on the test net. I took some work (had to start over three times), but she's purring now. I have only one problem with it. When I try to have the bitcoin daemon start when the system boots, I always get this error:

● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: failed (Result: start-limit-hit) since Thu 2016-02-11 11:28:06 EST; 5min ago

Feb 11 11:28:06 thundroid systemd[1]: bitcoind.service: Unit entered failed state.
Feb 11 11:28:06 thundroid systemd[1]: bitcoind.service: Failed with result 'resources'.
Feb 11 11:28:06 thundroid systemd[1]: bitcoind.service: Service hold-off time over, scheduling r
Feb 11 11:28:06 thundroid systemd[1]: Stopped Bitcoin daemon.
Feb 11 11:28:06 thundroid systemd[1]: bitcoind.service: Start request repeated too quickly.
Feb 11 11:28:06 thundroid systemd[1]: Failed to start Bitcoin daemon.
Feb 11 11:28:06 thundroid systemd[1]: bitcoind.service: Unit entered failed state.
Feb 11 11:28:06 thundroid systemd[1]: bitcoind.service: Failed with result 'start-limit-hit'.

I fix this by running: 

$ sudo pkill -9 -f bitcoind

Then:

$ sudo systemctl enable bitcoind
$ sudo systemctl start bitcoind

However, after $ sudo systemctl start bitcoind, I get the error: 

Job for bitcoind.service failed because a configured resource limit was exceeded. See "systemctl status bitcoind.service" and "journalctl -xe" for details.

At that point, I run: 

$ sudo mount -o rw,remount /

And afterwards, $ sudo systemctl start bitcoind, works like a charm. I have no problem getting LND up once the BTC testnet is running. 

Any thoughts on how I might streamline this process? Again, thank you for all of your hard work putting this together. I look forward to migrating to the mainnet and helping to validate transactions.

Knopfler23, 2018-11-18:
Hi, thank you for your follow-up. I'm using a WD Blue drive formatted NTFS+. 

When I try to read /home/bitcoin/.bitcoin/testnet3/debug.log, I get the error: "permission denied."
It seems that a number of these problems might be coming from the permissions are allocated? 

When you say disable the service unit, do you mean stop running the deamon and restart it manually as I have been doing, but instead of doing so from admin, using the bitcoin user?

Thanks again for your help!

Stadicus, 2018-11-22:
Yes, disable the systemd unit and start LND as posted here:
https://github.com/Stadicus/guides/issues/276#issuecomment-439678798

Do you get the "permission denied" error when accessing the `debug.log` both from users "admin" and "bitcoin"? If so, try to set the owner of the whole hard disk (but not the "swapfile") to user "bitcoin" like this:
``` 
sudo chown -R bitcoin:bitcoin /mnt/hdd
sudo chown root:root /mnt/hdd/swapfile
```

Stadicus, 2018-12-23:
Hej @Knopfler23, were you able to fix this issue?

Knopfler23, 2018-12-28:
Sadly, no. I ran those commands and tried to change permissions for both bitcoin and root users but the daemon still doesnt run automatically (I just had the chance to test it since we lost power an hour ago). I'm unsure where to turn next. As I mentioned before, I can get both the daemon and lightning to run once I shutdown the daemon manually and give R/W permissions for all users to all files. Otherwise, it doesnt work.

Stadicus, 2018-12-28:
Can you try to run through the Troubleshooting section and run bitcoind manually? Please post the relevant output of the `debug.log` here.
https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_70_troubleshooting.md

Stadicus, 2018-12-28:
It seems that your root filesystem - your sd card - is falling back to "read-only" mode, which is a sure sign of inconsistencies (either data corruption or hardware failure).
https://askubuntu.com/questions/197459/how-to-fix-sudo-unable-to-open-read-only-file-system

You can either try to fix it with `fsck` (see link above) or re-flash the Raspbian image (all your Bitcoin / Lightning stuff is still on the external harddisk), either with the same or a new sdcard.

Knopfler23, 2018-12-28:
Since the fsck commands did not work, I used the following to gain R/W privileges then started the troubleshooting guide. 

$ sudo mount -o rw,remount /

I went through the first steps, rebooted, and when I ran the second set of steps, this was the output:

admin@thundroid:~$ lsblk -o UUID,NAME,FSTYPE,SIZE,LABEL,MODEL
UUID                                 NAME        FSTYPE   SIZE LABEL  MODEL
                                     mmcblk1             14.9G        
52AA-6867                            ├─mmcblk1p1 vfat     128M boot   
e139ce78-9841-40fe-8823-96a304a09859 └─mmcblk1p2 ext4    14.7G rootfs 
c7ccc17d-63ab-4b1d-b2f7-8723ef26ee8c sda         ext4   931.5G                  
admin@thundroid:~$ sudo fsck /dev/sda1 -f
[sudo] password for admin: 
fsck from util-linux 2.27.1
e2fsck 1.42.13 (17-May-2015)
fsck.ext2: No such file or directory while trying to open /dev/sda1
Possibly non-existent device?
admin@thundroid:~$

Stadicus, 2018-12-29:
A read-only file system is really serious and must be fixed first, just mounting it again manually won't help in the long run.

Knopfler23, 2018-12-29:
Ok. Then I'll get a new micro SD and try re-flashing the image since that didnt work with the SD card I have. I'll report back. Thanks again for all of your help and always following up.

Knopfler23, 2018-12-31:
So I flashed my new SD and am trying to login, but now get this response:

@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
@    WARNING: REMOTE HOST IDENTIFICATION HAS CHANGED!     @
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
IT IS POSSIBLE THAT SOMEONE IS DOING SOMETHING NASTY!
Someone could be eavesdropping on you right now (man-in-the-middle attack)!
It is also possible that a host key has just been changed.
The fingerprint for the ECDSA key sent by the remote host is
SHA256:WqV/bWTWFPnfl3MK+4eWBPYeTerikLgrnBvcZCTSpkA.
Please contact your system administrator.
Add correct host key in /Users/jamesregan/.ssh/known_hosts to get rid of this message.
Offending ECDSA key in /Users/jamesregan/.ssh/known_hosts:5
ECDSA host key for 10.0.1.201 has changed and you have requested strict checking.
Host key verification failed.
Jamess-MacBook-Air:~ jamesregan$ 

Is there any way to reset this without having to setup the system again?

Knopfler23, 2018-12-31:
So I decided to just start over and now after formatting the SDA volume, I try to add the UUID to the fstab and get the error "error: read-only file system."

Knopfler23, 2019-01-01:
Another update: I've tried to go through the full setup process a number of times now and the file system reverts to read-only when I try to move the swap file. Any thoughts?

Knopfler23, 2019-01-01:
Thanks for your reply. I was fixed the first issue, but as for the second, I have tried two different SD cards, both 64gb. Both as SanDisk though. I just got a Samsung which I will try now. I'll report back soon. Thanks again for all of your help.

Knopfler23, 2019-01-01:
So now trying to go through the process, I get this error when trying to upgrade the system in the second setup step:

root@odroid:~# apt upgrade
E: Could not get lock /var/lib/dpkg/lock - open (11: Resource temporarily unavailable)
E: Unable to lock the administration directory (/var/lib/dpkg/), is another process using it?

Knopfler23, 2019-01-01:
Getting this error again when I try to move the swapfile:

root@odroid:~# sudo dphys-swapfile swapon
sudo: unable to resolve host odroid

Stadicus, 2019-01-01:
Not sure I interpret correctly, just to be sure:
* leave everything in the /etc/fstab as it is (it concerns the root file system on the sdcard)
* add a new line for the hdd

I have the following devices attached:
```
admin@odroid:~$ lsblk -o UUID,NAME,FSTYPE,SIZE,LABEL,MODEL
UUID                                 NAME        FSTYPE   SIZE LABEL  MODEL
0cc5980f-4fc8-4b98-90f6-cc4143dcec46 sda         ext4   931.5G
                                     mmcblk1             29.2G
52AA-6867                            |-mmcblk1p1 vfat     128M boot
e139ce78-9841-40fe-8823-96a304a09859 `-mmcblk1p2 ext4    29.1G rootfs
```

The `sda` is the external hdd. If you have partitions on it, you should see them as `sda1`, `sda2`, etc. and they have their own UUID you can use.

My `/etc/fstab` looks like this:
```
admin@odroid:~$ cat /etc/fstab
UUID=e139ce78-9841-40fe-8823-96a304a09859 / ext4 errors=remount-ro,noatime 0 1
LABEL=boot /media/boot vfat defaults 0 1

UUID=0cc5980f-4fc8-4b98-90f6-cc4143dcec46 /mnt/hdd ext4 noexec,defaults 0 0
```

(BTW: you can use three \` (backticks) to create code blocks)

Knopfler23, 2019-01-01:
That's my problem. I was erasing the other UUID. I'll restart and simply add a line for the new one. It makes sense that that is the root of the trouble.

Knopfler23, 2019-01-01:
When I get to the step of setting my bitcoin directory, I notice that your screenshot has a .nano folder while mine does not. I dont believe I've missed any steps, but now I'm unsure.

Stadicus, 2019-01-02:
When you monitor lnd.log, is it still syncing with bitcoind? That can take a few hours.

Otherwise you should be able to create a new address quite easily. But interesting thought regarding special characters, I usually avoid them (which also helps typing the password on other language keyboards). 

You might just delete your wallet file and try a different, easier password. But no idea if this is related:
`rm /home/bitcoin/.lnd/data/chain/bitcoin/testnet/wallet.db`

Knopfler23, 2019-01-02:
It is done syncing, but now I get this:
-- Logs begin at Tue 2019-01-01 23:43:14 EST. --
Jan 02 12:44:37 thundroid lnd[1047]: 2019-01-02 12:44:37.060 [INF] DISC: Attempting to bootstrap with: Authenticated Channel Graph
Jan 02 12:44:37 thundroid lnd[1047]: 2019-01-02 12:44:37.061 [INF] DISC: Attempting to bootstrap with: BOLT-0010 DNS Seed: [[test.nodes.lightning.directory soa.nodes.lightning.directory]]
Jan 02 12:44:57 thundroid lnd[1047]: 2019-01-02 12:44:57.067 [ERR] DISC: Unable to query bootstrapper BOLT-0010 DNS Seed: [[test.nodes.lightning.directory soa.nodes.lightning.directory]]: lookup soa.nodes.lightning.directory on 127.0.1.1:53: read udp 127.0.0.1:37172->127.0.1.1:53: i/o timeout
Jan 02 12:44:57 thundroid lnd[1047]: 2019-01-02 12:44:57.067 [INF] DISC: Obtained 0 addrs to bootstrap network with
Jan 02 12:44:57 thundroid lnd[1047]: 2019-01-02 12:44:57.067 [INF] DISC: Attempting to bootstrap with: Authenticated Channel Graph
Jan 02 12:44:57 thundroid lnd[1047]: 2019-01-02 12:44:57.068 [INF] DISC: Attempting to bootstrap with: BOLT-0010 DNS Seed: [[test.nodes.lightning.directory soa.nodes.lightning.directory]]
Jan 02 12:45:17 thundroid lnd[1047]: 2019-01-02 12:45:17.074 [ERR] DISC: Unable to query bootstrapper BOLT-0010 DNS Seed: [[test.nodes.lightning.directory soa.nodes.lightning.directory]]: lookup soa.nodes.lightning.directory on 127.0.1.1:53: read udp 127.0.0.1:38225->127.0.1.1:53: i/o timeout
Jan 02 12:45:17 thundroid lnd[1047]: 2019-01-02 12:45:17.074 [INF] DISC: Obtained 0 addrs to bootstrap network with
Jan 02 12:45:17 thundroid lnd[1047]: 2019-01-02 12:45:17.074 [INF] DISC: Attempting to bootstrap with: Authenticated Channel Graph
Jan 02 12:45:17 thundroid lnd[1047]: 2019-01-02 12:45:17.075 [INF] DISC: Attempting to bootstrap with: BOLT-0010 DNS Seed: [[test.nodes.lightning.directory soa.nodes.lightning.directory]]
Jan 02 12:45:37 thundroid lnd[1047]: 2019-01-02 12:45:37.081 [ERR] DISC: Unable to query bootstrapper BOLT-0010 DNS Seed: [[test.nodes.lightning.directory soa.nodes.lightning.directory]]: lookup soa.nodes.lightning.directory on 127.0.1.1:53: read udp 127.0.0.1:38776->127.0.1.1:53: i/o timeout
Jan 02 12:45:37 thundroid lnd[1047]: 2019-01-02 12:45:37.081 [INF] DISC: Obtained 0 addrs to bootstrap network with
Jan 02 12:45:37 thundroid lnd[1047]: 2019-01-02 12:45:37.082 [INF] DISC: Attempting to bootstrap with: Authenticated Channel Graph
Jan 02 12:45:37 thundroid lnd[1047]: 2019-01-02 12:45:37.082 [INF] DISC: Attempting to bootstrap with: BOLT-0010 DNS Seed: [[test.nodes.lightning.directory soa.nodes.lightning.directory]]

I tried deleting the old wallet using that command, but it said my permission was denied. I did so using "sudo" but now get this error when trying to create the wallet:

Generating fresh cipher seed...

[lncli] unable to generate seed: rpc error: code = Unimplemented desc = unknown service lnrpc.WalletUnlocker

Knopfler23, 2019-01-02:
So I restarted LND and was able to create the wallet and cipher. However, after unlocking the wallet, the LND log shows that it enters a failed state:

-- Logs begin at Wed 2019-01-02 12:49:24 EST. --
Jan 02 12:59:53 thundroid lnd[1012]: 2019-01-02 12:59:53.609 [INF] LTND: Version 0.4.1-beta commit=
Jan 02 12:59:53 thundroid lnd[1012]: 2019-01-02 12:59:53.609 [INF] LTND: Active chain: Bitcoin (network=testnet)
Jan 02 12:59:53 thundroid lnd[1012]: 2019-01-02 12:59:53.609 [INF] CHDB: Checking for schema update: latest_version=0, db_version=0
Jan 02 12:59:53 thundroid lnd[1012]: 2019-01-02 12:59:53.641 [INF] RPCS: password RPC server listening on 127.0.0.1:10009
Jan 02 12:59:53 thundroid lnd[1012]: 2019-01-02 12:59:53.641 [INF] RPCS: password gRPC proxy started at 127.0.0.1:8080
Jan 02 12:59:53 thundroid lnd[1012]: 2019-01-02 12:59:53.641 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create wallet, or `lncli unlock` to unlock already created wallet.
Jan 02 13:00:20 thundroid lnd[1012]: 2019-01-02 13:00:20.214 [INF] LNWL: Opened wallet
Jan 02 13:00:20 thundroid systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Jan 02 13:00:20 thundroid systemd[1]: lnd.service: Unit entered failed state.
Jan 02 13:00:20 thundroid systemd[1]: lnd.service: Failed with result 'exit-code'.

If I wait a little while it seems to fix itself. Is this something I should be concerned about?

Jan 02 13:01:20 thundroid systemd[1]: lnd.service: Service hold-off time over, scheduling restart.
Jan 02 13:01:20 thundroid systemd[1]: Stopped LND Lightning Daemon.
Jan 02 13:01:20 thundroid systemd[1]: Started LND Lightning Daemon.
Jan 02 13:01:20 thundroid lnd[1047]: Attempting automatic RPC configuration to bitcoind
Jan 02 13:01:20 thundroid lnd[1047]: Automatically obtained bitcoind's RPC credentials
Jan 02 13:01:20 thundroid lnd[1047]: 2019-01-02 13:01:20.609 [INF] LTND: Version 0.4.1-beta commit=
Jan 02 13:01:20 thundroid lnd[1047]: 2019-01-02 13:01:20.610 [INF] LTND: Active chain: Bitcoin (network=testnet)
Jan 02 13:01:20 thundroid lnd[1047]: 2019-01-02 13:01:20.610 [INF] CHDB: Checking for schema update: latest_version=0, db_version=0
Jan 02 13:01:20 thundroid lnd[1047]: 2019-01-02 13:01:20.646 [INF] RPCS: password RPC server listening on 127.0.0.1:10009
Jan 02 13:01:20 thundroid lnd[1047]: 2019-01-02 13:01:20.647 [INF] RPCS: password gRPC proxy started at 127.0.0.1:8080
Jan 02 13:01:20 thundroid lnd[1047]: 2019-01-02 13:01:20.647 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create wallet, or `lncli unlock` to unlock already created wallet.

Knopfler23, 2019-01-05:
No rush. I just wanted to follow up and see if you have any ideas about how to correct for this error. I am still able to unlock the LDN wallet, but as soon as I do the log shows that LND enters a failed state. This causes the wallet to re-lock, thus preventing me from generating a new address (or doing anything else). Any ideas on the culprit? Thanks again for your ongoing support.

Stadicus, 2019-01-06:
Did you have a look at the detailed log files, as described in
https://github.com/Fillippone/NodoBitcoinforDummies/blob/master/README.md

closed 2019-03-03 by Stadicus