raspibolt/raspibolt #276: LND fails after unlocking wallet

michielvbe, 2018-11-16:
After months of running fine, my LND node was 'Not reachable' anymore, so I restarted my Raspibolt. Unfortunately after successfully unlocking the wallet, it automatically restarts. 'Funny' enough I do get three different errors in the log files.

```
nov 16 17:53:06 raspberrypi lnd[2863]: 2018-11-16 17:53:06.673 [INF] LNWL: Finished rescan for 65 addresses (synced to block 0000000000000000000a0177bf9fc4ad4c92a61f9ab84dbf8fdbfaaa349414e4, height 550321)
nov 16 17:53:06 raspberrypi lnd[2863]: 2018-11-16 17:53:06.697 [INF] HSWC: Restoring in-memory circuit state from disk
nov 16 17:53:06 raspberrypi lnd[2863]: 2018-11-16 17:53:06.741 [INF] HSWC: Payment circuits loaded: num_pending=0, num_open=0
nov 16 17:53:06 raspberrypi lnd[2863]: 2018-11-16 17:53:06.771 [INF] HSWC: Trimming open circuits for chan_id=545608:2459:1, start_htlc_id=0
nov 16 17:53:06 raspberrypi lnd[2863]: 2018-11-16 17:53:06.772 [INF] HSWC: Trimming open circuits for chan_id=548938:115:1, start_htlc_id=0
nov 16 17:53:06 raspberrypi lnd[2863]: 2018-11-16 17:53:06.772 [INF] HSWC: Trimming open circuits for chan_id=548938:116:1, start_htlc_id=0
**nov 16 17:53:06 raspberrypi lnd[2863]: 2018-11-16 17:53:06.886 [ERR] SRVR: unable to create server: can't create router: ReadVarBytes: sig is larger than the max allowed size [count 147, max 80]
nov 16 17:53:06 raspberrypi systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
nov 16 17:53:06 raspberrypi systemd[1]: lnd.service: Unit entered failed state.
nov 16 17:53:06 raspberrypi systemd[1]: lnd.service: Failed with result 'exit-code'.**
nov 16 17:54:07 raspberrypi systemd[1]: lnd.service: Service hold-off time over, scheduling restart.
nov 16 17:54:07 raspberrypi systemd[1]: Stopped LND Lightning Daemon.
```

```
nov 16 17:58:03 raspberrypi lnd[3143]: 2018-11-16 17:58:03.874 [INF] HSWC: Restoring in-memory circuit state from disk
nov 16 17:58:03 raspberrypi lnd[3143]: 2018-11-16 17:58:03.895 [INF] LNWL: Done catching up block hashes
nov 16 17:58:03 raspberrypi lnd[3143]: 2018-11-16 17:58:03.895 [INF] LNWL: Finished rescan for 65 addresses (synced to block 0000000000000000000a0177bf9fc4ad4c92a61f9ab84dbf8fdbfaaa349414e4, height 550321)
nov 16 17:58:03 raspberrypi lnd[3143]: 2018-11-16 17:58:03.930 [INF] HSWC: Payment circuits loaded: num_pending=0, num_open=0
nov 16 17:58:03 raspberrypi lnd[3143]: 2018-11-16 17:58:03.961 [INF] HSWC: Trimming open circuits for chan_id=545608:2459:1, start_htlc_id=0
nov 16 17:58:03 raspberrypi lnd[3143]: 2018-11-16 17:58:03.961 [INF] HSWC: Trimming open circuits for chan_id=548938:115:1, start_htlc_id=0
nov 16 17:58:03 raspberrypi lnd[3143]: 2018-11-16 17:58:03.961 [INF] HSWC: Trimming open circuits for chan_id=548938:116:1, start_htlc_id=0
**nov 16 17:58:04 raspberrypi systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
nov 16 17:58:04 raspberrypi systemd[1]: lnd.service: Unit entered failed state.
nov 16 17:58:04 raspberrypi systemd[1]: lnd.service: Failed with result 'exit-code'.**
nov 16 17:59:04 raspberrypi systemd[1]: lnd.service: Service hold-off time over, scheduling restart.
nov 16 17:59:04 raspberrypi systemd[1]: Stopped LND Lightning Daemon.
```

```
Nov 16 17:59:45 raspberrypi lnd[3689]: 2018-11-16 17:59:45.921 [INF] ATPL: Processing new external signal
nov 16 17:59:46 raspberrypi lnd[3689]: 2018-11-16 17:59:46.051 [INF] DISC: gossipSyncer(02b3877314aff657216ceac039244fe94e7116af10cfffac0232fefa5903c62a58): no more chans to query
nov 16 17:59:46 raspberrypi lnd[3689]: 2018-11-16 17:59:46.051 [INF] DISC: gossipSyncer(02b3877314aff657216ceac039244fe94e7116af10cfffac0232fefa5903c62a58): applying gossipFilter(start=2018-11-16 16:59:46.05160458 +0100 CET m=-3558.295729910)
**nov 16 17:59:46 raspberrypi systemd[1]: lnd.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
nov 16 17:59:46 raspberrypi systemd[1]: lnd.service: Unit entered failed state.
nov 16 17:59:46 raspberrypi systemd[1]: lnd.service: Failed with result 'exit-code'.**
nov 16 18:00:46 raspberrypi systemd[1]: lnd.service: Service hold-off time over, scheduling restart.
```

- lncli version: 0.5 commit=3b2c807288b1b7f40d609533c1e96a510ac5fa6d
- bitcoincli version: Bitcoin Core RPC client version v0.16.0
- running 3 channels
- already checked ports on router: all open.

oitdmser, 2018-11-17:
After reboots my Debian-based lnd daemon needs multiple (5+) restarts by systemd to successfully launch. It usually dies with an error similar to your first log:

`SRVR: unable to create server: can't create router: ReadVarBytes: sig is larger than the max allowed size [count 147, max 80]`

Stadicus, 2018-11-18:
For testing, I'd disable the systemd unit first, reboot and start lnd manually:
```
admin $> sudo systemctl disable lnd
admin $> sudo shutdown -r now
```

You'll need two separate ssh sessions with user "bitcoin". Check if bitcoind is up and running, only then proceed to start lnd and unlock it in the second session.
```
bitcoin $1> bitcoin-cli getblockchaininfo
bitcoin $1> lnd

bitcoin $2> lncli unlock
```
What is the output of lnd in the first session?

100800100, 2018-11-18:
```
2018-11-18 18:57:59.365 [INF] LTND: Version 0.5.0-beta commit=3b2c807288b1b7f40d609533c1e96a510ac5fa6d
2018-11-18 18:57:59.366 [INF] LTND: Active chain: Bitcoin (network=mainnet)
2018-11-18 18:57:59.437 [INF] CHDB: Checking for schema update: latest_version=6, db_version=6
2018-11-18 18:57:59.449 [INF] RPCS: password RPC server listening on 127.0.0.1:10009
2018-11-18 18:57:59.450 [INF] RPCS: password gRPC proxy started at 127.0.0.1:8080
2018-11-18 18:57:59.450 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.
2018-11-18 18:58:42.234 [INF] LNWL: Opened wallet
2018-11-18 18:58:43.127 [INF] LTND: Primary chain is set to: bitcoin
2018-11-18 18:58:44.770 [INF] LTND: Initializing bitcoind backed fee estimator
2018-11-18 18:58:44.771 [INF] LNWL: Started listening for bitcoind block notifications via ZMQ on tcp://127.0.0.1:28332
2018-11-18 18:58:44.771 [INF] LNWL: Started listening for bitcoind transaction notifications via ZMQ on tcp://127.0.0.1:28333
2018-11-18 18:58:55.750 [INF] LNWL: The wallet has been unlocked without a time limit
2018-11-18 18:58:55.750 [INF] LTND: LightningWallet opened
2018-11-18 18:58:55.915 [INF] LNWL: Started rescan from block 0000000000000000001704b0fe599a5db63f4475353630e6f292c8d2a88bf013 (height 550324) for 65 addresses
2018-11-18 18:58:55.921 [INF] LNWL: Starting rescan from block 0000000000000000001704b0fe599a5db63f4475353630e6f292c8d2a88bf013
2018-11-18 18:58:56.387 [INF] HSWC: Restoring in-memory circuit state from disk
2018-11-18 18:58:56.542 [INF] HSWC: Payment circuits loaded: num_pending=0, num_open=0
2018-11-18 18:58:57.074 [INF] HSWC: Trimming open circuits for chan_id=545608:2459:1, start_htlc_id=0
2018-11-18 18:58:57.075 [INF] HSWC: Trimming open circuits for chan_id=548938:115:1, start_htlc_id=0
2018-11-18 18:58:57.075 [INF] HSWC: Trimming open circuits for chan_id=548938:116:1, start_htlc_id=0
2018-11-18 18:58:57.710 [INF] RPCS: gRPC proxy started at 127.0.0.1:8080
2018-11-18 18:58:57.710 [INF] RPCS: RPC server listening on 127.0.0.1:10009
2018-11-18 18:58:57.808 [INF] LTND: Waiting for chain backend to finish sync, start_height=550612
2018-11-18 19:00:21.041 [INF] RPCS: [listchannels] fetched 3 channels from DB
2018-11-18 19:02:16.649 [INF] RPCS: [listchannels] fetched 3 channels from DB
2018-11-18 19:05:23.602 [INF] RPCS: [listchannels] fetched 3 channels from DB
2018-11-18 19:07:53.954 [INF] LNWL: Rescan finished at 550324 (0000000000000000001704b0fe599a5db63f4475353630e6f292c8d2a88bf013)
2018-11-18 19:07:54.084 [INF] LNWL: Catching up block hashes to height 550324, this might take a while
2018-11-18 19:07:54.140 [INF] LNWL: Done catching up block hashes
2018-11-18 19:07:54.145 [INF] LNWL: Finished rescan for 65 addresses (synced to block 0000000000000000001704b0fe599a5db63f4475353630e6f292c8d2a88bf013, height 550324)
2018-11-18 19:07:54.757 [INF] LTND: Chain backend is fully synced (end_height=550613)!
2018-11-18 19:07:55.007 [INF] NTFN: New block epoch subscription
2018-11-18 19:07:55.013 [INF] HSWC: Starting HTLC Switch
2018-11-18 19:07:55.021 [INF] NTFN: New block epoch subscription
2018-11-18 19:07:55.320 [INF] NTFN: New block epoch subscription
2018-11-18 19:07:55.391 [INF] UTXN: Processing outputs from missed blocks. Starting with blockHeight=550324, to current blockHeight=550613
2018-11-18 19:07:55.394 [INF] UTXN: Attempting to graduate height=550325: num_kids=0, num_babies=0
2018-11-18 19:07:55.542 [INF] UTXN: Attempting to graduate height=550326: num_kids=0, num_babies=0
2018-11-18 19:07:55.786 [INF] UTXN: Attempting to graduate height=550327: num_kids=0, num_babies=0
```

Etcetera. And:

```
2018-11-18 19:13:54.748 [INF] CRTR: Block 0000000000000000001dc6856bbe7a51cfd2c4a9ec1cbd9a18cd178b0dd869e7 (height=550612) closed 0 channels
2018-11-18 19:13:55.682 [INF] CRTR: Block 00000000000000000029f0f5e485da9f20715bb20e6921d6f58406311b1eb6ab (height=550613) closed 0 channels
2018-11-18 19:13:55.682 [INF] CRTR: Graph pruning complete: 161 channels were closed since height 550324
2018-11-18 19:13:55.890 [INF] CMGR: Server listening on [::]:9735
2018-11-18 19:13:56.027 [INF] SRVR: Initializing peer network bootstrappers!
2018-11-18 19:13:56.028 [INF] SRVR: Creating DNS peer bootstrapper with seeds: [[nodes.lightning.directory soa.nodes.lightning.directory]]
2018-11-18 19:13:56.033 [INF] ATPL: Instantiating autopilot with cfg: (*main.autoPilotConfig)(0x1896ba0)({
 Active: (bool) true,
 MaxChannels: (int) 5,
 Allocation: (float64) 0.6,
 MinChannelSize: (int64) 20000,
 MaxChannelSize: (int64) 16777215,
 Private: (bool) false,
 MinConfs: (int32) 0
})

2018-11-18 19:13:56.039 [INF] DISC: Attempting to bootstrap with: Authenticated Channel Graph
2018-11-18 19:13:56.060 [INF] DISC: Obtained 3 addrs to bootstrap network with
```

Continuing with:

```
2018-11-18 19:19:20.631 [INF] CRTR: New channel discovered! Link connects 032a65fc24a44ba4334fe69ef9bbc07e299e5be956e4332347649312bebd51a82e and 03fb822818be083e0a954db85257a2911a3d55458b8c1ea4124b157e865a836d12 with ChannelPoint(b517401f217c3fc49b6bb9112958971e6a9119a3fbe0b2e1644f9b6351294124:0): chan_id=605103030315057152, capacity=0.09 BTC
2018-11-18 19:19:20.831 [INF] CRTR: Updated vertex data for node=03a5f83dfbfe9830406cb95ebc7231b12150a0f885c6ff0c59c1adfa9309796b02
[... cut ...]
I checked /etc/systemd/system/lnd.service + /etc/systemd/system/bitcoind.service and saw my configs were different than mentioned in https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_40_lnd.md#autostart-lnd and https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_30_bitcoin.md#autostart-bitcoind. 

I changed it, did a restart, unlocked my wallet and everything works fine!

Will check tomorrow if everything is still up and running. If yes, I will close.

@Stadicus thanks for all your hard work so far. Really appreciate it!

oitdmser, 2018-11-19:
Interesting. The main difference in my lnd.service was that mine had
```
Wants=publicip.service
After=publicip.service
```
instead of 
```
Wants=bitcoind.service
After=bitcoind.service
```
Just for kicks I edited lnd.service to use @Stadicus recommendation of bitcoind.service and rebooted. I don't have lnd auto-unlock, so I just waited until bitcoind was fully loaded and chugging along, then unlocked lnd. On the first try it failed:

```
2018-11-19 10:39:09.972 [INF] HSWC: Restoring in-memory circuit state from disk
2018-11-19 10:39:18.165 [INF] LNWL: Started rescan from block 00000000000000000017be79415b149ad501d3ead082ce84965c909550b8f560 (height 550734) for 234 addresses
2018-11-19 10:39:18.186 [INF] LNWL: Starting rescan from block 00000000000000000017be79415b149ad501d3ead082ce84965c909550b8f560
2018-11-19 10:39:18.752 [INF] LNWL: Rescan finished at 550734 (00000000000000000017be79415b149ad501d3ead082ce84965c909550b8f560)
2018-11-19 10:39:18.752 [INF] LNWL: Catching up block hashes to height 550734, this might take a while
2018-11-19 10:39:19.131 [INF] LNWL: Done catching up block hashes
2018-11-19 10:39:19.131 [INF] LNWL: Finished rescan for 234 addresses (synced to block 00000000000000000017be79415b149ad501d3ead082ce84965c909550b8f560, height 550734)
2018-11-19 10:39:19.503 [INF] HSWC: Payment circuits loaded: num_pending=206, num_open=0
2018-11-19 10:39:54.744 [INF] HSWC: Trimming open circuits for chan_id=521699:2306:1, start_htlc_id=48
2018-11-19 10:39:54.745 [INF] HSWC: Trimming open circuits for chan_id=521526:1693:1, start_htlc_id=38
2018-11-19 10:39:54.745 [INF] HSWC: Trimming open circuits for chan_id=550439:1344:0, start_htlc_id=0
2018-11-19 10:39:54.745 [INF] HSWC: Trimming open circuits for chan_id=532286:1974:1, start_htlc_id=0
2018-11-19 10:39:54.745 [INF] HSWC: Trimming open circuits for chan_id=533150:977:0, start_htlc_id=0
2018-11-19 10:39:54.745 [INF] HSWC: Trimming open circuits for chan_id=521769:1832:0, start_htlc_id=0
2018-11-19 10:39:54.745 [INF] HSWC: Trimming open circuits for chan_id=549258:2518:0, start_htlc_id=0
2018-11-19 10:39:54.745 [INF] HSWC: Trimming open circuits for chan_id=532207:840:0, start_htlc_id=0
2018-11-19 10:39:54.745 [INF] HSWC: Trimming open circuits for chan_id=530217:873:1, start_htlc_id=0
2018-11-19 10:39:54.745 [INF] HSWC: Trimming open circuits for chan_id=540001:1357:0, start_htlc_id=0
2018-11-19 10:39:54.745 [INF] HSWC: Trimming open circuits for chan_id=549572:1871:1, start_htlc_id=0
2018-11-19 10:39:54.745 [INF] HSWC: Trimming open circuits for chan_id=521489:1594:0, start_htlc_id=0
2018-11-19 10:39:54.745 [INF] HSWC: Trimming open circuits for chan_id=532733:1293:0, start_htlc_id=0
2018-11-19 10:39:54.746 [INF] HSWC: Trimming open circuits for chan_id=538437:2232:0, start_htlc_id=0
2018-11-19 10:39:56.505 [ERR] SRVR: unable to create server: can't create router: ReadVarBytes: sig is larger than the max allowed size [count 119, max 80]

2018-11-19 10:39:56.505 [INF] LTND: Shutdown complete

```

When I unlocked the second time it starts successfully.

oitdmser, 2018-11-19:
Anecdotally it seems that if I wait several minutes until bitcoind has 20+ peers I can start lnd without errors. Would we have more success if systemd waited longer to launch lnd?

Also, when I disable lnd.service and reboot, wait until bitcoind has 20+ peers, then run lnd at the command line under the lnd user it starts OK, but takes a VERY long time to terminate. Perhaps systemd isn't waiting long enough for lnd to close gracefully during shutdown?

Stadicus, 2018-11-22:
I retired the publicip script, as LND is now able to use UPnP to detect and broadcast it's public ip address. Should not make a big difference regarding your original question, however.

It may well be that LND starts, bitcoind is not ready yet, and LND shuts down. The service unit shold automatically restart, however, until LND keeps running. 

The reason that LND shuts down in your case seems to be this:
```
2018-11-19 10:39:56.505 [ERR] SRVR: unable to create server: can't create router: ReadVarBytes: sig is larger than the max allowed size [count 119, max 80]
```
Not sure I can help with that. Can you maybe post this issue in the LND github repo and keep us up-to-date here?

oitdmser, 2018-11-23:
Will do

Stadicus, 2018-12-23:
Hej @oitdmser, is your node up and running again, or still strugling?

closed 2019-01-08 by Stadicus