raspibolt/raspibolt #282: LND Mainnet

Freefromchainz, 2018-11-20:
After transfering the block and chainstate folder on WinSCP im running into this issue.
[image]
[image]

 
sudo tail -f /home/bitcoin/.bitcoin/debug.log
2018-11-20T16:42:20Z Opening LevelDB in /home/bitcoin/.bitcoin/blocks/index
2018-11-20T16:42:20Z Fatal LevelDB error: Corruption: 479 missing files; e.g.: /home/bitcoin/      .bitcoin/blocks/index/133569.ldb
2018-11-20T16:42:20Z You can use -debug=leveldb to get more complete diagnostic messages
2018-11-20T16:42:20Z Fatal LevelDB error: Corruption: 479 missing files; e.g.: /home/bitcoin/      .bitcoin/blocks/index/133569.ldb
2018-11-20T16:42:20Z : Error opening block database.
Please restart with -reindex or -reindex-chainstate to recover.
2018-11-20T16:42:20Z Aborted block database rebuild. Exiting.
2018-11-20T16:42:20Z Shutdown: In progress...
2018-11-20T16:42:20Z scheduler thread interrupt
2018-11-20T16:42:20Z Shutdown: done
2018-11-20T16:43:20Z

2018-11-20T16:43:20Z Bitcoin Core version v0.17.0.1 (release build)
2018-11-20T16:43:20Z InitParameterInteraction: parameter interaction: -whitelistforcerelay=1       -> setting -whitelistrelay=1
2018-11-20T16:43:20Z Assuming ancestors of block 0000000000000000002e63058c023a9a1de233554f28      c7b21380b6c9003f36a8 have valid signatures.
2018-11-20T16:43:20Z Setting nMinimumChainWork=0000000000000000000000000000000000000000028822      fef1c230963535a90d
2018-11-20T16:43:20Z Using the 'standard' SHA256 implementation
2018-11-20T16:43:20Z Default data directory /home/bitcoin/.bitcoin
2018-11-20T16:43:20Z Using data directory /home/bitcoin/.bitcoin
2018-11-20T16:43:20Z Using config file /home/bitcoin/.bitcoin/bitcoin.conf
2018-11-20T16:43:20Z Using at most 40 automatic connections (1024 file descriptors available)
2018-11-20T16:43:20Z Using 16 MiB out of 32/2 requested for signature cache, able to store 52      4288 elements
2018-11-20T16:43:20Z Using 16 MiB out of 32/2 requested for script execution cache, able to s      tore 524288 elements
2018-11-20T16:43:20Z Using 4 threads for script verification
2018-11-20T16:43:21Z scheduler thread start
2018-11-20T16:43:21Z HTTP: creating work queue of depth 16
2018-11-20T16:43:21Z Config options rpcuser and rpcpassword will soon be deprecated. Locally-      run instances may remove rpcuser to use cookie-based auth, or may be replaced with rpcauth. P      lease see share/rpcauth for rpcauth auth generation.
2018-11-20T16:43:21Z HTTP: starting 4 worker threads
2018-11-20T16:43:21Z Using wallet directory /home/bitcoin/.bitcoin
2018-11-20T16:43:21Z init message: Verifying wallet(s)...
2018-11-20T16:43:21Z Using BerkeleyDB version Berkeley DB 4.8.30: (April  9, 2010)
2018-11-20T16:43:21Z Using wallet wallet.dat
2018-11-20T16:43:21Z BerkeleyEnvironment::Open: LogDir=/home/bitcoin/.bitcoin/database ErrorF      ile=/home/bitcoin/.bitcoin/db.log
2018-11-20T16:43:21Z Cache configuration:
2018-11-20T16:43:21Z * Using 2.0MiB for block index database
2018-11-20T16:43:21Z * Using 12.2MiB for transaction index database
2018-11-20T16:43:21Z * Using 8.0MiB for chain state database
2018-11-20T16:43:21Z * Using 77.8MiB for in-memory UTXO set (plus up to 47.7MiB of unused mem      pool space)
2018-11-20T16:43:21Z init message: Loading block index...
2018-11-20T16:43:21Z Opening LevelDB in /home/bitcoin/.bitcoin/blocks/index
2018-11-20T16:43:21Z Fatal LevelDB error: Corruption: 479 missing files; e.g.: /home/bitcoin/      .bitcoin/blocks/index/133569.ldb
2018-11-20T16:43:21Z You can use -debug=leveldb to get more complete diagnostic messages
2018-11-20T16:43:21Z Fatal LevelDB error: Corruption: 479 missing files; e.g.: /home/bitcoin/      .bitcoin/blocks/index/133569.ldb
2018-11-20T16:43:21Z : Error opening block database.
Please restart with -reindex or -reindex-chainstate to recover.
2018-11-20T16:43:21Z Aborted block database rebuild. Exiting.
2018-11-20T16:43:21Z Shutdown: In progress...
2018-11-20T16:43:21Z scheduler thread interrupt
2018-11-20T16:43:21Z Shutdown: done
2018-11-20T16:44:21Z

Freefromchainz, 2018-11-20:
The bitcoind on the pi is running a level db debug log rite now
Also with -reindex -chainstate
The bitcoind on the pi keeps resetting I made it up too 480000 then went back to 220000
and still cant recover the 400 missing files
[image]

Stadicus, 2018-11-22:
I wouldn't reindex the chain directly on the Pi, that can take months. 

The most promising way forward is probably to resync your Bitcoin Core program on your regular computer with `txindex=1` from scratch again, making sure that all data files are present, and moving them via WinSCP to the Pi. 

I never ran into the issue that Bitcoin Core is missing files, but is still running. Do you have an idea how they disappeared on your computer?

Freefromchainz, 2018-11-22:
Thanks for the reply, I did few thing that could of lead to the missing files! First I downloaded bitcoin core to an external hard drive so I could utilize Option 2: 

Use existing hard disk with NTFS
If you want to use your existing hard disk that already contains the bitcoin mainnet blockchain, you
can simply mount it as is:
Identify the partition and note the UUID at the left (eg. “12345678”) and verify the FSTYPE (should be
“ntfs”)
$ sudo lsblk -o UUID,NAME,FSTYPE,SIZE,LABEL,MODEL
$ sudo apt-get install ntfs-3g
Open the file “/etc/fstab” in the Nano text editor and add the following line, but use the “UUID” noted
above, save and exit
$ sudo nano /etc/fstab
UUID=12345678 /mnt/hdd ntfs defaults,auto,umask=002,gid=bitcoin,users,rw 0 0
Please note that we mounted using umask=002,gid=bitcoin, which gives only the user “bitcoin” write
access. User “admin” can only read and must use sudo when writing to the disk.  

However this did not workout, then I checked your updated instructions and realized you removed that option. Probably because it wasn't working properly. So I transferred bitcoin core to my desktop and used the mounting option 1: which seems to be only option now! After everything was transferred, I formatted & mounted the external. Then I used winscp to transfer.

Freefromchainz, 2018-11-22:
Got this message with a new bitcoin core install never seen it before, hopefully this does the trick! [image]

Stadicus, 2018-11-23:
In theory, the NTFS option works, but it lead to so many issues that I removed it.

If you have a running Bitcoin Core version (with `txindex=1`) on your regular computer, you should be able to copy the blockchain data to the external Pi hdd with WinSCP just fine. Let me know how it works out.

Stadicus, 2018-12-23:
Hej @Freefromchainz, were you able to fix this issue?

Freefromchainz, 2018-12-23:
Hey there for mean time we are working with https://github.com/PierreRochard/node-launcher/releases/ but since this is a desktop solution which isn't energy efficient we are going to revisit the raspibolt in January! you can close this issue! take care happy holidays!

Stadicus, 2018-12-23:
Thanks, same to you! 👨‍🚀👍

closed 2018-12-23 by Stadicus