raspibolt/raspibolt #284: Translation in Italian of your guide

Fillippone, 2018-11-22:
Hello Stadicus,
I am trying to build a guide on setting up a Bitcoin node in Italian. I started translating your guide, with some degree of flexibility (added some paragraph on which components to buy, how to assemble then into a node, why you should run a node etc. I am instead translating your guide more literally in the technical part). In the preface i clearly stated that my guide is heavily based on your guide and provided a link to your repo.
I am just wondering if it is in any case relevant for you, if you have anything to object or you are ok with this. 
Thanks,
f

Stadicus, 2018-11-22:
Hej @Fillippone, that's great to hear! Go ahead, post the link here and feel free to also contribute to this guide directly if you find parts missing or not clear enough.

Fillippone, 2019-01-05:
First version is online at https://github.com/Fillippone/NodoBitcoinforDummies

I don’t know if you speak Italian, but I added an intro on why you should run a Bitcoin node and expanded somewhat how to actually assemble a node, something that gave me some headache while trying  powering the SSD directly from the Raspberry PI without using external power source.

Stadicus, 2019-01-08:
Thanks for the link, appreciate it!

closed 2019-01-08 by Stadicus