raspibolt/raspibolt #286: EXCEPTION: 15dbwrapper_error

Jeans5, 2018-11-23:
Hi,

I'm running Bitcoind 0.17 on a raspberry Pi 3. Everything ran smoothly until I did something wrong (I guess) but can't figure out what happened. Has someone experience with this one:

(I was trying to run LND on mainnet)

```
018-11-23T13:11:47Z Fatal LevelDB error: IO error: /home/bitcoin/.bitcoin/indexes/txindex/105794.ldb: Bad message
2018-11-23T13:11:47Z You can use -debug=leveldb to get more complete diagnostic messages
2018-11-23T13:11:47Z 

************************
EXCEPTION: 15dbwrapper_error       
Fatal LevelDB error: IO error: /home/bitcoin/.bitcoin/indexes/txindex/105794.ldb: Bad message       
bitcoin in txindex       

2018-11-23T13:12:48Z
```

Thanks in advance for helping me out.

Jeans5, 2018-11-23:
I think I did something wrong by doing this:

> admin@P2P:~ $ sudo chown -R bitcoin:bitcoin /mnt/hdd/bitcoin/
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/105827.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/102507.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/109792.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/107358.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/107264.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/099650.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/112180.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/099681.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/110119.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/099641.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/099401.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/094281.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/105544.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/111576.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/109465.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/106455.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/107900.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/110553.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/111553.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/086626.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/110118.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/106763.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/110196.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/107022.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/107582.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/107934.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/107547.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/110775.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/093165.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/099404.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/109420.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/105574.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/101027.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/104563.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/105547.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/107966.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/107280.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/107048.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/111834.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/107532.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/112252.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/111143.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/109314.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/111835.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/109631.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/109404.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/107531.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/107546.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/092085.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/107594.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/105831.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/111818.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/110490.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/111281.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/106078.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/105794.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/109708.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/093885.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/111546.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/086625.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/093164.ldb': Bad message
[... cut ...]
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/109678.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/092065.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/092077.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/111293.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/110489.ldb': Bad message
chown: changing ownership of '/mnt/hdd/bitcoin/indexes/txindex/106719.ldb': Bad message

Stadicus, 2018-11-24:
Ok, then I don't think it's a permission issue, everything looks good.

I'd suggest checking the file system. For this you need to unmount the hdd, and as we also have the swapfile on the hard disk, a restart without bitcoind / lnd autostart is required.

Disable the services and add a `#` in front of your mount entry in `/etc/fstab`, save & exit and reboot.
```
$ sudo systemctl disable bitcoind
$ sudo systemctl disable lnd
$ sudo nano /etc/fstab
#UUID=7b1b0117-11df-409d-b684-d0f54c78b792 /mnt/hdd ext4 noexec,defaults 0 0
$ sudo shutdown -r now
```

Now, let's check if the hdd is really not mounted any more (no entry for `/mnt/hdd`), get the NAME of the device (eg. `sda1`) and check the file system.
```
$ df -h
$ lsblk -o UUID,NAME,FSTYPE,SIZE,LABEL,MODEL
$ sudo fsck /dev/sda1 -f
```
**Any errors?**

To get back up and running, reverse the first steps:
```
$ sudo systemctl enable bitcoind
$ sudo systemctl enable lnd
$ sudo nano /etc/fstab
UUID=7b1b0117-11df-409d-b684-d0f54c78b792 /mnt/hdd ext4 noexec,defaults 0 0
$ sudo shutdown -r now
```

Stadicus, 2018-12-23:
@Jeans5 , did you manage to fix the issue?

closed 2019-01-08 by Stadicus

ramses2506, 2019-03-17:
I just experienced the same error: 

`EXCEPTION: 15dbwrapper_error       
Fatal LevelDB error: Corruption: block checksum mismatch: /home/bitcoin/.bitcoin/indexes/txindex/001290.ldb       
bitcoin in scheduler `

After after running your test @Stadicus you've mentioned above regarding HD, I didn't receive any errors. Bitcoind is trying to restart after error, but the appears again and again. Do you have a suggestion how I could try to solve this issue? Thank you