raspibolt/raspibolt #288: lnd mainnet  autopilot allocation

sako0938, 2018-11-28:
Hi, I have a small request, or would like some clarification. In Ind.conf there is the line:

```autopilot.allocation=0.6```

and I feel like this line deserves a comment to explain what it does, especially for using this on mainnet. Once switching over to mainnet and funding the lnd wallet, an on-chain transaction is done again to open the channel. However I didnt put enough money in my wallet where 60% of my total covers the transaction. So now I need to close the channel, wait for another block period, and restart  lnd with autopilot.allocation=1, or maybe create another channel manually with my full balance.

Either way since its either testnet and it doesnt matter as much, or mainnet and someone will specifically desire this feature of only a portion of their hot-wallet in a channel.

Stadicus, 2018-12-03:
Good point, will take a look and explain better / give some options.

Stadicus, 2018-12-28:
BTW: I extended the explanation here:
https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_40_lnd.md#lnd-configuration

But I am also thinking about disabling the `autopilot` feature in the standard config.

sako0938, 2018-12-28:
I think that wouldn't be a bad idea. The autopilot automatically opening channels with funds on chain isn't quite there yet from my testing. Maybe once AMP is live it will be better since you won't need a single channel that has the full liquidity for a transaction.