raspibolt/raspibolt #292: REST API connection for Joule

elkimek, 2018-11-29:
Hi,

I'm trying to connect to Joule - https://github.com/wbobeirne/joule-extension

I'm getting error when trying to connect to my Raspibolt via:

``` restlisten=0.0.0.0:8080 ```

error:

``` {"error":"connection error: desc = \"transport: authentication handshake failed: x509: certificate is valid for 127.0.0.1, ::1, 10.0.1.9, fe80::db34:afcd:c2e2:3326, not 0.0.0.0\"","code":13} ```

10.0.1.9 is my Raspibolt's internal IP.

lnd.conf:
```
# RaspiBolt LND Mainnet: lnd configuration
# /home/bitcoin/.lnd/lnd.conf

[Application Options]
debuglevel=info
# debughtlc=true
maxpendingchannels=5
alias=Elkim
color=#68F442
rpclisten=0.0.0.0:10009
restlisten=0.0.0.0:8080
listen=0.0.0.0:9735

[Bitcoin]
bitcoin.active=1

# enable either testnet or mainnet
#bitcoin.testnet=1
bitcoin.mainnet=1

bitcoin.node=bitcoind

[autopilot]
autopilot.active=1
autopilot.maxchannels=5
autopilot.allocation=0.6
```

I even tried:

```restlisten=10.0.1.9:8080```

Any ideas? Thanks

Stadicus, 2018-12-03:
I'll try that out myself in the coming days. The standard LND certificates are only valid for local access (127.0.0.1), not when accepting external requests. You need to create a new TLS certificat for that (similar to the one I described in the "Shango" bonus guide, but no guarantees).

elkimek, 2018-12-28:
Hey @Stadicus, any update on this?

dennisreimann, 2018-12-28:
I got it working by setting `extraip` to `0.0.0.0`, otherwise it failed for me too.

elkimek, 2018-12-28:
Just ```extraip=0.0.0.0``` not ````tlsextraip=0.0.0.0``` or ```restlisten``` ?

Stadicus, 2018-12-29:
I set it up, but I don't think Joule is ready for a Beginner's guide. It's only working reliably in Chrome, and I don't use that. Multiple problems in Firefox and Opera. 

To enable external access to the REST API, 
* add the following to `lnd.conf`
  ```
  restlisten=0.0.0.0:8080
  tlsextraip=0.0.0.0
  ```
* delete the old TLS certificates  
  `$ sudo rm /home/bitcoin/.lnd/tls.*`
* restart LND
* copy the TLS certificates to the admin user
  `$ sudo cp /home/bitcoin/.lnd/tls.cert /home/admin/.lnd`

More info here (in the "reckless" part):
https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_68_shango.md

dennisreimann, 2018-12-29:
Citing the Joule setup in FF:

> Firefox has an issue where addons cannot bypass cross-origin request prevention. If you are running lnd or the Lightning App, you will be unable to connect to your node.
> You will either have to setup a proxy server for your node that has CORS headers configured, or you will have to use another supported browser such as Chrome or Opera.

So a local proxy is the way to go atm, which feels like quite a hassle – and as far as I understand [this issue](https://github.com/wbobeirne/joule-extension/issues/99) it does not even work with the proxy atm.

Hopefully they'll [switch to the gRPC API](https://github.com/wbobeirne/joule-extension/issues/8) someday, as this woul eliminate the need to open up the REST API to the public. (one less thing to configure)

dennisreimann, 2018-12-29:
> I set it up, but I don't think Joule is ready for a Beginner's guide

Those who are interested can also have a look at the [step by step Joule setup guide](https://medium.com/lightning-power-users/bitcoin-lightning-joule-chrome-extension-ac149bb05cb9) by Pierre Rochard.

closed 2019-01-08 by Stadicus