raspibolt/raspibolt #294: How to fund channel open by counterpart

AndAntAnd, 2018-12-01:
Could you pls advise how can I fund channel opened to my node by somebody else?
I tried to use lncli openchannel … but it open new one with the same node

Stadicus, 2018-12-03:
Once a channel is open, you cannot change it's basic funding at the moment. Stuff like that is comming, google "splicing in / out" or "dual-party funding", but not quite yet.

But for your case there's no micromanagement of channels needed. Just open a few with other nodes, and let others connect to you, and everything should work fine.

Any special considerations I'm missing?

AndAntAnd, 2018-12-03:
I just noticed that some nodes I initiated channels with provided funding
from there side (I see both local and remote balance in those channels)

So I wanted to do fund channels initiated by other nodes to mine to have
balance in local.

After few months still there is no transaction in my node so I thought
maybe shall be both local and remote balances in one channel

Stadicus, 2018-12-04:
When other nodes open a channel with your node, all the funds are on their side. If you route payments, that balance can shift to your side (but the same balance would route outwards on a channel with a balance on your side). Once a channel is opened, there's no way to adjust the overall balance or additionally fund it.

closed 2018-12-04 by Stadicus