raspibolt/raspibolt #296: Basic Configuration

Kartereo, 2018-12-02:
Unable to located package linux-image-xu3.

Kartereo, 2018-12-02:
also

root@thundroid:~# sudo rm /var/swap
rm: cannot remove '/var/swap': Operation not permitted

Kartereo, 2018-12-02:
and I just got to this

MOVING THE SWAP FILE:
admin@thundroid:~$ sudo rm /var/swap

We trust you have received the usual lecture from the local System
Administrator. It usually boils down to these three things:

    #1) Respect the privacy of others.
    #2) Think before you type.
    #3) With great power comes great responsibility.

[sudo] password for admin:
rm: cannot remove '/var/swap': Operation not permitted

Kartereo, 2018-12-02:
and another one

SSH KEYS:

$ sudo mkdir /root/.ssh  
mkdir: cannot create directory ‘/root/.ssh’: File exists
admin@thundroid:~$ sudo cp/home/admin/.ssh/authorized_keys /root/.ssh/
sudo: cp/home/admin/.ssh/authorized_keys: command not found

Kartereo, 2018-12-02:
then I just got this

Bitcoin:

# manually check the fingerprint of the public key
admin@thundroid:~/download$ gpg --with-fingerprint ./laanwj-releases.asc
gpg: WARNING: no command supplied.  Trying to guess what you mean ...
pub   rsa4096 2015-06-24 [SC] [expires: 2019-02-14]
uid           Wladimir J. van der Laan (Bitcoin Core binary release signing key) <laanwj@gmail.com>

Kartereo, 2018-12-02:
and then

Bitcoin:

admin@thundroid:~/download$ sudo install -m 0755 -o root -g root -t /usr/local/bin bitcoin-0.17.0.1/bin/*
install: cannot stat 'bitcoin-0.17.0.1/bin/*': No such file or directory

Stadicus, 2018-12-02:
Are you by any chance using Ubuntu 18.04 instead of 16.04?

The command `sudo apt install linux-image-xu3` is needed for 16.04:
https://wiki.odroid.com/odroid-xu4/os_images/linux/ubuntu_4.14/20171213

...but not for 18.04:
https://wiki.odroid.com/odroid-xu4/os_images/linux/ubuntu_4.14/ubuntu_4.14

Stadicus, 2018-12-23:
Hej @Kartereo, any news?

Kartereo, 2018-12-23:
@Stadicus Yes, I was using 18.04; I will just wait for an updated version.

Kartereo, 2018-12-25:
I have just gone through and followed everything to a T. I got the following

admin@thundroid:~$ sudo rm /var/swap
[sudo] password for admin:
rm: cannot remove '/var/swap': Operation not permitted

Kartereo, 2018-12-25:
I also just got and at this point its as far as I can go

admin@thundroid:~/download$ sudo install -m 0755 -o root -g root -t /usr/local/bin bitcoin-0.17.0/bin/*
admin@thundroid:~/download$ bitcoind --version
-bash: /usr/local/bin/bitcoind: cannot execute binary file: Exec format error

Kartereo, 2018-12-25:
I finally got down to syncing and it is now showing me this.

 },
  "warnings": "Warning: unknown new rules activated (versionbit 28)"
}
admin@thundroid:~$ sudo tail -f /home/bitcoin/.bitcoin/testnet3/debug.log
2018-12-25T03:33:22Z UpdateTip: new best=000000000000003c263504a5ad89a5bee78909ae1eec92ef284f01568780bd28 height=1449411 version=0x20000000 log2_work=71.806647 tx=48426766 date='2018-12-25T02:48:55Z' progress=0.999966 cache=49.7MiB(440805txo) warning='31 of last 100 blocks have unexpected version'
2018-12-25T03:33:22Z UpdateTip: new best=00000000000000a7fecb8f1b60ef201099280b25af94e03bb088554c5542f1fc height=1449412 version=0x3fffe000 log2_work=71.806666 tx=48426780 date='2018-12-25T02:51:09Z' progress=0.999967 cache=49.7MiB(440814txo) warning='31 of last 100 blocks have unexpected version'
2018-12-25T03:33:22Z UpdateTip: new best=00000000000001250be6df04f8f010c62e5ca9c5960d725daba3e0596a8e9caa height=1449413 version=0x20000000 log2_work=71.806685 tx=48426823 date='2018-12-25T03:00:15Z' progress=0.999974 cache=49.7MiB(440851txo) warning='31 of last 100 blocks have unexpected version'
2018-12-25T03:33:22Z UpdateTip: new best=0000000000000102ec6ced30c1e17bef900af2bdb9d54d37bcaa8103026b26f7 height=1449414 version=0x20000000 log2_work=71.806704 tx=48426832 date='2018-12-25T03:01:46Z' progress=0.999975 cache=49.7MiB(440858txo) warning='30 of last 100 blocks have unexpected version'
2018-12-25T03:33:22Z UpdateTip: new best=00000000000000c968cae89af52b775a0038c278f80905fe48c24956ade23446 height=1449415 version=0x20000000 log2_work=71.806723 tx=48426838 date='2018-12-25T03:02:42Z' progress=0.999976 cache=49.7MiB(440859txo) warning='29 of last 100 blocks have unexpected version'
2018-12-25T03:33:22Z UpdateTip: new best=0000000000000035bafab69775772d45ea8fc62fe7e0fc7c83cb3a96223f0a89 height=1449416 version=0x20000000 log2_work=71.806742 tx=48426863 date='2018-12-25T03:06:57Z' progress=0.999980 cache=49.7MiB(440875txo) warning='29 of last 100 blocks have unexpected version'
2018-12-25T03:33:22Z UpdateTip: new best=00000000000001120bc4d335355035ff40650c3dbb73942583363933672aa293 height=1449417 version=0x20000000 log2_work=71.806761 tx=48426880 date='2018-12-25T03:10:16Z' progress=0.999982 cache=49.7MiB(440797txo) warning='29 of last 100 blocks have unexpected version'
2018-12-25T03:33:22Z UpdateTip: new best=00000000000000997f4a188d0a200fb350bb3e9cae4ea208cea49379bebb82a7 height=1449418 version=0x20000000 log2_work=71.80678 tx=48426928 date='2018-12-25T03:18:05Z' progress=0.999988 cache=49.7MiB(440861txo) warning='29 of last 100 blocks have unexpected version'
2018-12-25T03:33:22Z UpdateTip: new best=00000000000000fd89e3b0130982eeeadf9fd008881797158f1fc31b417dcc2b height=1449419 version=0x20000000 log2_work=71.806799 tx=48426975 date='2018-12-25T03:26:41Z' progress=0.999995 cache=49.7MiB(440897txo) warning='29 of last 100 blocks have unexpected version'
2018-12-25T03:38:25Z UpdateTip: new best=00000000000000f47e7698914aaa9e41274b6fe8944237f90e4a7b49d5285bed height=1449420 version=0x3fffe000 log2_work=71.806818 tx=4842703

Stadicus, 2018-12-26:
Removing the swap file is pretty optional, I wouldn't worry too much about that.

And the second output is the expected standard output. The Warning is not relevant, see here:
https://bitcoin.stackexchange.com/questions/46318/bitcond-v0-12-1-updatetip-6-of-last-100-blocks-have-unexpected-version

Seems you have pretty much synced up to the chain! 👍

Kartereo, 2018-12-26:
Ok nice, thank you!

closed 2018-12-29 by Stadicus