raspibolt/raspibolt #298: Tor Network no Got service ID

waffi121, 2018-12-06:
I just installed Tor. in the log I have:

```
2018-12-06T17: 50: 25Z InitParameterInteraction: parameter interaction: -proxy set -> setting -discover = 0
2018-12-06T17: 50: 25Z InitParameterInteraction: parameter interaction: -whitelistforcerelay = 1 -> setting -whitelistrelay = 1

2018-12-06T17: 54: 31Z torcontrol thread start
```

unfortunately he does not show me my onion address: tor: Got service ID

what I still see in the log:

`2018-12-06T17: 50: 25Z Config options rpcuser and rpcpassword will be deprecated soon. Locally-run instances may remove rpcuser to use cookie-based auth, or may be replaced with rpcauth. Please see share / rpcauth for rpcauth auth generation.
`

my bitcoin conf:

 Bitcoind options
server=1
daemon=1
txindex=1

Connection settings
rpcuser=raspibolt
rpcpassword=asf7asdf7asd7f7asdf

onlynet=ipv4
zmqpubrawblock=tcp://127.0.0.1:28332
zmqpubrawtx=tcp://127.0.0.1:28333

Raspberry Pi optimizations
dbcache=100
maxorphantx=10
maxmempool=50
maxconnections=40
maxuploadtarget=5000

proxy=127.0.0.1:9050
bind=127.0.0.1
listenonion=1

anyone have an idea what that may be?

Thank you!

Stadicus, 2018-12-18:
Can you remove the `onlynet=ipv4` option and try again?

The error you mentioned regarding the rpcuser / rpcpassword is for information only and not relevant in this case.

waffi121, 2018-12-18:
of course. Now I see the address in the btc log.

unfortunately I still get an unreachable via bitnodes.

with curl icanhazip.com

I still see my real IP.

I have 8 peers.

Can I keep Node running like that?

Thank you and nice Christmas :)

Stadicus, 2018-12-23:
Hej @waffi121, great that it's up and running. I also have issues with bitnodes. Check out this comment for an (even better) alternative to check your node:
https://github.com/Stadicus/guides/issues/189#issuecomment-449638447

Stadicus, 2018-12-28:
Thanks for the update, at least the bitnodes api seems to be quite reliable. 

My node being down is correct, as I am working on the guide, constantly switching between Tor & clearnet. :-)