raspibolt/raspibolt #299: Cannot ssh to raspberry pi anymore

dlawant, 2018-12-06:
Hey everyone, sorry for such a noobie question. My node works on BTC and LND (testnet), but now I cannot ssh into my raspberry pi anymore. I get the following message:

"admin@XXX.XXX.X.XX: Permission denied (publickey)."

I tried to disable ufw but it still doesn't work. Can someone help me?

Stadicus, 2018-12-08:
Do you think it's an issue with the SSH keys? If so, you can enable the login without keys again (comment the lines again). 

Regarding UFW, what is the IP address of your Pi? for the above mentioned config, it needs to be something like `192.168.0.*`.

dlawant, 2018-12-10:
That was it! Things are working normally again. The UFW config was correct.

Tks!

closed 2018-12-11 by Stadicus