raspibolt/raspibolt #300: Problems with LND and NAT-PMP

Xaspr, 2018-12-09:
Hi, 

I'm having an issue with LND and uPNP. Can anyone help me out with this? 

When running `lnd` and unlocking the wallet on bitcoin testnet, I get the following output and error:  

> Attempting automatic RPC configuration to bitcoind
> Automatically obtained bitcoind's RPC credentials
> 2018-12-09 11:31:57.565 [INF] LTND: Version 0.5.0-beta commit=3b2c807288b1b7f40d609533c1e96a510ac5fa6d
> 2018-12-09 11:31:57.565 [INF] LTND: Active chain: Bitcoin (network=testnet)
> 2018-12-09 11:32:45.980 [INF] CHDB: Checking for schema update: latest_version=6, db_version=6
> 2018-12-09 11:32:45.985 [INF] RPCS: password RPC server listening on 127.0.0.1:10009
> 2018-12-09 11:32:45.985 [INF] RPCS: password gRPC proxy started at 127.0.0.1:8081
> 2018-12-09 11:32:45.985 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.
> 2018-12-09 11:32:56.038 [INF] LNWL: Opened wallet
> 2018-12-09 11:32:56.436 [INF] LTND: Primary chain is set to: bitcoin
> 2018-12-09 11:32:56.442 [INF] LTND: Initializing bitcoind backed fee estimator
> 2018-12-09 11:32:56.443 [INF] LNWL: Started listening for bitcoind block notifications via ZMQ on tcp://127.0.0.1:29000
> 2018-12-09 11:32:56.444 [INF] LNWL: Started listening for bitcoind transaction notifications via ZMQ on tcp://127.0.0.1:29001
> 2018-12-09 11:33:02.753 [INF] LNWL: The wallet has been unlocked without a time limit
> 2018-12-09 11:33:02.753 [INF] LTND: LightningWallet opened
> 2018-12-09 11:33:02.771 [INF] LNWL: Catching up block hashes to height 1447134, this will take a while...
> 2018-12-09 11:33:02.780 [INF] HSWC: Restoring in-memory circuit state from disk
> 2018-12-09 11:33:02.782 [INF] HSWC: Payment circuits loaded: num_pending=0, num_open=0
> 2018-12-09 11:33:02.783 [INF] SRVR: Scanning local network for a UPnP enabled device
> 2018-12-09 11:33:02.786 [INF] LNWL: Caught up to height 0
> 2018-12-09 11:33:06.998 [ERR] SRVR: Unable to discover a UPnP enabled device on the local network: multiple NATs detected
> 2018-12-09 11:33:06.998 [INF] SRVR: Scanning local network for a NAT-PMP enabled device
> 2018-12-09 11:33:07.014 [ERR] SRVR: Unable to discover a NAT-PMP enabled device on the local network: fork/exec /usr/bin/ip: no such file or directory
> 2018-12-09 11:33:07.014 [ERR] SRVR: unable to create server: Unable to discover a NAT-PMP enabled device on the local network: fork/exec /usr/bin/ip: no such file or directory
> 
> 2018-12-09 11:33:07.015 [INF] LTND: Shutdown complete
> Unable to discover a NAT-PMP enabled device on the local network: fork/exec /usr/bin/ip: no such file or directory
> 

The problem shouldn't be in my router I think. The Raspberry Pi has a fixed internal IP of 192.168.1.20. Here's some screenshots from the Netgear router: 

[image]
[image]
[image]

My bitcoin.conf looks as follows: 

> server=1
> daemon=1
> testnet=1
> txindex=1
> deprecatedrpc=addwitnessaddress
> deprecatedrpc=signrawtransaction
> rpcuser=xxx
> rpcpassword=xxx
> zmqpubrawblock=tcp://127.0.0.1:29000
> zmqpubrawtx=tcp://127.0.0.1:29001
> dbcache=100
> maxorphantx=10
> maxmempool=50
> maxconnections=40
> maxuploadtarget=5000

lnd.conf is:

> [Application Options]
> debuglevel=info
> maxpendingchannels=5
> alias=xxxx
> color=#68F442
> nat=true
> restlisten=localhost:8081
> 
> [Bitcoin]
> bitcoin.active=1
> bitcoin.testnet=1
> bitcoin.node=bitcoind
> 
> [autopilot]
> autopilot.active=1
> autopilot.maxchannels=5
> autopilot.allocation=0.6

The UFW status is: 
   
[image]

Any help would be much appreciated.

Xaspr, 2018-12-11:
Solved.

closed 2018-12-11 by Xaspr

Stadicus, 2018-12-11:
I was puzzled by this issue, as I have almost the same router. What solved it?

Fillippone, 2018-12-11:
Exactely the same problem, very similar configuration (first router is a Netgear D7000 and second router is my provider one). Not sure I can setup the second one as bridge, but opened all the ports on both reouter AND UPnP on both…. but still i have almost same problem (opned an issue here).. any thoughts?

Xaspr, 2018-12-12:
Port forwarding and setting UPnP on both the router/modem and the LAN-connected router didn't work for me either. 

The routers are probably on a different subnet right? Can you setup the firewall and NAT on the modem/router only and point the LAN-connected router to the modem/router gateway? 

I did this the other way around, by doing firewall and NAT on the LAN-connected router only and bridging the modem/router. But this should be fine too I think.