raspibolt/raspibolt #301: LND PNP failure

Fillippone, 2018-12-11:
Hello, after launching LND and unlocking wallet with the second SSH Shell the server starts looking for a UPnP device, fails and shutdown. OF course my router and modem are UPnP acrivated. 

Attempting automatic RPC configuration to bitcoind
Automatically obtained bitcoind's RPC credentials
2018-12-11 23:04:46.446 [INF] LTND: Version: 0.5.1-beta commit=, build=production, logging=default
2018-12-11 23:04:46.446 [INF] LTND: Active chain: Bitcoin (network=testnet)
2018-12-11 23:04:46.446 [INF] CHDB: Checking for schema update: latest_version=7, db_version=7
2018-12-11 23:04:46.450 [INF] RPCS: password RPC server listening on 127.0.0.1:10009
2018-12-11 23:04:46.450 [INF] RPCS: password gRPC proxy started at 127.0.0.1:8080
2018-12-11 23:04:46.451 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.
2018-12-11 23:05:47.503 [INF] LNWL: Opened wallet
2018-12-11 23:05:47.916 [INF] LTND: Primary chain is set to: bitcoin
2018-12-11 23:05:47.943 [INF] LNWL: Started listening for bitcoind block notifications via ZMQ on tcp://127.0.0.1:28332
2018-12-11 23:05:47.943 [INF] LNWL: Started listening for bitcoind transaction notifications via ZMQ on tcp://127.0.0.1:28333
2018-12-11 23:05:47.943 [INF] LTND: Initializing bitcoind backed fee estimator
2018-12-11 23:05:53.271 [INF] LNWL: The wallet has been unlocked without a time limit
2018-12-11 23:05:53.281 [INF] LTND: LightningWallet opened
2018-12-11 23:05:53.297 [INF] LNWL: Catching up block hashes to height 1447434, this will take a while...
2018-12-11 23:05:53.373 [INF] HSWC: Restoring in-memory circuit state from disk
2018-12-11 23:05:53.384 [INF] LNWL: Caught up to height 0
2018-12-11 23:05:53.396 [INF] HSWC: Payment circuits loaded: num_pending=0, num_open=0
2018-12-11 23:05:53.418 [INF] SRVR: Scanning local network for a UPnP enabled device
2018-12-11 23:06:03.425 [ERR] SRVR: Unable to discover a UPnP enabled device on the local network: context canceled
2018-12-11 23:06:03.426 [INF] SRVR: Scanning local network for a NAT-PMP enabled device
2018-12-11 23:06:03.442 [ERR] SRVR: Unable to discover a NAT-PMP enabled device on the local network: fork/exec /usr/bin/ip: no such file or directory
2018-12-11 23:06:03.442 [ERR] SRVR: unable to create server: Unable to discover a NAT-PMP enabled device on the local network: fork/exec /usr/bin/ip: no such file or directory

2018-12-11 23:06:03.442 [INF] LTND: Shutdown complete
Unable to discover a NAT-PMP enabled device on the local network: fork/exec /usr/bin/ip: no such file or directory

My network setup is something like this: 
The RasPi is wired to my Netgear N7000 acting as LAN router, then the Netgeart itself is connected to my Internet provider Modem/Router. 
Both have correct manual port forwarding setup (identical on both devices) and UPnP activated (even if i didn't use it before LND).
I cannot se any port request of opening Ports from RaspPI on relevant page od Netgear Firmware.)

Any suggestion?

starkana, 2018-12-16:
@Fillippone I solved the problem on my end by adding a symlink to /usr/bin/ip.

`sudo ln -l /sbin/ip /usr/bin/ip.`

I think you have the exact issue as me. The failure point is here:

`Unable to discover a NAT-PMP enabled device on the local network: fork/exec **/usr/bin/ip: no such file or directory**`

starkana, 2018-12-17:
Well try it and let us know

Stadicus, 2018-12-18:
Can you try to replicate the following steps and tell me if you get different results somewhere? 

When checking where the binary `ip` is located (using `whereis ip`), I get the following locations:
`ip: /bin/ip /sbin/ip /usr/share/man/man7/ip.7.gz /usr/share/man/man8/ip.8.gz`

This works fine, as shown in the lnd.log file:
```
$ sudo cat /home/bitcoin/.lnd/logs/bitcoin/mainnet/lnd.log | grep UPnP
2018-12-18 08:25:42.717 [INF] SRVR: Scanning local network for a UPnP enabled device
2018-12-18 08:25:46.982 [INF] SRVR: Automatically set up port forwarding using UPnP to advertise external IP
```

I run the Raspbian Stretch Lite binary referenced in the guide, without having the package `ìproute2` manually installed. But it's essential that it is installed:
```
$ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"

$ dpkg -s iproute2
Package: iproute2
Status: install ok installed
```

Also, you might want to check if `ip` is accessible with your current $PATH settings:
```
 $ echo $PATH
/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
```

Anyway, if creating a symlink for `/usr/bin/ip` works, I will add that to the guide.

Stadicus, 2018-12-22:
I added additional information about `nat=true` in that context:
https://github.com/Stadicus/guides/commit/4b18002556499d13c98274cd0d5f58154075863d

@starkana: could you check the configuration as detailed above?

wetherweasel, 2018-12-22:
Thank you @Stadicus! Great work. 
Just to be sure: My router has this UPnP option enabled but I still got this:

```
[ERR] SRVR: Unable to discover a NAT-PMP enabled device on the local network: fork/exec /usr/bin/ip: no such file or directory
 [ERR] SRVR: unable to create server: Unable to discover a NAT-PMP enabled device on the local network: fork/exec /usr/bin/ip: no such file or directory
```
I disabled it in lnd.conf as described
How do I know my node is talking to the outside now? 
(Sorry if this question might be answered already — I am all new to this, github included 😬)

Stadicus, 2018-12-22:
Did you try creating looking for the missing `ip` executable and creating a link to it, as suggested above?
* `whereis ip` gives you the location (which can be different than LND expects)
* `sudo ln -l /sbin/ip /usr/bin/ip` creates a link (where /sbin/ip is your actual path, as detected by whereis

Otherwise, you're fine when disabling NAT. See latest explanation here: 
https://github.com/Stadicus/guides/commit/4b18002556499d13c98274cd0d5f58154075863d#diff-cf8452295da60ae23ef1c54a61d7bf2e

If it is disabled, other nodes will not be able to detect you. But you are still able to open and maintain channels from your side. If you can open channels, you're connected to the network.

nflaig, 2018-12-22:
@Stadicus In the commit you wrote:
**If your router does not support UPnP**, LND will still work, but your node will be a private Lightning node for your own payments and not able to route payments for others.

My router does not support UPnP and therefore I had to set `externalip` in `lnd.conf`. This makes my node public and able to route payments. I had to open the port 9735 on my router  and in case my IP changes I used this [solution](https://github.com/robclark56/RaspiBolt-Extras/blob/master/RB_extra_02.md) to restart LND and set a new `externalip` (has to be passed dynamically as parameter in `lnd.service` and not hard coded in `lnd.conf`).

This solution requires a bit more work then just having UPnP enabled and `nat=true` but my node seems to be public and able to route payments (I collected some fees).

nflaig, 2018-12-22:
@wetherweasel You can search your node on a LN explorer for example [1ML](https://1ml.com/testnet/) (this is for the testnet). There you can see if your node is a public node or not. 
You can also give someone your public key and IP to let them connect to you as a peer which as far as I know only works if your node is public.

wetherweasel, 2018-12-22:
Now nothing works anymore — think I broke ~~some~~everything. LND wont either start nor shut down anymore with `$ LND` or `CTRL C`.🤦‍♂️

wetherweasel, 2018-12-22:
Thanks for the answers. It seems that I broke ~~some~~everything. LND wont either start nor shut down anymore with `$ LND` or `CTRL C`.🤦‍♂️Also for some weird reason I cant access anything: 
```
$ lncli getinfo
[lncli] unable to read macaroon path (check the network setting!): open /home/bitcoin/.lnd/data/chain/bitcoin/mainnet/admin.macaroon: no such file or directory
```
but when I run: 

`$ sudo ls -la /home/bitcoin/.lnd/data/chain/bitcoin/testnet/`

it gives me this:

```
total 124428
drwx------ 2 bitcoin bitcoin      4096 Dec 22 17:26 .
drwx------ 3 bitcoin bitcoin      4096 Dec 22 17:26 ..
-rw------- 1 bitcoin bitcoin       232 Dec 22 17:26 admin.macaroon
-rw-r--r-- 1 bitcoin bitcoin       115 Dec 22 17:26 invoice.macaroon
-rw------- 1 bitcoin bitcoin     32768 Dec 22 18:15 macaroons.db
-rw-r--r-- 1 bitcoin bitcoin       183 Dec 22 17:26 readonly.macaroon
-rw------- 1 bitcoin bitcoin 139902976 Dec 22 20:17 wallet.db

```

And since I dont know what I did I guess I have to start from scratch. Any suggestions how I can do this? bitcoind seems to work so maybe I can skip this step. Sorry if I spam this. Where would be a better place to go with this? Any help appreciated.

nflaig, 2018-12-22:
@wetherweasel You don't have to start from scratch in the worst case just delete everything inside the `/home/bitcoin/.lnd/` folder and follow [Lightning](https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_40_lnd.md) (you can skip the installation).

`lncli getinfo` only works if you run LND on the mainnet and the error you got after running `lncli --network=testnet getinfo` might be because your wallet is still locked.

Stadicus, 2018-12-23:
@nflaig thanks for the link to the great guide of @robclark56. I previously had a similar script in the guide (see older version [here](https://github.com/Stadicus/guides/blob/d3cef6ba81f8b224f72aaaa8d0b182d2ae65b8d2/raspibolt/raspibolt_40_lnd.md), but it's just not suitable for a beginner's guide.

ninedigits, 2018-12-25:
Also running into the same issue. When I run `lnd`, I get:

`2018-12-25 14:45:26.510 [ERR] SRVR: Unable to discover a NAT-PMP enabled device on the local network: multiple NATs detected`

Logs:
`$ sudo cat /home/bitcoin/.lnd/logs/bitcoin/testnet/lnd.log | grep UPnP`
>2018-12-25 14:37:52.483 [INF] SRVR: Scanning local network for a UPnP enabled device
2018-12-25 14:38:02.508 [ERR] SRVR: Unable to discover a UPnP enabled device on the local network: context canceled

I suspect the issue may be related to my router(s). I'm using Google Wifi through a bridge on my ISP's router/modem. I enabled port forwarding on both, but perhaps I missed something there. Both routers supposedly support UPnP.

ninedigits, 2018-12-26:
Thanks for the reply @Stadicus. You are correct. I recall having issues with the Google Wifi / Comcast router in bridge mode when I first purchased the Google Wifi system, so I left the bridge mode disabled since the internet was otherwise working fine.

MrPaz, 2018-12-28:
ok - this is for other noobs. I was getting a UPnP error very similar error to the OP, but not exactly the same, To wit:

```
[ERR] SRVR: Unable to discover a UPnP enabled device on the local network: context canceled
[INF] SRVR: Scanning local network for a NAT-PMP enabled device
[ERR] SRVR: Unable to discover a NAT-PMP enabled device on the local network: read udp 192.168.0.106:54003->192.168.0.1:5351: recvfrom: connection refused
[ERR] SRVR: unable to create server: Unable to discover a NAT-PMP enabled device on the local network: read udp 192.168.0.106:54003->192.168.0.1:5351: recvfrom: connection refused
```
Sadly I spent a couple of hours browsing various threads (such as this one) trying various remedies with no luck, until I finally realized... it was UFW.  Earlier in the guide, when setting up ufw we do:

`ufw allow proto udp from 192.168.0.0/24 port 1900 to any comment 'allow local LAN SSDP for UPnP discovery'`

yup, for me should have been 192.168.**1**.0/24 .  I changed it for ssh, but failed to change for upnp.  hope this saves someone a headache.

nfaig's fix worked in the interim before I figured this out

cilphex, 2018-12-29:
Fixed my `multiple NATs detected` error by connecting to my primary router, rather than the google wifi setup attached to it.

Now getting the error @MrPaz was having:

```
SRVR: unable to create server: Unable to discover a NAT-PMP enabled device on the local network: read udp 10.0.0.111:39023->10.0.0.1:5351: recvfrom: connection refused
```

I don't think it's the firewall. I tried disabling it and then unlocking LND, but still get the same thing.

cilphex, 2018-12-30:
@ninedigits Yeah, I have the ports enabled in `ufw` and I have them being forwarded by the router:

[image]

[image]

badokun, 2019-01-09:
I've also had trouble getting uPnP to work, but after looking at the ufw logs it was pretty clear I had to open another/different port. 

`ufw allow proto udp from 192.168.1.0/24 port 32987 to any comment 'allow local LAN SSDP for UPnP discovery 32987'`

Looking at this for reference, https://askubuntu.com/questions/1028570/ufw-blocking-upnp-port-mapping, I was able to determine the 32987 port was used instead of the 1900.

Perhaps we can include this in the troubleshooting section. Happy to do it

Stadicus, 2019-01-11:
Can you point out, where & what in the LND log exactly shows what UPnP port is used? Did you search for similar issues in the LND repository?

badokun, 2019-01-11:
> Can you point out, where & what in the LND log exactly shows what UPnP port is used? Did you search for similar issues in the LND repository?

Firstly to view the logs:
` $ sudo su`
 ` $ tail /var/log/ufw.log`

I was able to isolate some of the messages, since I was monitoring what is happening as soon as I unlocked my Lightning wallet (which initiates the process). By looking at the two ip addresses involved you can confirm it's the Raspberry talking to the router. In the case below, `192.168.1.1` and `192.168.1.40`. Also the `PROTO=UDP` was indicative of a uPnP call. The ports involved are `SPT` and `DPT`. 
```
Jan  8 23:45:10 RaspiBolt kernel: [ 6368.941299] [UFW BLOCK] IN=eth0 OUT= MAC=xxxxxxx SRC=192.168.1.1 DST=192.168.1.40 LEN=388 TOS=0x00 PREC=0x00 TTL=64 ID=0 DF PROTO=UDP SPT=32969 DPT=55374 LEN=368
```

In my case, perhaps a router uPnP implementation, the ports are dynamically set. I'm sure there's something about this in the uPnP protocol spec, but short of reading through that I was just opening UDP comms between my Raspberry and the router. 

It's worth pointing out that other services on my home network don't exhibit this issue, e.g. Plex. That may just be because the PCs they're running on don't block UDP

badokun, 2019-01-11:
> https://www.yougetsignal.com/tools/open-ports/
> Is port Port 8333 and 18333 show as open when you check on site above?

Yes, confirmed open

Stadicus, 2019-01-29:
Not sure I can be of any help here. Did you try posting an issue directly on the LND github?

xkra, 2019-02-10:
First, I want to thank you for the excellent guide and the resources invested in enabling less tech-experienced users to set up their own RaspiBolt!

I have an issue, which from reading through this thread, seems to be similar but not the same:
`Feb 10 14:45:10 RaspiBolt lnd[840]: 2019-02-10 14:45:10.874 [INF] SRVR: Scanning local network for a UPnP enabled device

Feb 10 14:45:20 RaspiBolt lnd[840]: 2019-02-10 14:45:20.880 [ERR] SRVR: Unable to discover a UPnP enabled device on the local network: context canceled

Feb 10 14:45:20 RaspiBolt lnd[840]: 2019-02-10 14:45:20.880 [INF] SRVR: Scanning local network for a NAT-PMP enabled device

Feb 10 14:45:20 RaspiBolt lnd[840]: 2019-02-10 14:45:20.911 [ERR] SRVR: Unable to discover a NAT-PMP enabled device on the local network: read udp 192.168.1.122:43219->192.168.1.1:5351: recvfrom: connection refused

Feb 10 14:45:20 RaspiBolt systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE

Feb 10 14:45:20 RaspiBolt systemd[1]: lnd.service: Unit entered failed state.

Feb 10 14:45:20 RaspiBolt systemd[1]: lnd.service: Failed with result 'exit-code'.`

I have enabled NAT and UPnP in my router GUI, and have also set up port forwarding for bitcoin (8333), bitcoin test (18333), plus one for lightning (9735) - the latter I didn't find in the guide but figured out somehow.

Stadicus, 2019-02-21:
Great to hear. Do you think I should add some additional information to the guide?

Wriazuk, 2019-02-22:
> Great to hear. Do you think I should add some additional information to the guide?

Hi, thanks for the excellent guide. I'm also having trouble with UPNP, however. I've found that if I have UFW enabled I cannot connect to lnd and get the same "Unable to discover UPNP..." error.
When i disable UFW I can connect to the testnet, and have confirmed I was public on 1ML. 

Any idea whats going on and what to do from here?

Stadicus, 2019-03-01:
Have you tried to open UFW completely for UPnP?
https://github.com/Stadicus/guides/issues/301#issuecomment-452576155

Wriazuk, 2019-03-02:
> Have you tried to open UFW completely for UPnP?
> [#301 (comment)](https://github.com/Stadicus/guides/issues/301#issuecomment-452576155)

Wow, it worked! Thanks mate, cheers!

closed 2019-03-03 by Stadicus

dmar198, 2019-03-18:
I followed the recommendation to run this command: "sudo ln -l /sbin/ip /usr/bin/ip"

And I got this result: ln: invalid option -- 'l'

Did the method for creating symbolic links change?