raspibolt/raspibolt #302: lnd.service fails after setup with TOR

Cees2daBees, 2018-12-16:
Running:
Raspberry Pi 3 B+
Bitcoin Core Daemon version v0.17.0.1
lnd version 0.5.0-beta commit=3b2c807288b1b7f40d609533c1e96a510ac5fa6d

Was attempting to set up Tor using https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_69_tor.md

Set up Tor for Bitcoin Core seemed to work ok, except my node doesn't appear to be reachable at bitnodes.earn.com

Setting up Tor for LND created failed state and below log:

`Dec 16 16:29:17 RaspiBolt systemd[1]: Started LND Lightning Daemon.
Dec 16 16:29:18 RaspiBolt systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Dec 16 16:29:18 RaspiBolt systemd[1]: lnd.service: Unit entered failed state.
Dec 16 16:29:18 RaspiBolt systemd[1]: lnd.service: Failed with result 'exit-code'.
Dec 16 16:30:18 RaspiBolt systemd[1]: lnd.service: Service hold-off time over, scheduling restart.
Dec 16 16:30:18 RaspiBolt systemd[1]: Stopped LND Lightning Daemon.
Dec 16 16:30:18 RaspiBolt systemd[1]: Started LND Lightning Daemon.
Dec 16 16:30:18 RaspiBolt systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Dec 16 16:30:18 RaspiBolt systemd[1]: lnd.service: Unit entered failed state.
Dec 16 16:30:18 RaspiBolt systemd[1]: lnd.service: Failed with result 'exit-code'.
Dec 16 16:31:18 RaspiBolt systemd[1]: lnd.service: Service hold-off time over, scheduling restart.
Dec 16 16:31:18 RaspiBolt systemd[1]: Stopped LND Lightning Daemon.
Dec 16 16:31:18 RaspiBolt systemd[1]: Started LND Lightning Daemon.
Dec 16 16:31:18 RaspiBolt systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Dec 16 16:31:18 RaspiBolt systemd[1]: lnd.service: Unit entered failed state.
Dec 16 16:31:18 RaspiBolt systemd[1]: lnd.service: Failed with result 'exit-code'.
Dec 16 16:32:18 RaspiBolt systemd[1]: lnd.service: Service hold-off time over, scheduling restart.
Dec 16 16:32:18 RaspiBolt systemd[1]: Stopped LND Lightning Daemon.
Dec 16 16:32:18 RaspiBolt systemd[1]: Started LND Lightning Daemon.
Dec 16 16:32:19 RaspiBolt lnd[18654]: /home/bitcoin/.lnd/lnd.conf:25: unknown option: tor.active
Dec 16 16:32:19 RaspiBolt systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Dec 16 16:32:19 RaspiBolt systemd[1]: lnd.service: Unit entered failed state.
Dec 16 16:32:19 RaspiBolt systemd[1]: lnd.service: Failed with result 'exit-code'.
Dec 16 16:33:19 RaspiBolt systemd[1]: lnd.service: Service hold-off time over, scheduling restart.
Dec 16 16:33:19 RaspiBolt systemd[1]: Stopped LND Lightning Daemon.
Dec 16 16:33:19 RaspiBolt systemd[1]: Started LND Lightning Daemon.
Dec 16 16:33:19 RaspiBolt systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Dec 16 16:33:19 RaspiBolt systemd[1]: lnd.service: Unit entered failed state.
Dec 16 16:33:19 RaspiBolt systemd[1]: lnd.service: Failed with result 'exit-code'.
Dec 16 16:34:19 RaspiBolt systemd[1]: lnd.service: Service hold-off time over, scheduling restart.
Dec 16 16:34:19 RaspiBolt systemd[1]: Stopped LND Lightning Daemon.`

As you can see, there was an output that read: 
`/home/bitcoin/.lnd/lnd.conf:25: unknown option: tor.active`
but it didn't output that every cycle.

After commenting out the changes to lnd.conf 
`#tor.active=1
#tor.v3=1
#listen=localhost`

LND started working again and syncing.

Any thoughts?

Stadicus, 2018-12-18:
Can you try to run LND manually, as described in the Troubleshooting guide?
https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_70_troubleshooting.md#lnd

I'd leave the options commented out in the lnd.conf and try setting them via command line (eg. `lnd --tor.active`)

Cees2daBees, 2018-12-20:
Hi @Stadicus, so I did what you suggested and it worked! (note: I had to comment out `nat=true` though) However, when I try to go back to normal operations, I uncommented the tor lines again
`#tor.active=1`
`#tor.v3=1`
`#listen=localhost`
and LND still failed.

I thought since when running LND manually I set tor.active using `lnd --tor.active`, that maybe I would try uncommenting only the `tor.active=1` line. But, sadly, that didn't work either and I got the below output:

admin@RaspiBolt:~ $ sudo journalctl -f -u lnd
-- Logs begin at Thu 2016-11-03 17:16:42 UTC. --
Dec 20 00:50:54 RaspiBolt lnd[27249]: 2018-12-20 00:50:54.933 [INF] LTND: Active chain: Bitcoin (network=mainnet)
Dec 20 00:50:54 RaspiBolt lnd[27249]: 2018-12-20 00:50:54.933 [INF] CHDB: Checking for schema update: latest_version=6, db_version=6
Dec 20 00:50:54 RaspiBolt lnd[27249]: 2018-12-20 00:50:54.952 [INF] RPCS: password RPC server listening on 127.0.0.1:10009
Dec 20 00:50:54 RaspiBolt lnd[27249]: 2018-12-20 00:50:54.952 [INF] RPCS: password gRPC proxy started at 127.0.0.1:8080
Dec 20 00:50:54 RaspiBolt lnd[27249]: 2018-12-20 00:50:54.953 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.
Dec 20 00:51:15 RaspiBolt lnd[27249]: 2018-12-20 00:51:15.941 [INF] LNWL: Opened wallet
Dec 20 00:51:16 RaspiBolt lnd[27249]: 2018-12-20 00:51:16.299 [INF] LTND: Primary chain is set to: bitcoin
Dec 20 00:51:16 RaspiBolt lnd[27249]: 2018-12-20 00:51:16.307 [INF] LNWL: Started listening for bitcoind block notifications via ZMQ on tcp://127.0.0.1:28332
Dec 20 00:51:16 RaspiBolt lnd[27249]: 2018-12-20 00:51:16.307 [INF] LNWL: Started listening for bitcoind transaction notifications via ZMQ on tcp://127.0.0.1:28333
Dec 20 00:51:16 RaspiBolt lnd[27249]: 2018-12-20 00:51:16.307 [INF] LTND: Initializing bitcoind backed fee estimator
Dec 20 00:51:21 RaspiBolt lnd[27249]: 2018-12-20 00:51:21.686 [INF] LNWL: The wallet has been unlocked without a time limit
Dec 20 00:51:21 RaspiBolt lnd[27249]: 2018-12-20 00:51:21.686 [INF] LTND: LightningWallet opened
Dec 20 00:51:21 RaspiBolt lnd[27249]: 2018-12-20 00:51:21.924 [INF] SRVR: Proxying all network traffic via Tor (stream_isolation=false)! NOTE: Ensure the backend node is proxying over Tor as well
Dec 20 00:51:21 RaspiBolt lnd[27249]: 2018-12-20 00:51:21.930 [INF] HSWC: Restoring in-memory circuit state from disk
Dec 20 00:51:21 RaspiBolt lnd[27249]: 2018-12-20 00:51:21.940 [INF] HSWC: Payment circuits loaded: num_pending=3, num_open=0
Dec 20 00:51:22 RaspiBolt lnd[27249]: 2018-12-20 00:51:22.112 [INF] LNWL: Started rescan from block 000000000000000000187c341e93d4fa0040c501df5158f0ab676cbc79f048eb (height 554557) for 166 addresses
Dec 20 00:51:22 RaspiBolt lnd[27249]: 2018-12-20 00:51:22.135 [INF] LNWL: Starting rescan from block 000000000000000000187c341e93d4fa0040c501df5158f0ab676cbc79f048eb
Dec 20 00:51:22 RaspiBolt lnd[27249]: 2018-12-20 00:51:22.139 [INF] RPCS: RPC server listening on 127.0.0.1:10009
Dec 20 00:51:22 RaspiBolt lnd[27249]: 2018-12-20 00:51:22.140 [INF] RPCS: gRPC proxy started at 127.0.0.1:8080
Dec 20 00:51:22 RaspiBolt lnd[27249]: 2018-12-20 00:51:22.152 [INF] LTND: Waiting for chain backend to finish sync, start_height=554557
Dec 20 00:51:22 RaspiBolt lnd[27249]: 2018-12-20 00:51:22.154 [INF] LNWL: Rescan finished at 554557 (000000000000000000187c341e93d4fa0040c501df5158f0ab676cbc79f048eb)
Dec 20 00:51:22 RaspiBolt lnd[27249]: 2018-12-20 00:51:22.154 [INF] LNWL: Catching up block hashes to height 554557, this might take a while
Dec 20 00:51:22 RaspiBolt lnd[27249]: 2018-12-20 00:51:22.165 [INF] LNWL: Done catching up block hashes
Dec 20 00:51:22 RaspiBolt lnd[27249]: 2018-12-20 00:51:22.165 [INF] LNWL: Finished rescan for 166 addresses (synced to block 000000000000000000187c341e93d4fa0040c501df5158f0ab676cbc79f048eb, height 554557)
Dec 20 00:51:22 RaspiBolt lnd[27249]: 2018-12-20 00:51:22.172 [INF] LTND: Chain backend is fully synced (end_height=554557)!
Dec 20 00:51:22 RaspiBolt lnd[27249]: 2018-12-20 00:51:22.851 [ERR] SRVR: unable to start server: unable to write private key to file: open : no such file or directory
Dec 20 00:51:22 RaspiBolt lnd[27249]: 2018-12-20 00:51:22.854 [INF] LTND: Shutdown complete
Dec 20 00:51:22 RaspiBolt lnd[27249]: unable to write private key to file: open : no such file or directory
Dec 20 00:51:22 RaspiBolt systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Dec 20 00:51:22 RaspiBolt systemd[1]: lnd.service: Unit entered failed state.
Dec 20 00:51:22 RaspiBolt systemd[1]: lnd.service: Failed with result 'exit-code'.

Is there some other way I need to set the tor options in lnd.conf to get this to work?

elkimek, 2018-12-20:
Hi @Stadicus I have similar issue.

When I put following lines in lnd.conf lnd.service is failing to start. If I comment those, everything is just fine.

```
[Tor]
tor.active=1
tor.v3=1
listen=localhost
nat=true
```
logs:
```
Dec 20 18:00:25 ElkimLND lnd[6545]: /home/bitcoin/.lnd/lnd.conf:30: unknown option: listen
Dec 20 18:00:25 ElkimLND systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Dec 20 18:00:25 ElkimLND systemd[1]: lnd.service: Unit entered failed state.
Dec 20 18:00:25 ElkimLND systemd[1]: lnd.service: Failed with result 'exit-code'.
Dec 20 18:01:25 ElkimLND systemd[1]: lnd.service: Service hold-off time over, scheduling restart.
Dec 20 18:01:25 ElkimLND systemd[1]: Stopped LND Lightning Daemon.
Dec 20 18:01:25 ElkimLND systemd[1]: Started LND Lightning Daemon.
Dec 20 18:01:26 ElkimLND systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Dec 20 18:01:26 ElkimLND systemd[1]: lnd.service: Unit entered failed state.
Dec 20 18:01:26 ElkimLND systemd[1]: lnd.service: Failed with result 'exit-code'.
Dec 20 18:02:26 ElkimLND systemd[1]: lnd.service: Service hold-off time over, scheduling restart.
Dec 20 18:02:26 ElkimLND systemd[1]: Stopped LND Lightning Daemon.
Dec 20 18:02:26 ElkimLND systemd[1]: Started LND Lightning Daemon.
Dec 20 18:02:27 ElkimLND systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Dec 20 18:02:27 ElkimLND systemd[1]: lnd.service: Unit entered failed state.
Dec 20 18:02:27 ElkimLND systemd[1]: lnd.service: Failed with result 'exit-code'.
```

If I run it manually, I'm just getting:

```
/home/bitcoin/.lnd/lnd.conf:30: unknown option: listen
```

Thanks in advance. Running LND 0.5.1.

elkimek, 2018-12-20:
Hey yes, that works :) my IP is gone.

Now to the point how to make it working with systemctl :)

Cees2daBees, 2018-12-21:
I wonder if this is the problem, I just checked my Tor version and found that I'm running
`Tor version 0.2.9.16 (git-9ef571339967c1e5)`
And the guide says that LND requires Tor 0.3.3.6 or newer.
I'll try to update Tor and see what happens.

Cees2daBees, 2018-12-21:
Nope, updated to
`Tor version 0.3.4.9 (git-074ca2e0054fded1)`
but still getting LND failed state. =(

elkimek, 2018-12-21:
Same version here :(

elkimek, 2018-12-21:
Solved.

Under ``[Application Options]`` I had:

```listen=0.0.0.0:9735```

Changed to:

```listen=localhost```

Stadicus, 2018-12-22:
This is my current lnd.conf, which works without issues. If I use `nat=true`, I get a second ip-address URI, if I comment it out, it's gone after a LND restart.

```
[Application Options]
debuglevel=info
maxpendingchannels=5
#nat=true
tor.active=1
tor.v3=1
listen=localhost

[Bitcoin]
bitcoin.active=1
bitcoin.mainnet=1
bitcoin.node=bitcoind
```

My versions:
```
admin@RaspiBolt:~ $ tor --version
Tor version 0.3.4.9 (git-074ca2e0054fded1).
admin@RaspiBolt:~ $ lnd --version
lnd version 0.5.0-beta commit=3b2c807288b1b7f40d609533c1e96a510ac5fa6d
```

elkimek, 2018-12-22:
Tried your way, even closed all my channels and created new wallet. Still seeing IPV4 in the outputs :(

elkimek, 2018-12-22:
Fixed, my lnd.service was calling the IP script :)

Stadicus, 2018-12-23:
@Cees2daBees, can you verify your configuration with the updated guide here?  
https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_69_tor.md

RobJah, 2018-12-23:
Now all we need to know is how is lnd.service calling the IP script and better yet how will i stop that?

elkimek, 2018-12-23:
I guess you probably create the lnd.service according to old guide which was using the script like I did. Just change it as described here:

https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_40_lnd.md#autostart-lnd

RobJah, 2018-12-23:
Thnx. My lnd.service says:

[Unit]
Description=LND Lightning Daemon
Wants=bitcoind.service
After=bitcoind.service

# for use with sendmail alert
#OnFailure=systemd-sendmail@%n

[Service]
# get var PUBIP from file
EnvironmentFile=/run/publicip

ExecStart=/usr/local/bin/lnd --externalip=${PUBLICIP}
PIDFile=/home/bitcoin/.lnd/lnd.pid
User=bitcoin
Group=bitcoin
LimitNOFILE=128000
Type=simple
KillMode=process
TimeoutSec=180
Restart=always
RestartSec=60

[Install]
WantedBy=multi-user.target

what do i need to change?

Stadicus, 2018-12-23:
Yes, the guide used the IP script, as LND was not NAT UPnP-ready yet. Now that this works, I retired the script and you just have to adjust the lnd.service configuration pointed out by @elkimek.

Stadicus, 2018-12-23:
Something like this:

```
[Unit]
Description=LND Lightning Daemon
Wants=bitcoind.service
After=bitcoind.service

[Service]
ExecStart=/usr/local/bin/lnd 
PIDFile=/home/bitcoin/.lnd/lnd.pid
User=bitcoin
Group=bitcoin
LimitNOFILE=128000
Type=simple
KillMode=process
TimeoutSec=240
Restart=always
RestartSec=60

[Install]
WantedBy=multi-user.target
```

RobJah, 2018-12-23:
dont know what cees is talking about, but the lnd.service input solved my lnd issue wher it didnt start.
the only problem left is that https://bitnodes.earn.com/ still says my node is unreachable.

admin@RaspiBolt:~ $ curl icanhazip.com
62.163.166.236
admin@RaspiBolt:~ $ bitcoin-cli getpeerinfo | grep  local
    "addrlocal": "46.165.230.5:41122",
    "addrlocal": "192.160.102.169:39257",
    "addrlocal": "qwe54z7pugw7ilvh.onion:8333",
    "addrlocal": "94.230.208.147:47032",
    "addrlocal": "97.74.237.196:44457",
    "addrlocal": "51.15.3.40:40894",
    "addrlocal": "185.242.113.224:51498",
    "addrlocal": "185.220.100.253:3796",
    "addrlocal": "192.160.102.168:36361",

Stadicus, 2018-12-23:
bitnodes.earn.com seems to be broken for .onion addresses. My node is also not detected, but according to this link it's reachable:
https://bitnodes.earn.com/nodes/7qzi7ekmbkhwz36z.onion-8333/

elkimek, 2018-12-23:
Yeh bitnodes check works kinda often...

closed 2019-01-08 by Stadicus