raspibolt/raspibolt #304: Using tag and labels in github

cedricwalter, 2018-12-19:
Hi There :-)

Can you try to add tags in github to show where you need help, what you expect to have in 1.0, and so on.
I would like to help but for efficiency would like to filter which topics is more urgent than another.

regards
Cédric

Stadicus, 2018-12-21:
Hej guys, thanks for the offer. I guess I don't have too grand a vision for how this guide should evolve, so every idea is welcome.

What helps most is engaging with issues, help me solve open questions and improve the existing explanations in the guide.

Regarding new features, here are a few ideas.
* I removed the **lncli-web** section, as it was quite error-prone and I not really able to support it. Maybe there's an opportunity?
* A good description on how to generate external SSL certificates in combination with DynDNS (eg. to use for external access for mobile wallets like Shango, ZAP or Lightning Joule
* How to connect Lightning Joule

How exactly would you use lables? Create new issues with "Help wanted" for task, or something like that?

Stadicus, 2019-01-08:
Thanks @Septem151! 
https://github.com/Stadicus/guides/pull/333

closed 2019-02-21 by Stadicus