raspibolt/raspibolt #305: Failed to start bitcoind on mainnet

yhyuan, 2018-12-21:
Hi
After I copied the blocks and chainstate folders from anther machine with bitcoind 0.17.1, the bitcoind on the raspberry pi continues to print the following information. It seems something is wrong. Did someone have a clue to fix this issue?
> 2018-12-21T07:26:38Z Bitcoin Core version v0.17.0.1 (release build)
2018-12-21T07:26:38Z InitParameterInteraction: parameter interaction: -whitelistforcerelay=1 -> setting -whitelistrelay=1
2018-12-21T07:26:38Z Assuming ancestors of block 0000000000000000002e63058c023a9a1de233554f28c7b21380b6c9003f36a8 have valid signatures.
2018-12-21T07:26:38Z Setting nMinimumChainWork=0000000000000000000000000000000000000000028822fef1c230963535a90d
2018-12-21T07:26:38Z Using the 'standard' SHA256 implementation
2018-12-21T07:26:38Z Default data directory /home/bitcoin/.bitcoin
2018-12-21T07:26:38Z Using data directory /home/bitcoin/.bitcoin
2018-12-21T07:26:38Z Using config file /home/bitcoin/.bitcoin/bitcoin.conf
2018-12-21T07:26:38Z Using at most 40 automatic connections (1024 file descriptors available)
2018-12-21T07:26:38Z Using 16 MiB out of 32/2 requested for signature cache, able to store 524288 elements
2018-12-21T07:26:38Z Using 16 MiB out of 32/2 requested for script execution cache, able to store 524288 elements
2018-12-21T07:26:38Z Using 4 threads for script verification
2018-12-21T07:26:38Z scheduler thread start
2018-12-21T07:26:38Z HTTP: creating work queue of depth 16
2018-12-21T07:26:38Z Config options rpcuser and rpcpassword will soon be deprecated. Locally-run instances may remove rpcuser to use cookie-based auth, or may be replaced with rpcauth. Please see share/rpcauth for rpcauth auth generation.
2018-12-21T07:26:38Z HTTP: starting 4 worker threads
2018-12-21T07:26:38Z Using wallet directory /home/bitcoin/.bitcoin
2018-12-21T07:26:38Z init message: Verifying wallet(s)...
2018-12-21T07:26:38Z Using BerkeleyDB version Berkeley DB 4.8.30: (April  9, 2010)
2018-12-21T07:26:38Z Using wallet wallet.dat
2018-12-21T07:26:38Z BerkeleyEnvironment::Open: LogDir=/home/bitcoin/.bitcoin/database ErrorFile=/home/bitcoin/.bitcoin/db.log
2018-12-21T07:26:38Z Cache configuration:
2018-12-21T07:26:38Z * Using 2.0MiB for block index database
2018-12-21T07:26:38Z * Using 12.2MiB for transaction index database
2018-12-21T07:26:38Z * Using 8.0MiB for chain state database
2018-12-21T07:26:38Z * Using 77.8MiB for in-memory UTXO set (plus up to 47.7MiB of unused mempool space)
2018-12-21T07:26:38Z init message: Loading block index...
2018-12-21T07:26:38Z Opening LevelDB in /home/bitcoin/.bitcoin/blocks/index
2018-12-21T07:26:39Z Opened LevelDB successfully
2018-12-21T07:26:39Z Using obfuscation key for /home/bitcoin/.bitcoin/blocks/index: 0000000000000000

Stadicus, 2018-12-21:
That's all standard output. What happens next? Did you try to start it manually as described in the troubleshooting section?

yhyuan, 2018-12-21:
I got it. it is.
> bitcoind:

 chain.cpp:102: const CBlockIndex* CBlockIndex::GetAncestor(int) const: Assertion `pindexWalk->pprev' failed.
Aborted

Stadicus, 2018-12-22:
This is not a "normal" error and points to some underlying problem (hardware or operating system). There is a similar issue in the Bitcoin repo, please have a look at it.
https://github.com/bitcoin/bitcoin/issues/14750

This could be due to file system corruption, are you running Linux kernel 0.4.19? Check what kernel version are you using --> `$ uname -a` should be  
`Linux RaspiBolt 4.14.71-v7+ #1145 SMP Fri Sep 21 15:38:35 BST 2018 armv7l GNU/Linux`

I'd reformat the external hard disk and copy the blockchain from scratch. You could also try to ask for further assistance in the Bitcoin repo.

Stadicus, 2019-01-08:
Closting for now. Feel free to open a new issue if you need assistance.

closed 2019-01-08 by Stadicus