raspibolt/raspibolt #306: Problem with bitcoin-cli

isaactyc, 2018-12-23:
Hi. I'm trying to jump onto the mainnet. After copying everything over, I have been facing problems ever since. Opening a Putty session will hang the thing for at least 10mins. I cannot get bitcoin-cli nor lncli to work. When calling 'bitcoin-cli getblockchaininfo', I get "error: couldn't connect to server: EOF reached (code 1) (make sure server is running and you are connecting to the correct RPC port)". Trying to create a lnd wallet gives me "[lncli] unable to generate seed: rpc error: code = Unavailable desc = grpc: the connection is unavailable"

I have tried a few different solutions here to no avail. The solutions included: Changing the rpcpassword, [listing out the commands](https://github.com/Stadicus/guides/issues/136#issuecomment-405274999) <- the result of that can be found in [this pastebin](https://pastebin.com/YssR4hhn). I'm not quite sure what else to do or how to find the source of the error. Any tips on this would be great. Thank you!

Stadicus, 2018-12-23:
If the SSH setting hangs, this could be to heavy CPU / memory load. I would try the following steps (see [Troubleshooting guide](https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_70_troubleshooting.md) for additional details):
* disable systemd services bitcoind & lnd
* restart Pi
* start bitcoind manually from the command line and check output
* optionally: check system resource usage with `htop` from second ssh session

Judging from your pastebin, bitcoind gets shut down by the system somehow.

isaactyc, 2018-12-23:
I got it. When I typed sudo ufw status, I got an error couldn't stat '/etc/ufw/before6.rules' and couldn't stat '/etc/ufw/before.rules' 

typed sudo cp -p /usr/share/ufw/before6.rules /etc/ufw/
 and sudo cp -p /usr/share/ufw/before.rules /etc/ufw/

and after going through the rest of the troubleshooting, it has managed to work well. Thank you so much for the guide!

closed 2018-12-23 by isaactyc