raspibolt/raspibolt #310: System Overview problems

teasider, 2018-12-25:
Hi,
I'm running a bitcoind node + LND and wanted to use the system overview script
It operates smoothly when connecting on putty.
[image]

My issues are:
**1)** Even though It shows I am reachable through node checking website and port checking (8333 is open) - the script says "not reachable" in red.
When I run nc -z localhost:8333 it shows its open
When I run nc -z ip:8333 the program just hangs, might have something to do with that?
Info: I am not using any firewall currently on the linux box until i solve it and the connection is through a vpn that allows port-forwarding. (I am reachable on a node checking website and any port checking website)

**2)** No data on the LND column  - at all.
everything is setup well (directory is /home/username/.lnd on the script setup)
I have no idea why it doesn't draw data through there.

I would appreciate any help or information on the subject if anyone have any idea.
Thank you!

Septem151, 2019-01-04:
1) Check how many connections you have with `bitcoin-cli getconnectioncount` . If this number is 8 or lower, you only have outbound connections. If it's greater, congratulations! You have inbound connections and the system overview script is lying.

2) The LND column is really buggy and rarely shows any info for me, either, whether I'm in testnet or mainnet. I've only seen info there ONCE and I don't even know why. Other than that, it will say "Wallet locked" when my wallet is locked (but this only happens on mainnet).

rbuckhoy, 2019-01-04:
Per issue #233 this works with sudo for me.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

teasider, 2019-01-04:
Thank you - It is just a gimmick and was nice when SSHing to the machine.
I'll use the regular methods and also started using Zap with the LND node which works great.

closed 2019-01-04 by teasider