raspibolt/raspibolt #313: Unable to discover a UPnP enabled device on the local network: context canceled

hemzer, 2018-12-27:
Help!
Logs below.
Running:
Model	Linksys WRT1900ACS
Architecture	ARMv7 Processor rev 1 (v7l)
package MINIUPNPD is installed and running. Can see it in Chain zone_wan_forward  & Chain zone_wan_prerouting
I am not sure what else I need to put in here.  Stuck at this point. Yes I did read trough other tickets. was milling link to ip to which I have set it up. This issue after that correction. 

2018-12-26 20:23:00.917 [INF] LTND: LightningWallet opened
2018-12-26 20:23:02.138 [INF] LNWL: Catching up block hashes to height 1000007, this will take a while...
2018-12-26 20:23:02.156 [INF] LNWL: Caught up to height 0
2018-12-26 20:23:02.211 [INF] HSWC: Restoring in-memory circuit state from disk
2018-12-26 20:23:02.219 [INF] HSWC: Payment circuits loaded: num_pending=0, num_open=0
2018-12-26 20:23:02.232 [INF] SRVR: Scanning local network for a UPnP enabled device
2018-12-26 20:23:12.232 [ERR] SRVR: Unable to discover a UPnP enabled device on the local network: context canceled
2018-12-26 20:23:12.233 [INF] SRVR: Scanning local network for a NAT-PMP enabled device
2018-12-26 20:23:12.317 [ERR] SRVR: Unable to discover a NAT-PMP enabled device on the local network: read udp 192.168.1.1                45:40618->192.168.1.1:5351: recvfrom: connection refused
2018-12-26 20:23:12.317 [ERR] SRVR: unable to create server: Unable to discover a NAT-PMP enabled device on the local netw                ork: read udp 192.168.1.145:40618->192.168.1.1:5351: recvfrom: connection refused

Stadicus, 2018-12-29:
Seems like LND is not able to detect your UPnP device. You can try to pass the ip manually (see [here](https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_40_lnd.md) just below the `lnd.conf`.

Or you can try to address the issue why it is not detected in the first place. For that, the LND repo would be the right address.

Septem151, 2018-12-31:
I'm getting this same issue. I've checked that my Gateway is enabled for UPnP. I've added port forwarding for 9735 (tried both with and without). I've tried entering my own IP by running `lnd --externalip 1.2.3.4` . I've tried using both WiFi and Ethernet connections. Nothing I do seems to solve this issue besides disabling `nat=true` . Which is extremely disappointing since I want to run a FULL node and participate in routing transactions.

Septem151, 2018-12-31:
> @Septem151 Is your firewall configured for this? As far as I can see this is missing in the Lightning guide:
> 
> ```
> sudo ufw allow proto udp from 192.168.0.0/24 port 1900 to any comment 'allow local LAN SSDP for UPnP discovery'
> ```
> You might need to adjust the IP/subnet address to fit your needs. Just have a look what your other ufw settings look like:
> 
> ```
> sudo ufw status
> ```

Yes, I have this configured in my firewall. However since my IP starts with 10.0 , I have 
`sudo ufw allow proto udp from 10.0.0.0/24 port 1900 to any comment 'allow local LAN SSDP for UPnP discovery'`

I assumed I was supposed to put .0.0 at the end, regardless of the actual Static IP of my Pi. Is this assumption correct?

Stadicus, 2018-12-31:
Yes, this subnet mask covers all addresses after the last `.`, eg. 10.0.0.0 to 10.0.0.255.

Stadicus, 2019-01-08:
If your IP changes from time to time, you might want to try this script:
https://github.com/robclark56/RaspiBolt-Extras/blob/master/RB_extra_02.md

closed 2019-01-08 by Stadicus