raspibolt/raspibolt #314: Bitcoin received from a wallet was inexplicably transferred out of my wallet

DecisiveIndecisive, 2018-12-27:
I used [this](https://coinfaucet.eu/en/btc-testnet/) faucet to get some bitcoin to 2N5aYoFPhgKewcgACusnpJ8krMggutBFb7n. You can see the transaction [here](https://testnet.smartbit.com.au/address/2N5aYoFPhgKewcgACusnpJ8krMggutBFb7n).

Cmd was giving me the following and then all balances became 0. After investigation, I noticed there was a transaction that cleared out the wallet and the current balance, according to smartbit is 0.
```
{
    "total_balance": "13806779",
    "confirmed_balance": "0",
    "unconfirmed_balance": "13806779"
}
```

However now if I checkbalance I get the following... and am very confused:
```
{
    "total_balance": "551941",
    "confirmed_balance": "551941",
    "unconfirmed_balance": "0"
}
```

listchaintxns shows:
```
{
    "transactions": [
        {
            "tx_hash": "689b1a53fe3f9b99d536ab842445afbe5968747fb1acf2da3c26bf60d2153865",
            "amount": "13806779",
            "num_confirmations": 15,
            "block_hash": "0000000000000012d1921e87a652db1cb2278626ae4a635a0ac24923fad91bfe",
            "block_height": 1449764,
            "time_stamp": "1545928536",
            "total_fees": "0",
            "dest_addresses": [
                "2N5aYoFPhgKewcgACusnpJ8krMggutBFb7n",
                "2NBBoxSWdfXe3juKACA3jjvdZv16Eq5Uwcw"
            ]
        },
        {
            "tx_hash": "14b334746874fc02425cd04ccc0c07304d09530733747ff12e770bb4f384f74a",
            "amount": "-8284244",
            "num_confirmations": 5,
            "block_hash": "00000000000054aaf396c9dc592d55d87abb5506e10d4ec371dfd0acdce93077",
            "block_height": 1449774,
            "time_stamp": "1545935222",
            "total_fees": "177",
            "dest_addresses": [
                "tb1qj232zz89r44x6tefvdkmv2vnq04ulpuncpr65p",
                "tb1ql9nt9rwag9u5gvte3tr6nfmfp5ehpkgpxfd5prgamxzdp0kaul2qnr8ju6"
            ]
        },
        {
            "tx_hash": "a60a552774ed0dd771ef2d31f27e804890448f611dbe31773f7c837f09c2b2b0",
            "amount": "-4970594",
            "num_confirmations": 5,
            "block_hash": "00000000000054aaf396c9dc592d55d87abb5506e10d4ec371dfd0acdce93077",
            "block_height": 1449774,
            "time_stamp": "1545935222",
            "total_fees": "154",
            "dest_addresses": [
                "tb1qld3x2y5my5gtxqf6mv9w5xhr4zyp0gunqyq7vw",
                "tb1qzs9dnw7ggvzq3ejsvdnxtztmf7x8q8wuadhra7hyj7klk7jd0vms56p2nr"
            ]
        }
    ]
}
```

Where did my bitcoin go?

Stadicus, 2018-12-28:
You should be able to see your bitcoin commited to channels. There will be a minor difference because of on-chain fees, but in sum you should be able to see your initial amount, now split on-chain and in channels.

https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_40_lnd.md#lnd-in-action

DecisiveIndecisive, 2018-12-29:
Ah! I understand now that the autopilot feature will take 60% of the funds, commit to a channel and run that process again for a max of 5 channels... I think. I've found the remaining bitcoin in another channel and have closed my channels to see almost the entire balance returned to the original address.

closed 2018-12-29 by DecisiveIndecisive