raspibolt/raspibolt #315: EPS - Electrum unable to connect to EPS

victorscones, 2018-12-27:
I am unable to connect to EPS v0.1.6 using Electrum v3.3.2. Electrum is displaying 'Not connected' and the console window is blank.

The following is the debug.log from EPS. 

```
2018-12-27 13:45:32,467460 [  LOG] Starting electrum server
2018-12-27 13:45:32,467890 [  LOG] Listening for Electrum Wallet on ('0.0.0.0', 50002)
2018-12-27 13:45:37,298599 [  LOG] Electrum connected from ('192.168.29.176', 54366)
2018-12-27 13:45:37,302773 [DEBUG] => {"jsonrpc": "2.0", "method": "server.version", "id": 90, "params": ["3.3.2", "1.4"]}
2018-12-27 13:45:37,303896 [DEBUG] <= {"jsonrpc": "2.0", "params": ["3.3.2", "1.4"], "result": ["ElectrumPersonalServer 0.1", "0.1"], "method": "server.version", "id": 90}
2018-12-27 13:45:37,304774 [DEBUG] => {"jsonrpc": "2.0", "result": null, "id": 90}
```

Stadicus, 2018-12-28:
Working on updating this bonus guide as we speak, thanks for letting me know.

Stadicus, 2019-01-08:
Guide is updated, can you try to check if it fixes your connection issues?
https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_64_electrum.md

victorscones, 2019-01-09:
I no longer have any connection issues after using the latest update of the guide. Thanks!

closed 2019-01-09 by victorscones