raspibolt/raspibolt #316: SSH key setup only includes link for Windows

cilphex, 2018-12-28:
The section `Raspberry Pi > Login with SSH keys` requires the user to `Set up SSH keys for the "admin" user`, but the only link it provides to learn how to do that is for Windows/PuTTY. It would be helpful for it to include examples for other Mac and Linux, too.

Stadicus, 2018-12-28:
Do you have a recommondation for a good guide? What are you using?

cilphex, 2018-12-28:
I don't yet - I'm on a Mac. I'd love to do this part but might skip for now and come back.

I think on linux & mac this can all be done via command line? If so, my preference would be to just have the instructions for unix-based systems inline, with an alternative "If you're a Windows user, follow this guide..." link.

jraffile, 2019-01-01:
@cilphex In linux / mac you can use the [github ssh keygen steps](https://help.github.com/articles/connecting-to-github-with-ssh/) (as they're pretty well documented), and stick the public key in `/home/admin/.ssh/authorized_keys` on the pi. 

@Stadicus I can fork and put a PR in for updating the `raspibolt_20_pi.md` with the unix / linux steps.  It's really the least I could do to help improve an already **phenomenal** guide.

Stadicus, 2019-01-02:
Hej @jraffile, I'd appreciate that very much, thanks! I think it's important to verify that all external guides end with the same state / ssh keyfiles, or we need to make three short sections per OS (link to external guide + after-guide steps).

Stadicus, 2019-01-08:
Instructions by @jraffile have been merged! 💯

closed 2019-01-08 by Stadicus