raspibolt/raspibolt #318: Docs missing symlink

cilphex, 2018-12-28:
When following this guide, I found I needed an additional step that isn't documented.

After unlocking `lnd`, I was getting the following error:

```
2018-12-28 11:34:08.314 [ERR] SRVR: Unable to discover a NAT-PMP enabled device on the local network: fork/exec /usr/bin/ip: no such file or directory
```

The fix was to add a symlink to `/bin/ip`:

```
sudo ln -s /bin/ip /usr/bin/ip
```

Now I get a different error:

```
2018-12-28 12:10:12.128 [ERR] SRVR: Unable to discover a NAT-PMP enabled device on the local network: multiple NATs detected
```

But at least the symlink fixes that first case...

Stadicus, 2018-12-29:
Thanks, I just updated the guide, can you please check if everything works as intended for you? 

For LND to detect the external ip address and configure port forwarding, your router must have UPnP enabled, see explanation directly below `lnd.conf`.

hgedia, 2019-01-02:
This fails for me as user bitcoin is not in the list of sudoer's.