raspibolt/raspibolt #320: Updating LND

jodobear, 2018-12-29:
This is more of an anomaly than an issue:

I just updated my LND from `v0.5` to `v0.5.1`. LND is running and i can normally operate lncli. I don't have any channels open so can't comment on that. The anomaly is:

The screen shows `LND v0.5-beta` while output of `lncli --version` is `lncli version 0.5.1-beta commit=`

I think while updating lnd was running. Can this be the cause? Should i clean install lnd? If so, how do i remove lnd from my pi?

Thank you!

Stadicus, 2018-12-29:
I did not try LND 0.5.1 as the changes are minimal and not really relevant for this Beginner's Guide. What "screen" do you mean?

jodobear, 2018-12-31:
I mean, i have the 3.5" lcd screen attached to my Pi. Also, to be clear i'm using @rootzoll's Raspiblitz SD image for my Pi and since it's heavily based on your work and refers to your instructions, i posted the question here as even i follow your guides. Apologies if i shouldn't have posted the question here..

jodobear, 2018-12-31:
Also, a lot of people have been facing issues with LND 0.5.1 update with lost balances and closed channels, i wanted to update first before i seriously start opening channels and routing.

Septem151, 2019-01-01:
I've already created an issue with the lnd repository here: https://github.com/lightningnetwork/lnd/issues/2389