raspibolt/raspibolt #322: Error creating wallet on LND Mainnet

DecisiveIndecisive, 2018-12-30:
When creating a wallet with my original 24word mnemonic I receive the following error: `[lncli] rpc error: code = Unknown desc = wrong seed version`

I successfully created a new wallet using the same password I used last time, as suggested however after restarting and trying to unlock the wallet I I receive this error `[lncli] rpc error: code = Unknown desc = wallet not found`

What does the first error mean, and is the second error caused by a mismatched password like you warned against?

Stadicus, 2019-01-01:
It actually should not matter anymore (since LND 0.5) how you create the mainnet wallet. Are you moving from testnet to mainnet? If so, just creating a new wallet should work. 

When writing "created a new wallet using the same password" you are referring to the wallet password, not the cipher passphrase, right? I usually do not use an additional passphrase, but the password is mandatory.

If you **do not have any coins** in the current (not found) mainnet wallet and **no open mainnet channels**, you might just want to delete the mainnet data and create a new wallet:
`rm /home/bitcoin/.lnd/data/chain/bitcoin/mainnet/*`

DecisiveIndecisive, 2019-01-06:
Yes I was referring to the wallet password, but I also tried import with my original cipher and that's what through the first error I mentioned. 

I can not run the command you mentioned, I get the following error `rm: cannot remove '/home/bitcoin/.lnd/data/chain/bitcoin/mainnet/': Is a directory`

I'm struggling to find an answer to this problem outside of posting to public channels :(

Stadicus, 2019-01-06:
`rm /home/bitcoin/.lnd/data/chain/bitcoin/mainnet/*` should delete all files **within** that directory, did you maybe omit the `*`?

DecisiveIndecisive, 2019-01-06:
Sorry, I meant to include the asterisk in my post. I get this error:
`rm: cannot remove '/home/bitcoin/.lnd/data/chain/bitcoin/mainnet/*': No such file or directory`

DecisiveIndecisive, 2019-01-06:
I'm happy to move this conversation to Slack LND Devs. I've gone through the files with WinSCP and can't find anything out of ordinary in there either...

Stadicus, 2019-01-08:
If you do not have any coins in LND yet, you can also just delete the whole LND folder and start over. That should completely wipe all LND application data.
`rm -rf /home/bitcoin/.lnd/*`

closed 2019-01-29 by Stadicus