raspibolt/raspibolt #323: Mount HD alternative instruction

Nursing-Guy, 2018-12-31:
I have partially completed the guide a couple of months ago and had to restart from scratch for life reasons and discovered that I had no idea how to mount the hard drive without reformatting it (and honestly can't find it using -lsblk -o ETC,ETC,ETC in it's current incarnation.)

I am referring to https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_20_pi.md

>Mounting external hard disk
To store the blockchain, we need a lot of space. As a server installation, the Linux native file system Ext4 is the best choice for the external hard disk, so we will format the hard disk, erasing all previous data. The external hard disk is then attached to the file system and can be accessed as a regular folder (this is called “mounting”). 

>:warning: **Previous data on this hard disk will be deleted!**

>* Plug your hard disk into the running Pi and power the drive up. 

>* Get the NAME for main partition on the external hard disk  
>  `$ lsblk -o UUID,NAME,FSTYPE,SIZE,LABEL,MODEL` 

>* Format the external hard disk with Ext4 (use [NAME] from above, e.g `/dev/sda1`)  
  `$ sudo mkfs.ext4 /dev/[NAME]`

>* Copy the UUID that is provided as a result of this format command to your local (Windows) notepad. 

>* Edit the fstab file and the following as a new line (replace `UUID=123456`) at the end  
  `$ sudo nano /etc/fstab`  
  `UUID=123456 /mnt/hdd ext4 noexec,defaults 0 0` 

>* Create the directory to add the hard disk and set the correct owner  
  `$ sudo mkdir /mnt/hdd`

>* Mount all drives and check the file system. Is “/mnt/hdd” listed?  
  `$ sudo mount -a`  
  `$ df /mnt/hdd`

>Filesystem     1K-blocks  Used Available Use% Mounted on
/dev/sda1      479667880 73756 455158568   1% /mnt/hdd

>*  Set the owner  
  `$ sudo chown -R bitcoin:bitcoin /mnt/hdd/`

>* Switch to user "bitcoin", navigate to the hard disk and create the bitcoin directory.  
  `$ sudo su - bitcoin`  
  `$ cd /mnt/hdd`  
  `$ mkdir bitcoin`  
  `$ ls -la`

>* Create a testfile in the new directory and delete it.  
  `$ touch bitcoin/test.file`  
  `$ rm bitcoin/test.file`

>* Exit the "bitcoin" user session  
  `$ exit` 

>If this command gives you an error, chances are that your external hard disk is mounted as “read only”. This must be fixed before proceeding. If you cannot fix it, consider reformatting the external hard disk.

>👉 additional information: [external storage configuration](https://www.raspberrypi.org/documentation/configuration/external-storage.md)

So obviously I have formatted this hard drive before. I thought maybe I will have to reformat it anyways as I've gotten as far as "###Hardening your pi" on a previous incarnation and don't know if previous formatting will work flawlessly with essentially a new device.

Unfortunately when I type  `$ lsblk -o UUID,NAME,FSTYPE,SIZE,LABEL,MODEL` I don't get the HD UUID. Should I attach this to my PC and reformat it and start it from scratch as well? I would have to redownload the testnet blockchain but I will survive.

Nursing-Guy, 2018-12-31:
Well after attaching the HD to my laptop by USB I have no idea how to reformat it, it doesn't show up as a storage device on my windows 10 device and I can only find it in "Devices and Printers"

Stadicus, 2018-12-31:
If you connect your hdd to the Pi, it should show up with `lsblk -o UUID,NAME,FSTYPE,SIZE,LABEL,MODEL`, even when not mounted, formatted or not. Just skip the formatting and use the UUID provided by this command for the fstab entry.

I do this all the time, no need to download the blockchain again (important for mainnet and all LND channel information).

If it does not show up, there's something wrong with the hardware (good usb cable? enough power? faulty hdd?)

Nursing-Guy, 2018-12-31:
You're right Stadicus. I'm not sure at what point they started talking to each other but after a few resets of each piece of hardware everything started working fine.

closed 2018-12-31 by Nursing-Guy

Stadicus, 2019-01-01:
Great to hear. 👍