raspibolt/raspibolt #326: lncli --network=testnet np2wkh rpc error

Nursing-Guy, 2018-12-31:


Stadicus, 2019-01-01:
The error above does not concern `bitcoind` but the gRPC API that is exposed by LND itself and can be used to connect to LND, eg. as a wallet or web interface.

* are you running additional software that connects to LND on this port?
* is your port opened in UFW?

Nursing-Guy, 2019-01-02:
This is a fresh install of LND on a device I wiped clean. Unless the guide is installing LND on the external hard drive and it didn't get written over on the fresh install.

I'm looking into UFW right now, but the command I'm calling is `lncli --network=testnet np2wkh`

Nursing-Guy, 2019-01-02:
I've left the program running for the past two days since I've been busy with other tasks. I just attempted to recreate the error and received the proper response. I'm uncertain of the cause, but I suspect that either the lightning network or the bitcoin client weren't caught up to the network. I believe the guide shows how to check this, I likely just misinterpreted the results of the calls.

closed 2019-01-02 by Nursing-Guy

Nursing-Guy, 2019-01-02:
I now have a working LND node. I have to get to work but I'm excited to get some blockachinos. I've learned a lot, I am super thankful to you Stadicus for writing this guide and being so helpful with this Linux and LND newcomer. I've learned a lot about how much more there is to learn and it will probably be a little while before I get #Reckless