raspibolt/raspibolt #328: Tor V3 Onion on v 0.99

zer0trst, 2019-01-01:
Hello to all and Happy New Year!

Hoping to get some assistance on this one. I am trying to run through TOR v3 onion service. I have Setup my Raspiblitz today using the latest image (raspiblitz-0.99-2018-12-24.img.gz)

I tried the new option of setting from the main menu:  Go to SERVICES and Run behind TOR
[image]

It gives me this screen for a split second and then takes me back to the main menu. From there it does nothing further.

[image]

I rebooted in the hopes it would do something but still nothing..

So I went and Followed the steps outlined here: https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_69_tor.md

added sources and installed tor-arm...

Added "bitcoin" to the group "debian-tor"

      $ cat /etc/group | grep debian-tor
         debian-tor:x:123:bitcoin

un-commented and added in "torrc" file

      $ sudo nano /etc/tor/torrc

[image]

Added lines to "bitcoin.conf" file

      $ sudo nano /home/bitcoin/.bitcoin/bitcoin.conf

[image]

Archived log file and restarted bitcoin core

      $ sudo mv /home/bitcoin/.bitcoin/debug.log /home/bitcoin/.bitcoin/debug.log.old
      $ sudo systemctl start bitcoind

ran the verification 

      $ sudo tail /home/bitcoin/.bitcoin/debug.log -f -n 200

[image]
[image]

and 

      $ bitcoin-cli getnetworkinfo

[image]

Everything looks right i think? but here's where I run into a brick wall....

I run 

    $ curl icanhazip.com

and my IP shows up:

[image]

Now when I run 

      $ bitcoin-cli getpeerinfo | grep  local

my IP is not included but my onion address is...

[image]

So i then go through the steps for LND as follows: 

      $ sudo nano /home/bitcoin/.lnd/lnd.conf

[image]

and when I run

      lncli getinfo 
      lncli getnodeinfo [YOUR_PUBKEY]

My IP is exposed here as well

[image]
[image]

So final thing I have tried is adding these lines to "/etc/tor/torrc" as someone on reddit had pointed out a few months back

[image]

then I grabbed the address out of /var/lib/tor/lnd_service/hostname and added it to /home/bitcoin/.lnd/lnd.conf in "externalip"

not sure i've put this in the right place or not??

[image]

This seems to have done something because after waiting about 10 minutes, when I run 

     lncli getinfo

Under URI's I now get both my public key against my exposed IP but also my public key against what appears to be a V3 onion address but it's not the address in /var/lib/tor/lnd_service/hostname or what I put under "externalip" in lnd.conf

[image]
[image]

[image]

so i'm stuck here..

any ideas? something I've missing along the way perhaps?

Stadicus, 2019-01-01:
Wow, I think this is the first issue that forced me to work in parallel windows... :-)

**Tor on Bitcoin Core**
seems to work perfectly well
* Tor is running
* onion address is created and used to connect to Bitcoin p2p network
* Bitcoind traffic for ipv4 & ipv6 is routed over Tor proxy

It is perfectly normal that your IP is exposed when you call a normal internet address like icanhazip.com, that has nothing to do with Tor or Bitcoin. This command is only used to get the reference clearnet address (I should make that more clear in the guide). The fact that you cannot find your IP address in "getpeerinfo" verifies that bitcoind does not use your clearnet address.

**Tor for Lightning**
there might be an issue
* the commands listed should suffice to completely deactivate that your node is announced on clearnet. I can reliably switch between clearnet & Tor by just changing the lnd.conf.
* additional steps like you described should not be necessary.
* did you restart LND?

zer0trst, 2019-01-02:
Hey Stadicus thanks for getting back to me. 

Had an inkling that the the icanhazip report was right thanks for confirming. 

I did restart LND but nothing has changed. I am going to format my SD, restart all over again and see if anything changes just in case there was something I did wrong somewhere. I have also shared this over at rootzoll/raspiblitz to see if anyone there has any ideas

I'll report my findings

Thanks for your help! :)

Stadicus, 2019-01-02:
I guess, you also see your clearnet address with `lncli getinfo`? This should definitely not be the case. Please check with the RaspiBlitz project why that could be. In my experience, the changes highlighted in my guide work reliably, but I might not be aware of specific configurations that RaspiBlitz implementend.

Closing for now (cannot support other projects), but would appreciate feedback if you figure it out.

closed 2019-01-02 by Stadicus

zer0trst, 2019-01-03:
@Stadicus thanks for your help. 

To save anyone else the confusion , I ended up getting a resolve:

In the Raspiblitz 0.99 menu under Services you have to press space-bar on "Run behind TOR" to make your selection then hit enter and it runs the tor setup script.

I wasn't hitting space bar to make the selection so subsequently nothing would happen when I hit enter... (Facepalm :P) - 

This will only run Onion V2 service though. Now when I run 

        lncli getnodeinfo >pub key< 

I only get my onion address - IP doesnt show up at all.

Thanks! :)