raspibolt/raspibolt #336: Remove txindex

Fillippone, 2019-01-07:
I commented on the pull request from @Stadicus, but i got no answer, so posting here because maybe nobody cares about merged pull requests. 
Could you please explain what did you do here? Why did you remove the txindex=1 option? Txindex=1 is needed to run a full node, isn’t it?

Cees2daBees, 2019-01-07:
Hi @Fillippone, Stadicus tweeted recently about transaction index no longer being needed to run a full node. Link below. I'm guessing this means txindex=1 is no longer required. However, I haven't changed my config yet to remove it.

https://twitter.com/Stadicus3000/status/1079677441956868097

Fillippone, 2019-01-07:
Thanks for the answer. So it was Bitcoin Core v. 17.0 that enables that. Only not sure how are the performances of the indexing process on a Raspberry PI device: how long is it going to take? Two weeks? And what about LND?

saubyk, 2019-01-07:
I reindexed on Raspberry pi. It took about 6 hours to complete. LND synced back in about 20 mins or so.

Fillippone, 2019-01-07:
Mhh. It took mine two weeks. How’s that?

Fillippone, 2019-01-08:
Thank you @Stadicus for the clear explanation. 
On the Bitcoin Core 17.0 release notes i read: "... If bitcoind is run with -txindex on a node that is already partially or fully synced without one, the transaction index will be built in the background and become available once caught up. "
So I understand that the transaction index is built on the background. IF you activate the txindex=1 on a Raspberry PI the reindexing of the whole chain can take several weeks to complete. IS that "background process" of similar length? Am I correct? did you actually test how long does it takes? (can you do that by 'bitcoin-cli getblockchaininfo'?

Stadicus, 2019-01-08:
It's new that it builds up the txindex in the background. Before that, bitcoind was not operational during the reindexing.

closed 2019-01-08 by Stadicus