raspibolt/raspibolt #338: Visible from outside

badokun, 2019-01-09:
https://github.com/Stadicus/guides/blob/2743596230618e141cd51adecb552af5115ea0f1/raspibolt/resources/20-raspibolt-welcome#L122

Not sure what this line does, but I get `Not reachable`. When I use an external website to confirm whether the port is open it seems to be fine - http://www.canyouseeme.org/

Here's my ufw setup, and I also confirmed it's setup correctly on my router with a port forward
```
Status: active

To                         Action      From
--                         ------      ----
22                         ALLOW       192.168.1.0/24             # allow SSH from local LAN
Anywhere                   ALLOW       192.168.1.0/24 1900/udp    # allow local LAN SSDP for UPnP discovery
9735                       ALLOW       Anywhere                   # allow Lightning
8333                       ALLOW       Anywhere                   # allow Bitcoin mainnet
18333                      ALLOW       Anywhere                   # allow Bitcoin testnet
Anywhere                   ALLOW       192.168.1.0/24/udp         # allow local LAN SSDP for UPnP discovery all UPD Open
9735 (v6)                  ALLOW       Anywhere (v6)              # allow Lightning
8333 (v6)                  ALLOW       Anywhere (v6)              # allow Bitcoin mainnet
18333 (v6)                 ALLOW       Anywhere (v6)              # allow Bitcoin testnet

```

ejose19, 2019-01-09:
Probably the same issue as mine, router lacking NAT Loopback functionability, see:

https://github.com/Stadicus/guides/issues/79#issuecomment-449058366

You have 2 options, revert script to bitnodes check or change $public_port variable to obtain ipv6 instead of ipv4 (with ipv6 I didn't had any issue)

badokun, 2019-01-10:
Thanks will take a look at those options. My router is a ZTE Japanese one so most likely lacking in industry standard features

badokun, 2019-01-10:
I tried running this command
`$ curl  https://bitnodes.earn.com/api/v1/nodes/me-18333/`
but got 
`{"success":false}`

Interestingly doing the same on my Windows machine on the same local network returns `{"success":true}`

I disabled ufw altogether to see if that's the issue but it still fails.

ejose19, 2019-01-10:
If on the other computer it returns success:true I doubt it's an issue with ufw. Any way I can help you troubleshoot this?

On mine the command you pasted works properly

closed 2019-02-21 by Stadicus