raspibolt/raspibolt #341: Sudo not working

isaactyc, 2019-01-11:
Hello. After about a week of leaving my node alone without much change, I opened Putty to make some changes to my node. I found that calling sudo gave me the following errors 

sudo: unable to stat /etc/sudoers: Structure needs cleaning
sudo: no valid sudoers sources found, quitting
sudo: unable to initialize policy plugin

I googled and I want to try [this solution](https://raspberrypi.stackexchange.com/questions/68995/sudo-error-structure-needs-cleaning), but in order to unmount the external hard drive for the raspibolt, i have to disable bitcoind and lnd, both of which require sudo. Which doesn't work. So I'm a little stuck here, can I just go ahead and try the solution? My concern is that I risk damaging the file system further without any backup

isaactyc, 2019-01-12:
Hmmm I can't even switch to the bitcoin user. Using "sudo su - bitcoin" gives me the same error messages. Essentially anything that involves sudo is kinda stuck 😢

Stadicus, 2019-01-20:
Seems like file system corruption, but not on the external hdd, but the sdcard. I don't think you can fix that from the Pi, as you cannot unmount the rootfs while running from it. 

Before fiddling around for hours, I'd just set up the sdcard from scratch (maybe with a different card, if it is a hardware defect). All important stuff is on the hdd, so you just need to configure the os and install the applications.

isaactyc, 2019-01-25:
Hi Stadicus, sorry for the late reply (work has been busy). Does that mean I'll just turn off the pi? My concern is with it corruption anything else. So once I've configured the os, I don't have to go through the trouble of reinstalling and re syncing with the blockchain right?

Stadicus, 2019-01-25:
Correct. I did that this week myself, it works without issue as everything important is on the external hdd. Just install all the software, set the systemd units and create the symbolic links .bitcoin & .lnd. No need to download the blockchain again.

isaactyc, 2019-01-28:
Ok, will do! So will I have to close all my channels before I can do this, or will powering down the pi straightaway be ok? (sorry, a little unsure and I'm trying to be careful with the bitcoins in the channel haha)

Stadicus, 2019-01-29:
Yeeeeaaaaah...... in theeeeory it works perfectly fine. If something goes wrong, however, your money in the channels will be lost. So I'd err on the safe side and close the channels (Bitcoin transactions are a few cents at the moment). Otherwise: don't blame me.

isaactyc, 2019-02-03:
No worries, only "blame" you get is making me even more interested in bitcoin and lightning. :) 

Welp, things just seem to get worse and worse. my putty couldn't access the pi at all, the window would close the moment i typed 'admin'. so can't even close my channels if i wanted to :( anyway, I've taken the plunge and powered down my pi, sd card is not working after multiple tries, so i got a new one. 

Can I ask, does that mean I'll follow the "Raspberry Pi" section of the Raspibolt guide? I find that I am unable to access the 'admin' account any longer, so does that mean to authorization keys I created has to be re-created?

isaactyc, 2019-02-03:
Also, I'm not too sure what you mean by "set the systemd units" What units is it I have to set? And also, under "mounting the external hard disk, there's a warning that says "previous data on this hard drive will be deleted!", how would i mount it back?

Stadicus, 2019-02-04:
Yes, with the new sdcard, you need to set up the operating system from scratch and follow the guide.

* Ignore the "format external hdd" part, just mount the existing one.
* If you still have your private & public keys for ssh, you can just add this to the "authorized_keys" file again, no need to recreate them.
* Install bitcoin & lnd again, but after creating the symbolic link to the external hdd (with `ln -s ...`) the necessary directories & config files are already there, just check them.
* You also need to recreate the "systemd units" (the `.service` files) for the serives to startup automatically.

isaactyc, 2019-02-05:
Finally finally got everything up and running again, at the moment it seems like no bitcoins were lost. Thank you Stadicus!! :)

closed 2019-02-05 by isaactyc