raspibolt/raspibolt #343: Cannot verify keys of torproject / installed version not uptodate

raindogdance, 2019-01-12:
I'm trying to set up Tor, following the "bonus" part of the guide.  

After doing `sudo apt install dirmngr` (see the output of which [here](https://pastebin.com/py1HevGC)), the next step in the guide is verifying the keys with `gpg --keyserver keys.gnupg.net --recv A3C4F0F979CAA22CDBA8F512EE8CBC9E886DDD89`. 
But it gives out `gpg: keyserver receive failed: No data`  

Next step in guide is `gpg --export A3C4F0F979CAA22CDBA8F512EE8CBC9E886DDD89 | sudo apt-key add -` , which gives out `gpg: WARNING: nothing exported
gpg: no valid OpenPGP data found.` 

I still proceeded to install Tor (with $ sudo apt update and $ sudo apt install tor tor-arm), but checking the version it said `Tor version 0.2.9.16 (git-9ef571339967c1e5)` (should have been 0.3.3.6 or newer, acc to guide), so I didn't proceed any further.  

Although I set up my node with an older version of the guide, I do have a Raspberry Pi 3 and it runs on Raspbian Stretch Lite (which are both requirements, acc to guide).   

Appreciate the help!

Stadicus, 2019-01-20:
I can reproduce the error. Try `gpg --recv A3C4F0F979CAA22CDBA8F512EE8CBC9E886DDD89`, that should work.
https://www.torproject.org/docs/debian.html.en#packages

Stadicus, 2019-01-20:
Did you correctly add the custom Tor repository and run `apt update`?

* `$ dpkg -s tor` shows installed version:
```
Package: tor
Status: install ok installed
Priority: optional
Section: net
Installed-Size: 3611
Maintainer: Peter Palfrader <weasel@debian.org>
Architecture: armhf
Version: 0.3.4.9-1~d90.stretch+1
```
* `apt-cache showpkg tor` shows available packages in repositories. Version 0.2.9 comes from the standard Raspbian, I guess that's the issue.
```
Package: tor
Versions: 
0.3.4.9-1~d90.stretch+1 (/var/lib/apt/lists/deb.torproject.org_torproject.org_dists_stretch_main_binary-armhf_Packages) (/var/lib/dpkg/status)
 Description Language: 
                 File: /var/lib/apt/lists/raspbian.raspberrypi.org_raspbian_dists_stretch_main_binary-armhf_Packages
                  MD5: 9106e8e4b3843ebd532ba2b89615a92e
 Description Language: 
                 File: /var/lib/apt/lists/deb.torproject.org_torproject.org_dists_stretch_main_binary-armhf_Packages
                  MD5: 9106e8e4b3843ebd532ba2b89615a92e

0.2.9.16-1 (/var/lib/apt/lists/raspbian.raspberrypi.org_raspbian_dists_stretch_main_binary-armhf_Packages)
 Description Language: 
                 File: /var/lib/apt/lists/raspbian.raspberrypi.org_raspbian_dists_stretch_main_binary-armhf_Packages
                  MD5: 9106e8e4b3843ebd532ba2b89615a92e
 Description Language: 
                 File: /var/lib/apt/lists/deb.torproject.org_torproject.org_dists_stretch_main_binary-armhf_Packages
                  MD5: 9106e8e4b3843ebd532ba2b89615a92e
```

raindogdance, 2019-01-20:
Couldn't fix the issue, still the same as in OP. I hope the provided output of the commands give some insights (sorry if it's messy).

The custom repository seems to be added correctly in the `sources.list` file. The file looks like this:  

> deb http://mirrordirector.raspbian.org/raspbian/ stretch main contrib non-free rpi
# Uncomment line below then 'apt-get update' to enable 'apt-get source'
#deb-src http://archive.raspbian.org/raspbian/ stretch main contrib non-free rpi

> deb https://deb.torproject.org/torproject.org stretch main
deb-src https://deb.torproject.org/torproject.org stretch main  

I also did `apt update`, but it gives out some error messages:  

> Err:5 https://deb.torproject.org/torproject.org stretch InRelease                                           
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 74A941BA219EC810
Reading package lists... Done                                                                               
W: GPG error: https://deb.torproject.org/torproject.org stretch InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 74A941BA219EC810
E: The repository 'https://deb.torproject.org/torproject.org stretch InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.  

Output of `dpkg -s tor` is:  
>Package: tor
Status: install ok installed
Priority: optional
Section: net
Installed-Size: 3822
Maintainer: Peter Palfrader <weasel@debian.org>
Architecture: armhf  

And output of `apt-cache showpkg tor` is:  

>Package: tor
Versions: 
0.2.9.16-1 (/var/lib/apt/lists/mirrordirector.raspbian.org_raspbian_dists_stretch_main_binary-armhf_Packages) (/var/lib/dpkg/status)
 Description Language: 
                 File: /var/lib/apt/lists/mirrordirector.raspbian.org_raspbian_dists_stretch_main_binary-armhf_Packages
                  MD5: 9106e8e4b3843ebd532ba2b89615a92e

The command `gpg --recv A3C4F0F979CAA22CDBA8F512EE8CBC9E886DDD89` gives out: 

> gpg: keyserver receive failed: No data

Stadicus, 2019-01-21:
It all comes down to not being able to import the Tor signing keys first. Without that, you cannot receive packages from the repository, so you are stuck with the old "official" version of Raspbian.

What's your `gpg` version?
```
admin@RaspiBolt:~ $ gpg --version
gpg (GnuPG) 2.1.18
libgcrypt 1.7.6-beta
```

raindogdance, 2019-01-21:
Seems to be the same as yours:  

> $ gpg --version
gpg (GnuPG) 2.1.18
libgcrypt 1.7.6-beta

Stadicus, 2019-01-29:
Just tried again, now it works the other way round... seems to be a tricky subject.
```
admin@odroid:/$ gpg --recv A3C4F0F979CAA22CDBA8F512EE8CBC9E886DDD89
gpg: keyserver receive failed: No data
admin@odroid:/$ gpg --keyserver keys.gnupg.net --recv A3C4F0F979CAA22CDBA8F512EE8CBC9E886DDD89
gpg: key EE8CBC9E886DDD89: 2 duplicate signatures removed
gpg: key EE8CBC9E886DDD89: 82 signatures not checked due to missing keys
gpg: key EE8CBC9E886DDD89: "deb.torproject.org archive signing key" not changed
gpg: Total number processed: 1
gpg:              unchanged: 1
```

raindogdance, 2019-01-29:
I just tried again and it works, although I changed absolutely nothing since the last message here :) 

`gpg --export A3C4F0F979CAA22CDBA8F512EE8CBC9E886DDD89 | sudo apt-key add -
key EE8CBC9E886DDD8` now gives out:  

> 82 signatures not checked due to missing keys
OK  

which I assume is a good result?  
In any case, I could proceed with the installation now and `tor --version` now shows 0.3.5.7.  

So I will proceed with further set up and I guess this issue solved itself for whatever reason :) 
Probably the problem was somewhere on the TOR repository side or something?

raindogdance, 2019-01-31:
Just reporting that I managed to set up everything :)  

Unrelated to the original issue, but don't want to open a new one (yet):  

So the command `bitcoin-cli getpeerinfo | grep  local` doesn't show my clear IP anymore, only my onion address and some random IP addresses. So I assume my node is being reached under its onion address (and not under clear IP anymore).  

But when I go to https://bitnodes.earn.com/ and put in my onion address + 8333, it says "unreachable". Also, when I go to https://canyouseeme.org, it says that my port 8333 is not opened (for my clear IP), although I haven't changed anything in router settings or anywhere else that might concern that (ports 8332 and 9735 are also shown as open).  

So I'm a bit confused - is my node still reachable (given that I have peers connecting to me through my onion address) or not?

Amiga500, 2019-02-03:
I have the same issue above:

"the command bitcoin-cli getpeerinfo | grep local doesn't show my clear IP anymore, only my onion address and some random IP addresses. So I assume my node is being reached under its onion address (and not under clear IP anymore).

But when I go to https://bitnodes.earn.com/ and put in my onion address + 8333, it says "unreachable". Also, when I go to https://canyouseeme.org, it says that my port 8333 is not opened (for my clear IP)"

raindogdance, 2019-02-03:
In another issue someone reports that bitnodes has issues with onion addresses: https://github.com/Stadicus/guides/issues/365#issuecomment-459482273 

So maybe it's just something on bitnodes' end. On the other hand, shouldn't my port 8333 on
https://canyouseeme.org/ show as "open"? Or is it ok if it shows "closed", because my node doesn't listen on this port under my clear IP anymore?

larry-beck, 2019-02-03:
Hi, I am having same type problem, but get the following when issuing the first gpg command:

admin ~ ฿ gpg --keyserver keys.gnupg.net --recv A3C4F0F979CAA22CDBA8F512EE8CBC9E886DDD89
gpg: packet(13) too large
gpg: read_block: read error: Invalid packet
gpg: no valid OpenPGP data found.
gpg: Total number processed: 0

Any help?

larry-beck, 2019-02-03:
admin ~ ฿ gpg --version
gpg (GnuPG) 2.1.18
libgcrypt 1.7.6-beta

Stadicus, 2019-02-04:
@larry-beck what key server did you use? The current set seems notoriously unreliable, need to add a better one to the guide...

Stadicus, 2019-02-04:
Regarding bitnodes and checking public availability: see this comment here
https://github.com/Stadicus/guides/issues/189#issuecomment-449638447

Please make sure that port 8332 is closed, this is sensitive and allows everybody to remotely control your bitcoind.

raindogdance, 2019-02-04:
@Stadicus  - thank you! 

I can confirm that putting my node onion address + port like this (as in your linked comment):   https://bitnodes.earn.com/nodes/[nodeonionaddress]-8333/    seems to show my node as reachable. Good news :) 

And thanks for the advice with the 8332 port. I don't think it's mentioned in the guide to close it after switching to mainnet, so perhaps it would be good to add this advice somewhere.

larry-beck, 2019-02-04:
> @larry-beck what key server did you use? The current set seems notoriously unreliable, need to add a better one to the guide...

I tried these four, I believe that number 3 finally worked and allowed me to download the keys in question for accessing the tor update servers.

1. hkps.pool.sks-keyservers.net
2. keys.gnupg.net
3. pgp.uni-mainz.de
4. pgp.mit.edu

janglorumo, 2019-02-14:
I had the same issue as OP. Worked around it.
"After doing `sudo apt install dirmngr` (see the output of which [here](https://pastebin.com/py1HevGC)), the next step in the guide is verifying the keys with `gpg --keyserver keys.gnupg.net --recv A3C4F0F979CAA22CDBA8F512EE8CBC9E886DDD89`.
But it gives out
> gpg: keyserver receive failed: No data.

Next step in guide is `gpg --export A3C4F0F979CAA22CDBA8F512EE8CBC9E886DDD89 | sudo apt-key add -` , which gives out
> gpg: WARNING: nothing exported
> gpg: no valid OpenPGP data found."

I have same issue here as well. 
"the command bitcoin-cli getpeerinfo | grep local doesn't show my clear IP anymore, only my onion address and some random IP addresses. So I assume my node is being reached under its onion address (and not under clear IP anymore).

But when I go to https://bitnodes.earn.com/ and put in my onion address + 8333, it says "unreachable".

mjslawson, 2019-02-17:
Has anyone else received the error:

> gpg: packet(13) too large
> gpg: read_block: read error: Invalid packet
> gpg: no valid OpenPGP data found.
> gpg: Total number processed: 0

I've tried all @larry-beck recommended key servers and @Stadicus recommended command without specifying keyserver:

> gpg --recv A3C4F0F979CAA22CDBA8F512EE8CBC9E886DDD89

gitro83, 2019-02-21:
@mjslawson yes, I've got the same error when using the  "pgp.uni-mainz.de" server, for all the other servers It returns the "gpg: keyserver receive failed: No data"-message.
It still doesn't work.

dschoel, 2019-02-22:
Experiencing the same issue as @mjslawson and @gitro83 
One progressed past the `gpg: keyserver receive failed: No data` message, but produced this:
`admin@raspberrypi:~ $ gpg --keyserver pgp.uni-mainz.de --recv A3C4F0F979CAA22CDBA8F512EE8CBC9E886DDD89
gpg: packet(13) too large
gpg: read_block: read error: Invalid packet
gpg: no valid OpenPGP data found.
gpg: Total number processed: 0`

justinmoon, 2019-02-24:
[These instructions from the Tor website[(https://www.torproject.org/docs/debian.html.en#ubuntu) cover an alternative method to get the key

```
$ curl https://deb.torproject.org/torproject.org/A3C4F0F979CAA22CDBA8F512EE8CBC9E886DDD89.asc | gpg --import
$ gpg --export A3C4F0F979CAA22CDBA8F512EE8CBC9E886DDD89 | apt-key add -
```

Instead of 

```
$ gpg --keyserver keys.gnupg.net --recv A3C4F0F979CAA22CDBA8F512EE8CBC9E886DDD89
$ gpg --export A3C4F0F979CAA22CDBA8F512EE8CBC9E886DDD89 | sudo apt-key add -
```

At least I think this does the same thing?

gitro83, 2019-02-24:
Hi,

Thank you! That seems to work for me!

dschoel, 2019-02-24:
This also worked for me.  Thanks.

Stadicus, 2019-03-03:
Updated the guide, thanks @justinmoon! 
https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_69_tor.md

closed 2019-03-03 by Stadicus