raspibolt/raspibolt #345: lnd.service main process exited, code

paintufast, 2019-01-15:
So I have been working just fine on the mainnet for a few days. I decided the RaspiBolt is ready to move to its more permanent location in my home. I shut it down, moved it and turned it back on, now this is happening. I am at a 100% total loss...

It will ask for wallet unlock, then immediately fails after being unlocked. When I try to unlock again, it repeats...Any ideas?

[image]

paintufast, 2019-01-15:
Ok so this happened multiple times to me (enough that I needed to come here). Then after sitting for about 15 minutes while I was posting I tried one more time out of sheer disgust...and it fired right up. 

I am still open to input on what to check for or look out for in case this happens again. Thanks all.

paintufast, 2019-01-15:
Further information - Now multiple times this has happened post restart, however if I just wait about 10 minutes and then try it works fine.

Septem151, 2019-01-19:
Error code 28 in bitcoind is the status code for "Still warming up" meaning your bitcoind hasn't finished booting up or syncing. When you first start up your pi, or start the bitcoind service, it is loading bitcoind. You can observe the status of this by checking the log file
```$ sudo tail -f /home/bitcoin/.bitcoin/debug.log```
You have to wait for this process to finish and start printing along the lines of 
```UpdateTip: new best=00000000000000000001708576de1406e4ec9a57a778390a48790a18c6bdb6c7 height=559125 version=0x20400000 log2_work=90.253607 tx=374561593 date='2019-01-19T06:15:41Z' progress=1.000000 cache=5.1MiB(15500txo) warning='25 of last 100 blocks have unexpected version'```
before unlocking your lnd wallet.

Stadicus, 2019-01-20:
Yes, the LND startup process can fail several times until bitcoind is ready. Usually, it does not take that long, but although its quite hacky there's no harm in it.

This is also the reason why the auto-unlock script (in bonus section) checks the uptime to see if the Pi has been rebooted (wait 180s), or only LND service is being restarted (wait only 10s). You can increase the wait time, if these values do not work for you.

closed 2019-01-21 by paintufast