raspibolt/raspibolt #346: No space left on device

mcgotts, 2019-01-16:
Hi there - I set up my raspibolt just a little over a week ago as per this guide.  Since then, it appears my /dev/root directly has ballooned up to ~30G and is now full.  

Upon investigation, it appears as though these log files are the culprits:
- daemon.log at 13G
- syslog.l at 3.8G
- syslog at 3.8G
- daemon.log.1 at 3.4G

Are these log files supposed to be so big?  If so - is there an easy way to move them onto my /mnt directory without affecting the operation of my LND node?

badokun, 2019-01-17:
Can't answer your question directly as it relates to moving it to your /mnt directory, but if you want insights into what's going on on your Raspberry I've put together a Monitoring guide under the Bonus section. The pull request is still open, but in the mean time you can look at https://github.com/badokun/guides/blob/master/raspibolt/raspibolt_71_monitoring.md. If you scroll down to the bottom you can get an idea of the graphs you'll get. Disk usage is also in there

Stadicus, 2019-01-20:
No, these logs are not supposed to be that big. Usually, they are rotaded, compressed and deleted after some time. It seems that something creates a lot of noise. Can you share some content of the logs on pastebin.com?

kadlma, 2019-02-02:
> Hi there - I set up my raspibolt just a little over a week ago as per this guide. Since then, it appears my /dev/root directly has ballooned up to ~30G and is now full.
>
> Upon investigation, it [...]

I do have exactly the same problem as you do, after reboot the node is full of "No space left on device" errors.

`Filesystem      Size  Used Avail Use% Mounted on
/dev/root        30G   30G     0 100% /
devtmpfs        460M     0  460M   0% /dev
tmpfs           464M     0  464M   0% /dev/shm
tmpfs           464M  6.3M  458M   2% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           464M     0  464M   0% /sys/fs/cgroup
/dev/mmcblk0p1   44M   23M   22M  51% /boot
tmpfs            93M     0   93M   0% /run/user/1000
tmpfs            93M     0   93M   0% /run/user/1001`

Stadicus, 2019-02-04:
Can you check the content of the biggest logfiles and post some of your finding (repeated error messages, etc) here?

faket0shi, 2019-02-12:
I've had the same problem for a couple of days now. I copy a fragment of the syslog file because daemon.log is too big to open.

Feb 12 06:27:07 FakeToshi lnd[733]: 2019-02-12 06:27:07.940 [DBG] PEER: Sending ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=605208583406354432, mflag$
Feb 12 06:27:07 FakeToshi lnd[733]: 2019-02-12 06:27:07.940 [DBG] PEER: Sending ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=605208583406354432, mflag$
Feb 12 06:27:07 FakeToshi lnd[733]: 2019-02-12 06:27:07.941 [DBG] PEER: Sending ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=605208583406354432, mflag$
Feb 12 06:27:07 FakeToshi lnd[733]: 2019-02-12 06:27:07.941 [DBG] PEER: Sending ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=612680864475119617, mflag$
Feb 12 06:27:07 FakeToshi lnd[733]: 2019-02-12 06:27:07.941 [DBG] PEER: Sending ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=612680864475119617, mflag$
Feb 12 06:27:07 FakeToshi lnd[733]: 2019-02-12 06:27:07.941 [DBG] PEER: Sending ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=605205284848795648, mflag$
Feb 12 06:27:07 FakeToshi lnd[733]: 2019-02-12 06:27:07.941 [DBG] PEER: Sending ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=605208583406354432, mflag$
Feb 12 06:27:07 FakeToshi lnd[733]: 2019-02-12 06:27:07.941 [DBG] PEER: Sending ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=615324090335232001, mflag$
Feb 12 06:27:07 FakeToshi lnd[733]: 2019-02-12 06:27:07.941 [DBG] PEER: Sending ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=612680864475119617, mflag$
Feb 12 06:27:07 FakeToshi lnd[733]: 2019-02-12 06:27:07.941 [DBG] PEER: Sending ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=605208583406354432, mflag$
Feb 12 06:27:07 FakeToshi lnd[733]: 2019-02-12 06:27:07.941 [DBG] PEER: Sending ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=605208583406354432, mflag$
Feb 12 06:27:07 FakeToshi lnd[733]: 2019-02-12 06:27:07.942 [DBG] PEER: Sending ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=605208583406354432, mflag$
Feb 12 06:27:07 FakeToshi lnd[733]: 2019-02-12 06:27:07.942 [DBG] PEER: Sending ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=612680864475119617, mflag$
Feb 12 06:27:07 FakeToshi lnd[733]: 2019-02-12 06:27:07.942 [DBG] PEER: Sending ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=612680864475119617, mflag$
Feb 12 06:27:07 FakeToshi lnd[733]: 2019-02-12 06:27:07.942 [DBG] PEER: Sending ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=612680864475119617, mflag$
Feb 12 06:27:07 FakeToshi lnd[733]: 2019-02-12 06:27:07.941 [DBG] PEER: Sending ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=612680864475119617, mflag$
Feb 12 06:27:07 FakeToshi lnd[733]: 2019-02-12 06:27:07.942 [DBG] PEER: Sending ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=605205284848795648, mflag$
Feb 12 06:27:07 FakeToshi lnd[733]: 2019-02-12 06:27:07.942 [DBG] PEER: Sending ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=605208583406354432, mflag$
Feb 12 06:27:07 FakeToshi lnd[733]: 2019-02-12 06:27:07.942 [DBG] PEER: Sending ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=615324090335232001, mflag$
Feb 12 06:27:07 FakeToshi lnd[733]: 2019-02-12 06:27:07.942 [DBG] PEER: Sending ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=605205284848795648, mflag$
Feb 12 06:27:07 FakeToshi lnd[733]: 2019-02-12 06:27:07.941 [DBG] PEER: Sending ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=605208583406354432, mflag$
Feb 12 06:27:07 FakeToshi lnd[733]: 2019-02-12 06:27:07.942 [DBG] PEER: Sending ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=605205284848795648, mflag$
Feb 12 06:27:07 FakeToshi lnd[733]: 2019-02-12 06:27:07.942 [DBG] PEER: Sending ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=615324090335232001, mflag$
Feb 12 06:27:07 FakeToshi lnd[733]: 2019-02-12 06:27:07.943 [DBG] PEER: Sending ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=612680864475119617, mflag$
Feb 12 06:27:07 FakeToshi lnd[733]: 2019-02-12 06:27:07.943 [DBG] PEER: Sending ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=615324090335232001, mflag$
[... cut ...]
      4 drwxr-xr-x 11 root root       4096 Jan 18 13:48 ..
      4 drwxr-xr-x  2 root root       4096 Feb  2 11:23 apt
      4 drwx--x--x  2 root root       4096 Nov 13 14:11 lightdm
      4 drwxr-x---  2 root adm        4096 Aug 13 21:49 samba
      4 -rw-------  1 root utmp       1536 Jan 21 15:26 btmp
      0 -rw-r--r--  1 root root          0 Nov 13 14:10 bootstrap.log
      0 -rw-r--r--  1 root root          0 Nov 13 14:10 fontconfig.log

yesterday I deleted the logs, so the size of the logs is less than 12 hours.

Stadicus, 2019-02-12:
Not sure why this FakeToshi node causes so many channel updates. But these are Debugging messages. Are you sure you have `debuglevel=info` set in the `lnd.conf`?

faket0shi, 2019-02-13:
You're right, I was in debuglevel=debug but I do not know how it has changed. I have left it in =info and it seems that everything is correct, thank you.

robclark56, 2019-02-19:
More on this ... I also ran out of space on /

I found my logs were full of gazillions of messages like this:
```
Feb 19 08:17:34 RaspiBolt lnd[19836]: 2019-02-19 08:17:34.353 [ERR] DISC: unable to validate channel update announcement for short_chan_id=(lnwire.ShortChannelID) 557818:1437:1
Feb 19 08:17:34 RaspiBolt lnd[19836]: : invalid signature for channel update (*lnwire.ChannelUpdate)(0x226a41b0)({
Feb 19 08:17:34 RaspiBolt lnd[19836]:  Signature: (lnwire.Sig) (len=64 cap=64) {
Feb 19 08:17:34 RaspiBolt lnd[19836]:   00000000  68 0b 03 8e 2c 65 45 05  82 9d f4 2b 28 54 76 ef  |h...,eE....+(Tv.|
Feb 19 08:17:34 RaspiBolt lnd[19836]:   00000010  34 32 25 e0 4d a5 f3 2c  0a 3e 0e 07 cc 91 0d 46  |42%.M..,.>.....F|
Feb 19 08:17:34 RaspiBolt lnd[19836]:   00000020  28 e8 1c bd 93 0f 03 e3  1e d9 18 52 70 ae 74 de  |(..........Rp.t.|
Feb 19 08:17:34 RaspiBolt lnd[19836]:   00000030  ac 40 84 85 b5 07 46 a9  64 a5 c4 a8 be 55 6e 9c  |.@....F.d....Un.|
Feb 19 08:17:34 RaspiBolt lnd[19836]:  },
Feb 19 08:17:34 RaspiBolt lnd[19836]:  ChainHash: (chainhash.Hash) (len=32 cap=32) 000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f,
Feb 19 08:17:34 RaspiBolt lnd[19836]:  ShortChannelID: (lnwire.ShortChannelID) 557818:1437:1,
Feb 19 08:17:34 RaspiBolt lnd[19836]:  Timestamp: (uint32) 1550535425,
Feb 19 08:17:34 RaspiBolt lnd[19836]:  Flags: (lnwire.ChanUpdateFlag) 259,
Feb 19 08:17:34 RaspiBolt lnd[19836]:  TimeLockDelta: (uint16) 144,
Feb 19 08:17:34 RaspiBolt lnd[19836]:  HtlcMinimumMsat: (lnwire.MilliSatoshi) 1000 mSAT,
Feb 19 08:17:34 RaspiBolt lnd[19836]:  BaseFee: (uint32) 1000,
Feb 19 08:17:34 RaspiBolt lnd[19836]:  FeeRate: (uint32) 1
Feb 19 08:17:34 RaspiBolt lnd[19836]: })
```
My lnd.conf has
```
[Application Options]
debuglevel=info
debughtlc=false
```

Stadicus, 2019-02-21:
Seems that the automatic log rotation is not always sufficient. Have a look at this here, I think this topic needs to be included in the guide. Any help appreciated! :-)

https://www.networkworld.com/article/3218728/linux/how-log-rotation-works-with-logrotate.html

robclark56, 2019-02-21:
not elegant...but here is how I quickly worked around this

admin ~  ฿  ls /etc/cron.hourly/clean_logs
-rwxr-xr-x 1 root root 58 Feb 18 14:18 /etc/cron.hourly/clean_logs
admin ~  ฿  cat /etc/cron.hourly/clean_logs
#!/bin/bash
rm /var/log/syslog.*
rm /var/log/daemon.log.*
admin ~  ฿

Stadicus, 2019-03-03:
Ok, that's blunt force... :-)

robclark56, 2019-03-03:
:-)  Yes .... had to find a quick fix as running out of space on "/" is not cool.

But .... this weekend I migrated from V0.4.2 to V0.5.2. Guess what? All those messages filling up my log files have stopped!

My GUESS: V0.4.2 was not happy with some messages coming out of the newer LND versions.

My other Guess?: Prior to the migration, I closed a few channels (as that was the advice from others who had seen funds lost if upgrading). Maybe one of those channels was responsible.

Anyway - there is a datapoint for you.

closed 2020-02-28 by Stadicus