raspibolt/raspibolt #350: Electrum Wallet on Testnet

badokun, 2019-01-17:
I'm running the EPS and when launching the portable version of Electrum with 
```
electrum-3.3.2-portable.exe --oneserver --server 192.168.1.40:50002:s --testnet
```
able to connect successfully, however when creating the wallet in Electrum, 
* Standard Wallet
* I already have a seed
* Options, check BIP39 seed
* enter the 24 seed words as generated by the lncli create command

I get checksum: failed. When proceeding, the resulting addresses all start with `tb`, whereas the ones created on the RaspiBolt with `lncli --network=testnet newaddress np2wkh` looks different.

Any ideas? Seems like Electrum doesnt like the 24 seeds generated thru lncli

MrPaz, 2019-01-18:
electrum uses a 12 word seeds.  the 24 word wallet generated by lncli is a lnd lightning wallet, which, unfortunately, won't work with electrum.

badokun, 2019-01-18:
Right, but the raspibolt guide helps you set up a personal server only to be let down further down the road. What's the point?

badokun, 2019-01-19:
Fair enough, I was just hoping I could also manage my wallet that was created as part of `Lnd create`.

closed 2019-01-19 by badokun