raspibolt/raspibolt #352: Update "Bonus guide: Auto-unlock LND on startup"

daniel-vera-g, 2019-01-17:
This is more a known open task for someone who knows how to auto-start lnd after start up.

Even though in the [Bonus guide: Auto-unlock LND on startup](https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_6A_auto-unlock.md) the script to autostart says `Updated for LND 0.5 and above` it does, I can confirm that the whole process does not work as intended(As it says in the header: `Please note: this guide has not been updated to LND 0.5 yet and might not work as intended.`).

After doing every step and restarting LND, LND hangs up in: `Jan 17 17:19:39 dvg-raspiBolt systemd[1]: Started LND Lightning Daemon.
`

Maybe someone knows how to fix it.

Stadicus, 2019-01-29:
I updated the Auto Unlock script:
https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_6A_auto-unlock.md

Can you give it a try? The logging is better now, most issues come with the TLS certificates.

daniel-vera-g, 2019-01-29:
> I updated the Auto Unlock script:
> [/raspibolt/raspibolt_6A_auto-unlock.md@`master`](https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_6A_auto-unlock.md)
> 
> Can you give it a try? The logging is better now, most issues come with the TLS certificates.

Nice Thx, will give it a try this weekend and update this issue ;)

Stadicus, 2019-02-08:
Thanks for your feedback, glad to hear! And good to know that a RPi2 is also well suited for the task (I never personally checked).

compane, 2019-02-09:
I'm running into an TLS error. Any advise?

{"error":"all SubConns are in TransientFailure, latest connection error: connection error: desc = \"transport: authentication handshake failed: x509: cannot validate certificate for 127.0.0.1 because it doesn't contain any IP SANs\"","code":14}

lnd version: 0.5.2-beta
bitcoin: 0.17.1

Thanks!

daniel-vera-g, 2019-02-10:
> I'm running into an TLS error. Any advise?
> 
> {"error":"all SubConns are in TransientFailure, latest connection error: connection error: desc = "transport: authentication handshake failed: x509: cannot validate certificate for 127.0.0.1 because it doesn't contain any IP SANs"","code":14}
> 

Can you give more information about the issue?

What did you before the issue?
What action causes this TLS error?

compane, 2019-02-10:
Environment
Odroid xu4 / Bitcoin (0.17.1) and LND (0.5.2) node on an external ssd.

1. What I did
++Moved from ubuntu 16.04 to 18.04. Updated LND from 0.5.1 to 0.5.2. BTC and LND nodes are running fine.
2. What causes this error?
++Set up auto unlock via [guide ](https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_6A_auto-unlock.md). Node restart looks like this:

> Feb 10 14:18:27 thundroid systemd[1]: Starting LND Lightning Daemon...
> Feb 10 14:18:27 thundroid lnd[13376]: Attempting automatic RPC configuration to bitcoind
> Feb 10 14:18:27 thundroid lnd[13376]: Automatically obtained bitcoind's RPC credentials
> Feb 10 14:18:27 thundroid lnd[13376]: 2019-02-10 14:18:27.730 [INF] LTND: Version: 0.5.2-beta commit=v0.5.2-beta, build=production, logging=default
> Feb 10 14:18:27 thundroid lnd[13376]: 2019-02-10 14:18:27.730 [INF] LTND: Active chain: Bitcoin (network=mainnet)
> Feb 10 14:18:27 thundroid lnd[13376]: 2019-02-10 14:18:27.731 [INF] CHDB: Checking for schema update: latest_version=7, db_version=7
> Feb 10 14:18:27 thundroid lnd[13376]: 2019-02-10 14:18:27.745 [INF] RPCS: password RPC server listening on 127.0.0.1:10009
> Feb 10 14:18:27 thundroid lnd[13376]: 2019-02-10 14:18:27.745 [INF] RPCS: password gRPC proxy started at 127.0.0.1:8080
> Feb 10 14:18:27 thundroid lnd[13376]: 2019-02-10 14:18:27.746 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.
> Feb 10 14:18:37 thundroid systemd[1]: Started LND Lightning Daemon.

daniel-vera-g, 2019-02-10:
> Environment
> Odroid xu4 / Bitcoin (0.17.1) and LND (0.5.2) node on an external ssd.
>
> 1. What I did
>    ++Moved from ubuntu 16.04 to 18.04. Updated LND from 0.5.1 to 0.5.2. BTC and LND nodes are [...]

Hmm, I've not had the time to test the updated version but this was the exact output I got and that's why I opened this issue.  
I also am not using Ubuntu.  
I suppose your `systemd --version` is higher 230+.  

Will try to make time test the updated version this week and maybe I can help you out.

compane, 2019-02-10:
> I suppose your `systemd --version` is higher 230+.

Yes. That was my motivation to move from 16.04 to 18.04 --> systemd 237

Stadicus, 2019-02-21:
@compane: 
* do you have the `restlisten=localhost:8080` in your lnd.conf?
* did you delete the TLS files after the change, so that new certificates have been created?
* did you copy the new tls files to the admin .lnd folder? (not relevant for this issue, just to use lncli in general)

lechatnieger, 2019-02-21:
hi same error here, lnd is not auto unlocking

i have to manually run "lncli unlock"

i added restlisten=localhost:8080

i ' am using latest unlock script.

Where is located tls files? it is safe to cancel the tls file?

thank you

compane, 2019-02-21:
> @compane:
> 
> * do you have the `restlisten=localhost:8080` in your lnd.conf?
> * did you delete the TLS files after the change, so that new certificates have been created?
> * did you copy the new tls files to the admin .lnd folder? (not relevant for this issue, just to use lncli in general)

Thank you so much! Deleting the TLS files solved my problem.

compane, 2019-02-21:
> [...] 
> Where is located tls files? it is safe to cancel the tls file?
> 
> thank you

The files (tls.cert and tls.key) are located in '/home/bitcoin/.lnd'.

lechatnieger, 2019-02-22:
after deleting tls files i receive the followng error
[lncli] rpc error: code = Unavailable desc = all SubConns are in TransientFailure, latest connection error: connection error: desc = "transport: authentication handshake failed: x509: certificate signed by unknown authority"

and i cannot unlock lnd wallet

what can i do now?

thank you

Stadicus, 2019-03-01:
To explain: adding additional ip addresses or hostnames (like localhost) to the lnd.conf file does not automatically add these to the TLS certificate. You need to delete the old certificates so that LND generates a new one including the additional interfaces. 

Until you do, the unlock script cannot authenticate to the API and gets a TLS error.

dawiepoolman, 2019-03-21:
I get the same error as @daniel-vera-g even after deleting tls and macaroons
I suppose this is why the issue is still open?

dawiepoolman, 2019-03-22:
I re-tried deleting tls.cert and recreated under the bitcoin user and transferred to admin as explained in the guide and it works fine now

closed 2020-02-28 by Stadicus