raspibolt/raspibolt #353: Problem when running sudo tail -f

CryptoAstroNut, 2019-01-17:
I have just finished uploading the blockchain mainnet and am attempting to start the bitcoind.
I was able to use:
$ sudo systemctl start bitcoind  
$ systemctl status bitcoind.service  -This returns "active (running)" 

When I attempt to use "$sudo tail -f /home/bitcoin/.bitcoin/debug.log" it returns:
tail: cannot open '/home/bitcoin/.bitcoin/debug.log' for reading: No such file or directory
tail: no files remaining
I attemtped- "$ bitcoin-cli getblockchaininfo" it returns:
error could not connect to the server 127.0.0.1:8332 

Any idea where I screwed up?

CryptoAstroNut, 2019-01-17:
Update: I powered down and restarted the Raspi and was able to get the debug.log to run. but I keep getting Error loading block data base and then it aborts block database rebuild, and shutdowns and then repeats that log over and over again.

CryptoAstroNut, 2019-01-18:
second update: I believe I discovered the issue. I was using a previously loaded version of core and it must not have have txindex=1  
I believe to fix this I need to reindex, but am unsure of the exact command to do this.

Septem151, 2019-01-19:
Are you running on testnet? If you're on testnet, you need to run
```$ sudo tail -f /home/bitcoin/.bitcoin/testnet3/debug.log```
if you're running on mainnet, you would use
```$ sudo tail -f /home/bitcoin/.bitcoin/debug.log```
You shouldn't need txindex=1 set in order to open the file of blocks. Try going into the .bitcoin folder
```$ cd /home/bitcoin/.bitcoin```
and then do ```ls```. Make sure you see folders for: "blocks" and "chainstate".
[image]

CryptoAstroNut, 2019-01-19:
I'm attempting to switch over to the mainnet. This is what  I see when I use ls 
[image]

Stadicus, 2019-01-20:
Not sure if your data directory is corrupted, or if it's just the missing txindex. As txindex is no longer needed, you can delete it from `bitcoin.conf` and start bitcoind manually. Follow the steps in the "Troubleshooting" section carefully:
https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_70_troubleshooting.md

If you keep getting this error, maybe the blockchain data is corrupt. In that case, I'd just copy it again from your regular computer (again, no need for txindex anymore).

CryptoAstroNut, 2019-01-20:
I went through the troubleshooting right before posting the last two screenshots. I'll try uploading again. Thanks. 
One last question, I am having to resync bitcoin core to my computer after doing the uplaod, is this normal? It basically restarted the whole wallet to day 1.

Stadicus, 2019-01-21:
No, I cannot imagine why Bitcoin Core on your regular computer would complain just because you copied some files from the data directory? Did you close the application before sync? If not, maybe some files got locked and confused the app, causing a resync?

CryptoAstroNut, 2019-01-21:
I'm pretty sure it was completed. took a full day. Being that this is the first time 've done anything like this there is a good chance that I missed or messed something up along the line. Core is almost done resyncing so I'll be able to try to upload again shortly.

Stadicus, 2019-01-29:
Ok, just make sure to stop all running software (Bitcoin Core on your regular computer, bitcoind & lnd on the Pi) before copying / changing data in the respective working directories. Let me know how it goes.

CryptoAstroNut, 2019-01-31:
got everything resynced and uploaded both files again. Im getting a different error code now. it looks like it start to load the block index successfully but then runs into an issue when using the obfuscation key. 
[image]

CryptoAstroNut, 2019-02-08:
The original block and chainstate files are being transferred from my laptop to an external HDD using winSCP. Should I delete from the laptop and start over or delete from the HDD, If from the HDD what would be the way to do that?

CryptoAstroNut, 2019-02-08:
so I decided to download core once again but on a different computer and while doing this I came to a possible answer to my problem. I had the 64bit version on my laptop and I noticed in the troubleshooting that only the 32bit version would work on the pi. Could this have been the issue?

Cees2daBees, 2019-02-08:
I don't think that would be it. I've also got the 64-bit version of Core on my laptop and the 32-bit version on my raspibolt. I don't think it affects the blockchain data.

CryptoAstroNut, 2019-02-09:
Well I'm not sure what the issue was but it's fixed. I deleted the two directories like you said and using a different computer I downloaded the 32 bit version and now everything seems to be working fine. Like Cees said downloading the 32bit may not have been the issue but its working now so who knows. Thank you everyone for your help and hopefully you wont see me in this chat are anytime soon. lol

closed 2019-02-10 by CryptoAstroNut