raspibolt/raspibolt #355: bitcoin-qt not using the blockchain of bitcoind

walterkok, 2019-01-19:


Stadicus, 2019-01-20:
bitcoin-qt stores some settings apart from the config file, but it should not be a problem to share the data directory. I don't think this is advisable, though, as my guess is that running bitcoind & bitcoin-qt at the same time (eg. by mistake, just once) could lead to problem.

That being said, you can configure both the data dir and the config file explicitly:
```
-conf=<file> 	  Specify configuration file (default: bitcoin.conf)
-datadir=<dir> 	Specify data directory 
```
https://en.bitcoin.it/wiki/Running_Bitcoin

walterkok, 2019-01-20:
Thank you Stadicus, I did install everything according to your guide (other than allowing VNC server afterwards) so the following should do the trick to test (would appreciate your validation!) 

As admin:

$ sudo systemctl stop bitcoind
$ sudo systemctl stop lnd

then Open VNC session, in that session a console
$ sudo su - bitcoin
$ cd /usr/local/bin/ #in this location is bitcoin-qt

$ bitcoin-qt -conf=/home/bitcoin/.bitcon/bitcoin.conf -datadir=/home/bitcoin/.bitcon/testnet3/

Stadicus, 2019-01-21:
Seems legit, except for the `.bitcon` typo... :-)

Did you try it out yet?

walterkok, 2019-01-21:
Hi Stadicus, I managed to get it working today, the following did the trick:

As admin:

$ sudo systemctl stop bitcoind
$ sudo systemctl stop lnd

then Open VNC session, in that session a console
It is imporant that the VNC session is also logged in as bitcoin otherwise the bitcoin-qt does not start (cannot display its output). So I logged also into the Raspi VNC environment as bitcoin
$ # sudo su - bitcoin # now this is not needed
$ cd /usr/local/bin/ #in this location is bitcoin-qt

$ bitcoin-qt -conf=/home/bitcoin/.bitcoin/bitcoin.conf -datadir=/home/bitcoin/.bitcon
(so no reference to the testnet3 directory). 

Bitcoin-qt started nicely on the Testchain (testnet=1 is in .conf) and I managed to get some testcoins in the wallet from a faucet. Thanks !

walterkok, 2019-01-21:
last command should be
$ bitcoin-qt -conf=/home/bitcoin/.bitcoin/bitcoin.conf -datadir=/home/bitcoin/.bitcoin
(no /testnet3/ but also no typo :-)

Stadicus, 2019-01-25:
Great, thanks for the update!

closed 2019-01-25 by walterkok